evcc-io/evcc #19159: UI: Messaging Configured zeigt immer "nein"

TobiasHuber1980, 2025-02-25:
### Describe the bug

seit 0.200.3 Ref: [Config: show messaging configured state (v2) (](https://github.com/evcc-io/evcc/commit/c59f1b1ca76cd5031eb866a91849f76c94ebaaf6) zeigt UI "Benachrichtigungen Konfiguriert
"Nein", ist jedoch konfiguriert und das Messaging funktioniert auch.

### Steps to reproduce

1. Update auf 0.200.3
2. Anzeige im UI Benachrichtigungen steht auf "Konfiguriert Nein"

[image]

3.
...

### Configuration details

```yaml
alles im UI konfiguriert
```

### Log details

```text
habe auch mal an der Messaging Konfiguration etwas geändert, reboot, bleibt auf "aus"
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.3

TobiasHuber1980, 2025-02-25:
War schon seit längerem im UI konfiguriert. Ich meine das war copy past aus der yaml. 
Evtl. daher? https://github.com/evcc-io/evcc/commit/af8a57544aa059cd3a799030dc885678a374593a

[image]

VolkerK62, 2025-02-25:
betrifft auch "modbusproxy" und "hems"
ist bei mir auch schon länger im UI konfiguriert (nicht per migrate)

Lako-Unbekannt, 2025-02-25:
@TobiasHuber1980 man kann die geschwärzten Stellen lesen.

andig, 2025-02-25:
Wir finden es generell gut, auf Screenshots zu verzichten ;)

naltatis, 2025-02-25:
> Wir finden es generell gut, auf Screenshots zu verzichten ;)

Das würd ich nicht ausnahmslos unterschreiben 😄

naltatis, 2025-02-25:
https://github.com/evcc-io/evcc/pull/19184 ist nicht die Lösung. Ich kann das reproduzieren. Fix kommt ...

closed 2025-02-25 by naltatis