evcc-io/evcc #19168: UI Config loadpoint [db:2] charger: not found: db:1

documentONE, 2025-02-25:
### Describe the bug

Since Update from 0.200.1 to 0.200.2 I can't add a charger via the UI configuration and evcc won't start
Also creating new configuration without charger results in same error

### Steps to reproduce

1. creating loadpoint in UI config
2. setting up meter, battery, pv
3. restart

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.iot # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 15s # control cycle interval

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /data/evcc/evcc.db
```

### Log details

```text
[main ] INFO 2025/02/25 11:02:51 evcc 0.200.3
[main ] INFO 2025/02/25 11:02:51 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/25 11:02:51 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/25 11:02:52 listening at :7070
[bmw ] DEBUG 2025/02/25 11:02:52 identity.Login - database token found
[main ] FATAL 2025/02/25 11:02:55 loadpoint [db:2] charger: not found: db:1
[main ] FATAL 2025/02/25 11:02:55 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.3

naltatis, 2025-02-25:
@documentONE I can't reproduce this. Please provide more information. Did you try removing the db?

naltatis, 2025-02-25:
@documentONE You can also send me you `evcc.db` to michael@evcc.io for inspection.

naltatis, 2025-02-25:
I was able to reproduce this. This inconsistent state can occur of you:

- add a new loadpoint with a charger
- edit loadpoint, edit charger, delete charger
- restart

Now we've a loadpoint with an invalid charger (delete) reference. 

@andig how do we want to solve this? Clean way would if the delete charger call would remove the charger ref at the loadpoint. Doing this via an "update loadpoint" api call after the "delete charger" succeeded would be an option but we don't have transactional security (browser/network errors possible).

In any case we need to be able to handle loadpoints without an associated charger.

Nitrox912, 2025-02-25:
Ich konnte das Problem beheben, indem ich ein Backup eingespielt habe.

In meinem Fall war bei Loadpoint db:2 das Fahrzeug db:5 als Standard definiert. Nachdem ich dieses erst im Loadpoint db:2 entfernte, neu startete und dann das Fahrzeug löschte, kam es zu keinem Fehler mehr.

naltatis, 2025-02-26:
> In meinem Fall war bei Loadpoint db:2 das Fahrzeug db:5 als Standard definiert.

Ja, passt ins Fehlerbild. Danke.

olikra, 2025-03-10:
Hier ist es nun auch unter 0.200.8 passiert. Habe die ursprüngliche yaml-Konfiguration wieder eingespielt und die experimentelle Funktion in der UI ausgeschaltet. 

Werden trotzdem die Konfigurationen aus der DB genommen? 
Muss ich also die DB editieren und alle Konfigurationen löschen um die stabile Nutzung via yaml zu ermöglichen?

andig, 2025-03-10:
Warum sollte das nicht passieren?

andig, 2025-03-10:
Ja. Deshalb gibt es dieses offene Issue. Keine neue Erkenntnis.

closed 2025-03-12 by andig