evcc-io/evcc #19170: Update auf 200.3 cannot Create Vehicle

theoldphilip, 2025-02-25:
### Describe the bug

[main ] ERROR 2025/02/25 12:45:40 creating vehicle Xceed failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: deviceid not found
[main ] ERROR 2025/02/25 12:45:40 creating vehicle NiroEV failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: deviceid not found

Vor dem Update auf 200.3 war alles in Ordnung

### Steps to reproduce

1.every Reboot
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
[main ] INFO 2025/02/25 12:45:37 evcc 0.200.3
[main ] INFO 2025/02/25 12:45:37 using config file: /config/evcc.yaml
[db ] INFO 2025/02/25 12:45:37 using sqlite database: /data/evcc.db
[mqtt ] INFO 2025/02/25 12:45:38 connecting evcc-603342046 at tcp://localhost:1883
[mqtt ] DEBUG 2025/02/25 12:45:38 tcp://localhost:1883 connected
[main ] INFO 2025/02/25 12:45:38 listening at :7070
[main ] ERROR 2025/02/25 12:45:40 creating vehicle Xceed failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: deviceid not found
[main ] ERROR 2025/02/25 12:45:40 creating vehicle NiroEV failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: deviceid not found
[lp-1 ] DEBUG 2025/02/25 12:45:40 set title: kleine Wallbox
[lp-1 ] DEBUG 2025/02/25 12:45:40 set priority: 0
[lp-1 ] DEBUG 2025/02/25 12:45:40 set smart cost limit: <nil>
[lp-1 ] DEBUG 2025/02/25 12:45:40 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-1 ] DEBUG 2025/02/25 12:45:40 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/02/25 12:45:40 set session energy limit: 0
[lp-1 ] DEBUG 2025/02/25 12:45:40 set session soc limit: 0
[lp-1 ] DEBUG 2025/02/25 12:45:40 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000aaa78c}
[lp-1 ] DEBUG 2025/02/25 12:45:40 set default mode: pv
[lp-1 ] DEBUG 2025/02/25 12:45:40 set phases: 1
[lp-2 ] DEBUG 2025/02/25 12:45:40 set title: Wallbox
[lp-2 ] DEBUG 2025/02/25 12:45:40 set priority: 1
[lp-2 ] DEBUG 2025/02/25 12:45:40 set smart cost limit: <nil>
[lp-2 ] DEBUG 2025/02/25 12:45:40 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-2 ] DEBUG 2025/02/25 12:45:40 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-2 ] DEBUG 2025/02/25 12:45:40 set session energy limit: 0
[lp-2 ] DEBUG 2025/02/25 12:45:40 set session soc limit: 0
[lp-2 ] DEBUG 2025/02/25 12:45:40 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000c936dc}
[lp-2 ] DEBUG 2025/02/25 12:45:40 set default mode: pv
[lp-2 ] DEBUG 2025/02/25 12:45:40 set phases: 3
[lp-1 ] DEBUG 2025/02/25 12:45:42 charge total import: 487.114kWh
[lp-2 ] DEBUG 2025/02/25 12:45:42 charge total import: 3710.900kWh
[site ] WARN 2025/02/25 12:45:43 vehicle 'NiroEV' does not support automatic detection
[site ] WARN 2025/02/25 12:45:43 vehicle 'Xceed' does not support automatic detection
[site ] INFO 2025/02/25 12:45:43 site config:
[site ] INFO 2025/02/25 12:45:43 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/25 12:45:43 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2025/02/25 12:45:43 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2025/02/25 12:45:43 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site ] INFO 2025/02/25 12:45:43 vehicles:
[site ] INFO 2025/02/25 12:45:43 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site ] INFO 2025/02/25 12:45:43 vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2025/02/25 12:45:43 loadpoint 1:
[lp-1 ] INFO 2025/02/25 12:45:43 mode: pv
[lp-1 ] INFO 2025/02/25 12:45:43 charger: power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/02/25 12:45:43 meters: charge ✓
[lp-1 ] INFO 2025/02/25 12:45:43 charge: power ✓ energy ✓ currents ✗
[lp-2 ] INFO 2025/02/25 12:45:43 loadpoint 2:
[lp-2 ] INFO 2025/02/25 12:45:43 mode: off
[lp-2 ] INFO 2025/02/25 12:45:43 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-2 ] INFO 2025/02/25 12:45:43 meters: charge ✓
[lp-2 ] INFO 2025/02/25 12:45:43 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/02/25 12:45:43 set buffer soc: 90
[site ] DEBUG 2025/02/25 12:45:43 set buffer start soc: 95
[site ] DEBUG 2025/02/25 12:45:43 set priority soc: 90
[site ] DEBUG 2025/02/25 12:45:43 set battery discharge control: true
[site ] DEBUG 2025/02/25 12:45:43 set grid charge limit: 0.1
[lp-1 ] DEBUG 2025/02/25 12:45:43 phase timer inactive
[lp-1 ] DEBUG 2025/02/25 12:45:43 pv timer inactive
[lp-1 ] INFO 2025/02/25 12:45:43 vehicle updated: unknown -> Xceed
[lp-2 ] DEBUG 2025/02/25 12:45:43 phase timer inactive
[lp-2 ] DEBUG 2025/02/25 12:45:43 pv timer inactive
[lp-2 ] INFO 2025/02/25 12:45:43 vehicle updated: unknown -> NiroEV
[site ] WARN 2025/02/25 12:45:43 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2025/02/25 12:45:43 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

200.3

andig, 2025-02-25:
> Vor dem Update auf 200.3 war alles in Ordnung

200.2 geht noch?

theoldphilip, 2025-02-25:
Jetzt leider auch nicht mehr, ich vermute mal die KIA Api .....

MichaelWitkenkamp, 2025-02-25:
Moin,

Ich hatte eben den gleichen Fehler. Jetzt nach einem Neustart hat er von meinen XCeed wieder alle Daten ohne Fehler abgeholt.

closed 2025-02-25 by andig