evcc-io/evcc #19173: enable und disable Trigger überschreiben sich gegenseitig

tipuraneo, 2025-02-25:
### Describe the bug

# Beobachtung
Zunächst betrug der Überschuss >4.140 W. Auto wurde dreiphasig geladen. Dann wurde eine Verbraucher zugeschaltet, der Überschuss lag zwischen 800 W (= Bezug) < Überschuss < 4.140 W. Evcc startet den enable timer bei 3 m, um auf einphasige Ladung umzuschalten. Soweit ist das ok. Dann wurde ein taktender Verbraucher (Herd) zugeschaltet, der den Bezug auf > 800 W stiegen lies. Evcc startete den disable timer mit 5 m. Der taktende Verbraucher schaltete sich vor Ablauf des disable times ab, der Überschuss stieg wieder auf 800 W (= Bezug) < Überschuss < 4.140 W. und der enable timer startete erneut bei 3 m. Während der taktende Verbraucher lief, lag der Überschuss vereinfacht gesagt zwischen 0 und 4.140 W. Innerhalb des enable times schaltete sich der taktende Verbraucher wieder ein und der disable timer wurde auf 5 m hochgesetzt. Das führte dazu, dass durchgängig dreiphasig mit 6 A und mehr Strom aus dem Netz geladen wurde als gewünscht.

# Fehler
Die Timer zum Umschalten von drei- auf einphasig und der Abschalt-Timer überschreiben sich gegenseitig und starteten immer wieder von neuem. Das führt dazu, nicht von drei- einphasig umgeschaltet wird.

# mögliche Lösung
Statt immer wieder neu startende Timer für die Hysterese zu verwenden, sollten, [wie hier vorgeschlagen](https://github.com/evcc-io/evcc/issues/17151), rollende Mittelwerte verwendet werden.

### Steps to reproduce

siehe Beobachtung

### Configuration details

```yaml
loadpoints:
  - title: Wallbox
    charger: wallbox1
    priority: 3
    vehicle: Model3
    enable:
      delay: 3m
      threshold: -700
    disable:
      delay: 5m
      threshold: 800
```

### Log details

```text
keine
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.1

andig, 2025-02-26:
> Statt immer wieder neu startende Timer für die Hysterese zu verwenden, sollten, https://github.com/evcc-io/evcc/issues/17151, ein rollender Mittelwerte für die Überschussermittlung verwendet werden.

Das haben wir nicht umgesetzt weil dieser Mittelwert nicht zu den Werten aus der UI passt und daher für viele Anwender unerwartetes Verhalten bedeutet.

closed 2025-02-26 by andig

tipuraneo, 2025-02-28:
> Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

Da ich das Verhalten nur zufällig am Smartphone beobachtet habe, es nur sehr selten auftritt und es nur schwer zu reproduzieren ist, habe ich leider keine Logs. Ist solches Verhalten denn anhand des Quellcodes denkbar? Hat jemand anderes ähnliches beobachtet?

> Das haben wir nicht umgesetzt weil dieser Mittelwert nicht zu den Werten aus der UI passt und daher für viele Anwender unerwartetes Verhalten bedeutet.

Das Argument ist für mich nicht nachvollziehbar, da aus der UI auch nicht hervorgeht, ab welchem Überschuss/Bezug Trigger ausgelöst werden. Eine sinnvolle Regelung sollte nicht zulasten der Nachvollziehbarkeit gehen. Für eine bessere Nachvollziehbarkeit könnte man die Mittelwerte entweder als "⌀ Netzbezug" und "⌀ Einspeisung" unterhalb der jeweiligen Momentanwerte oder als zusätzliche Spalte link neben den Momentanwerten (wie CO2-Wert) ausweisen.

[image]