evcc-io/evcc #19181: Keba Wallbox wird falscher Hersteller zugeordnet

jove01, 2025-02-25:
### Describe the bug

Ich habe die komplette evcc.yaml teils manuell auf "UI" umgestellt. Seitdem bekomme ich den Fehler

> Wallbox: charger out of sync: expected disabled, got enabled

Zuerst habe ich das Default-Intervall von 10s auf 30s umgestellt. Jedoch keine Verbesserung. Dann ist mir bei der Loadpoint Definition aufgefallen:

[image]

und dann

[image]

Hier sieht man den Hersteller "BMW" ob das jetzt der Grund für das ursprüngliche Problem ist.

Die Wallbox ist eine Keba P30 Deutschlandversion mit Version V3.10.51 (also Voraussetzung erfüllt), daher der zusätzliche Energiezähler. Doch hier ist auch ein falsches Model zugeordnet:

[image]

[image]

Tatsächlich ist es ein 

[image]

In der Keba sehe ich jetzt einen "nicht behebbaren Fehler", den ich eher als Folgeproblem sehe.

Doch ein Löschen des LP im UI und Aktivierung in der Yaml brachte nicht den Erfolg. Morgen werde ich dieWallbox neustarten.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
siehe oben
```

### Log details

```text
siehe oben
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

Noch nicht installiert

andig, 2025-02-25:
Das ist https://github.com/evcc-io/evcc/issues/19018

closed 2025-02-25 by andig

jove01, 2025-02-25:
Vielen Dank.
Ich warte also bezüglich der Benennung ab. 

Ich habe die Wallbox über die Elektro-Sicherung neu gestartet. Der Fehler an der Keba ist damit temporär beseitigt und ich hoffe, dass der Keba-Fehler dann nicht mehr auftritt.