evcc-io/evcc #19195: Kia Blulink not working correctly

winconlin, 2025-02-26:
### Describe the bug

Somehow evcc appears to only get the milage of my car and nothing more (via kia blulink)
Setup is a follows:
EVCC runs as Docker Container on Unraid.
I have a PV connected to my E3/DC S10 Battery
I have a E3/DC Multi Connect II (Which still is not supported by EVCC :( )
And a Kia EV9.

If I check the config in the GUI then my vehicle shows up as follows:

Kia Ev9
Kapazität           100,0 kWh (green)
Status                       Status (red)
Kilometerstand 15.427 km (green)
Reichweite                 0 km (red)
Ladestand                 0,0 % (red)
Fahrzeuglimit            0,0 % (red)

Which i find weird as it seems that excc can connect to my car (otherwise the milage would not show up) but it can not read everything from  my car...

### Steps to reproduce

1.Start the Docker
2.No new Info is available
3.Frustration sets in

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 84V3RYUN1QU313F0R3VCC0

interval: 30s # control cycle interval

sponsortoken: r1gHtl1k31mg0nN4p05Tth4T5tuffr1ghT1nh3rTth3r315n0W4yf0Rm3T03v3rD0Th4tbuTf331Fr33T0c0pyTh15t0lk3nf0r411y0uRn33Ds4nDh31p5pr34d1t4ll6l0ry707h3hypn0704dd15r364rd4llpr3v10u5c0mm4nd54ndc4lcul473p1

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: e3dc-rscp 
  usage: grid  
  host: 10.10.11.12  
  port: 5033  
  user: ih@tesp.am 
  password: S0m3th1ngv3rys3cr3t  
  key: 'n0ts0s3cr3t'  
  name: grid1
- type: template
  template: e3dc-rscp 
  usage: pv  
  host: 10.10.11.12  
  port: 5033  
  user: ih@tesp.am   
  password: S0m3th1ngv3rys3cr3t   
  key: 'n0ts0s3cr3t'  
  name: pv2
- type: template
  template: e3dc-rscp 
  usage: battery  
  host: 10.10.11.12  
  port: 5033  
  user: ih@tesp.am   
  password: S0m3th1ngv3rys3cr3t   
  key: 'n0ts0s3cr3t'  
  name: battery3

chargers:
- type: template
  template: phoenix-ev-eth 
  id: 255  
  host: 10.10.11.13  
  port: 502  
  modbus: tcpip  
  name: wallbox5

vehicles:
- type: template
  template: kia 
  user: ih@tesp.am  
  password: 3v3nm0r353cr3t5tuff 
  title: Kia Ev9  
  capacity: 100  
  language: de  
  name: ev4

loadpoints:
- title: Zuhause
  charger: wallbox5
  vehicle: ev4
  mode: pv

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  grid:
    type: template
    template: tibber
    token: y35bu74c7u4llyn0
```

### Log details

```text
[main ] INFO 2025/02/26 09:25:02 evcc 0.200.3 (8eccba69)
[main ] INFO 2025/02/26 09:25:02 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/26 09:25:02 using sqlite database: /.evcc/evcc.db
[main ] INFO 2025/02/26 09:25:03 listening at :7070
[site ] INFO 2025/02/26 09:25:04 site config:
[site ] INFO 2025/02/26 09:25:04 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/26 09:25:04 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2025/02/26 09:25:04 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2025/02/26 09:25:04 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site ] INFO 2025/02/26 09:25:04 vehicles:
[site ] INFO 2025/02/26 09:25:04 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1 ] INFO 2025/02/26 09:25:04 loadpoint 1:
[lp-1 ] INFO 2025/02/26 09:25:04 mode: pv
[lp-1 ] INFO 2025/02/26 09:25:04 charger: power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/02/26 09:25:04 meters: charge ✗
[lp-1 ] DEBUG 2025/02/26 09:25:04 phase timer inactive
[lp-1 ] DEBUG 2025/02/26 09:25:04 pv timer inactive
[lp-1 ] INFO 2025/02/26 09:25:04 vehicle updated: unknown -> Kia Ev9
[lp-1 ] ERROR 2025/02/26 09:25:04 charger enabled: write tcp 10.10.10.19:37742->10.10.11.13:502: write: connection reset by peer
[site ] DEBUG 2025/02/26 09:25:04 ----
[lp-1 ] DEBUG 2025/02/26 09:25:04 charge power: 0W
time="2025-02-26T09:25:04+01:00" level=info msg="Connecting to 10.10.11.12:5033"
time="2025-02-26T09:25:04+01:00" level=info msg="Connecting to 10.10.11.12:5033"
time="2025-02-26T09:25:04+01:00" level=info msg="Connecting to 10.10.11.12:5033"
time="2025-02-26T09:25:04+01:00" level=info msg="successfully connected to 10.10.11.12:5033"
time="2025-02-26T09:25:04+01:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2025-02-26T09:25:04+01:00" level=info msg="successfully connected to 10.10.11.12:5033"
time="2025-02-26T09:25:04+01:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2025-02-26T09:25:04+01:00" level=info msg="successfully connected to 10.10.11.12:5033"
time="2025-02-26T09:25:04+01:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2025-02-26T09:25:04+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
[site ] DEBUG 2025/02/26 09:25:04 grid power: -3W
time="2025-02-26T09:25:04+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
[site ] DEBUG 2025/02/26 09:25:04 pv 1 power: 996W
time="2025-02-26T09:25:04+01:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
[site ] DEBUG 2025/02/26 09:25:04 battery 1 power: -207W
[site ] DEBUG 2025/02/26 09:25:04 battery 1 soc: 3%
[site ] DEBUG 2025/02/26 09:25:04 site power: -210W
[lp-1 ] DEBUG 2025/02/26 09:25:04 vehicle odometer: 15427km
[lp-1 ] ERROR 2025/02/26 09:25:04 charger status: read tcp 10.10.10.19:37750->10.10.11.13:502: read: connection reset by peer
[site ] DEBUG 2025/02/26 09:25:34 ----
[lp-1 ] DEBUG 2025/02/26 09:25:34 charge power: 0W
[site ] DEBUG 2025/02/26 09:25:34 pv 1 power: 978W
[site ] DEBUG 2025/02/26 09:25:34 grid power: -15W
[site ] DEBUG 2025/02/26 09:25:34 battery 1 power: -228W
[site ] DEBUG 2025/02/26 09:25:34 battery 1 soc: 4%
[site ] DEBUG 2025/02/26 09:25:34 site power: -243W
[lp-1 ] ERROR 2025/02/26 09:25:34 charger status: write tcp 10.10.10.19:44700->10.10.11.13:502: write: connection reset by peer
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.3 (8eccba69)

andig, 2025-02-26:
> I have a E3/DC Multi Connect II (Which still is not supported by EVCC :( )

E3DC is supported

> Somehow evcc appears to only get the milage of my car and nothing more (via kia blulink)

    evcc vehicle --log trace,db:error

winconlin, 2025-02-26:
> > I have a E3/DC Multi Connect II (Which still is not supported by EVCC :( )
> 
> Does it have a documented api?
> 
I think so there have been a couple of attemts as far as i know:
https://github.com/evcc-io/evcc/discussions/14122
https://www.e3dc.com/wp-content/uploads/2022/05/E3DC_TDB_Wallbox-multi-connect.pdf
https://community.symcon.de/uploads/short-url/z6Yc7LiO6m9lJt8r5Aif539GbHI.pdf
https://community.simon42.com/t/e3dc-wallbox-multi-connect-2-leistung-auslesen/27164

> > Somehow evcc appears to only get the milage of my car and nothing more (via kia blulink)
> 
> ```
> evcc vehicle --log trace,db:error
> ```
Hope this does not contain any important Data - changed a few things like tolkens and stuff
`[main  ] INFO 2025/02/26 10:26:25 evcc 0.200.3 (8eccba69)
[main  ] INFO 2025/02/26 10:26:25 using config file: /etc/evcc.yaml
[kia   ] TRACE 2025/02/26 10:26:25 POST https://prd.eu-ccapi.kia.com:8080/api/v1/spa/notifications/register
[kia   ] TRACE 2025/02/26 10:26:26 {"pushRegId":"0647270206C83C3EC414974D62A317A3581277AC25FF058A67F350D4238C7610A15A6770C","pushType":"APNS","uuid":"526f2495b9-d2e8-47f6-8d39-1723ffd4e7d0c"}
--
{"retCode":"S","resCode":"0000","resMsg":{"deviceId":"a3615f21-46b3-4ca4-8fae-14ad846623e1"},"msgId":"c0626000-f423-11ef-8e6a-6bd7d912c3c8"}
[kia   ] TRACE 2025/02/26 10:26:26 GET https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=fdc285c00-0a2f-4c64-bcb4-2cfb1512007340a&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/redirect
[kia   ] TRACE 2025/02/26 10:26:26 <a href="https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset">Found</a>.
[kia   ] TRACE 2025/02/26 10:26:26 GET https://prd.eu-ccapi.kia.com:8080/web/v1/user/authorize?cache=reset
[kia   ] TRACE 2025/02/26 10:26:26 <!doctype html><html lang="en"><head><meta charset="utf-8"><meta name="viewport" content="width=device-width,initial-scale=1,shrink-to-fit=no,maximum-scale=1,user-scalable=0"><meta name="theme-color" content="#ffffff"><link href="https://fonts.googleapis.com/css?family=Nanum+Gothic:400,700,800&amp;subset=korean" rel="stylesheet"><link href="https://fonts.googleapis.com/icon?family=Material+Icons" rel="stylesheet"><link rel="manifest" href="/web/v1/user/manifest.json"><link rel="icon" href="/web/v1/user/favicon_kia.ico" type="image/x-icon"><title></title><script defer="defer" src="/web/v1/user/static/js/main.716bbfd9.js"></script><link href="/web/v1/user/static/css/main.7c0207e2.css" rel="stylesheet"></head><body><noscript>You need to enable JavaScript to run this app.</noscript><div id="root"></div></body></html>
[kia   ] TRACE 2025/02/26 10:26:26 POST https://prd.eu-ccapi.kia.com:8080/api/v1/user/language
[kia   ] TRACE 2025/02/26 10:26:26 {"lang":"de"}
[kia   ] TRACE 2025/02/26 10:26:26 GET https://prd.eu-ccapi.kia.com:8080/api/v1/user/integrationinfo
[kia   ] TRACE 2025/02/26 10:26:26 {"userId":"86744f75-6a8e-4666-85d0-16d7cb4c14bb","serviceId":"fdc85c00-0a2f-4c64-bcb4-2cfb1500730a","serviceName":"UVO"}
[kia   ] TRACE 2025/02/26 10:26:26 GET https://eu-account.kia.com/auth/realms/eukiaidm/protocol/openid-connect/auth?client_id=572e0304-5f8d-4b4c-9dd5-41aa84eed160&scope=openid+profile+email+phone&response_type=code&hkid_session_reset=true&redirect_uri=https://prd.eu-ccapi.kia.com:8080/api/v1/user/integration/redirect/login&ui_locales=en&state=fdc85c00-0a2f-4c64-bcb4-2cfb1500730a:86744f75-6a8e-4666-85d0-16d7cb4c14bb
[kia   ] TRACE 2025/02/26 10:26:26 <!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"  "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
    <html xmlns="http://www.w3.org/1999/xhtml" class="login-pf"
                    lang="en"
    >

    <head>
        <meta charset="utf-8">
        <meta http-equiv="Content-Type" content="text/html; charset=UTF-8" />
        <meta name="robots" content="noindex, nofollow">
        <meta http-equiv="Expires" content="Mon, 06 Jan 1990 00:00:01 GMT">
        <meta http-equiv="Expires" content="-1">
        <meta http-equiv="Pragma" content="no-cache">
        <meta http-equiv="Cache-Control" content="no-cache">

                <meta name="viewport" content="width=device-width,initial-scale=1"/>
        <title>Log in to Kia Account</title>
        <link rel="icon" href="/auth/resources/19ggl/login/kia-account-2022/img/favicon.ico?v=2" />

            
            <script type="text/javascript" src="https://cdn.cookielaw.org/consent/dea61314-e1fe-4d63-8605-6b28e56f24ea/OtAutoBlock.js?v=1" ></script>
            <script src="https://cdn.cookielaw.org/scripttemplates/otSDKStub.js?v=1" data-document-language="true" type="text/javascript" charset="UTF-8" data-domain-script="dea61314-e1fe-4d63-8605-6b28e56f24ea" ></script>
            <script type="text/javascript">function OptanonWrapper() { }</script>
            

                <link href="/auth/resources/19ggl/login/kia-account-2022/node_modules/bootstrap/dist/css/bootstrap.css?v=10" rel="stylesheet" />
                <link href="/auth/resources/19ggl/login/kia-account-2022/css/kia-login-2022.css?v=10" rel="stylesheet" />

                <script src="/auth/resources/19ggl/login/kia-account-2022/node_modules/jquery/dist/jquery.min.js" type="text/javascript"></script>
                <script src="/auth/resources/19ggl/login/kia-account-2022/node_modules/bootstrap/dist/js/bootstrap.min.js" type="text/javascript"></script>
[... cut ...] teControl":{"SleepMode":1},"ConnectedService":{"OTA":{"ControllerStatus":0}},"DrivingReady":0,"Version":"CCU2025022607281701","Date":"20250226072817.000","Offset":"1","Location":{"Date":"20250226072358.847","Heading":13,"Offset":1,"Servicestate":0,"TimeStamp":{"Day":26,"Hour":8,"Min":23,"Mon":2,"Sec":58,"Year":2025},"Version":"HU2025022608235801","GeoCoord":{"Altitude":0,"Latitude":234,"Longitude":234,"Type":0},"Speed":{"Unit":0,"Value":511.5}}}}},"msgId":"c0f3a380-f423-11ef-954b-c2efea609e71"}`

andig, 2025-02-26:
Bitte formatieren- danke!

andig, 2025-02-26:
> I think so there have been a couple of attemts as far as i know:

Die Doku enthält keinerlei Register über die sich die WB steuern ließe. Bitte an E3DC wenden.

winconlin, 2025-02-26:
> > I think so there have been a couple of attemts as far as i know:
> 
> Die Doku enthält keinerlei Register über die sich die WB steuern ließe. Bitte an E3DC wenden.

Email an E3DC ist raus.
Bleibt die frage warum evcc den Ladestand meines Kias nicht richtig anzeigt?

mfuchs1984, 2025-02-27:
Es sieht auf den ersten Blick in die logs so aus, als würden die Daten in der bluelink API Response schlicht fehlen.

winconlin, 2025-02-27:
> Es sieht auf den ersten Blick in die logs so aus, als würden die Daten in der bluelink API Response schlicht fehlen.

In allen anderen Anfragen sind die Infos aber vorhanden. Z.b. sehe ich in Home Assistent über die kia hyundai blulink integration ale Informationen wie zb Ladezustand etc.

andig, 2025-03-04:
Sieht aus als wäre das hier eine komplett neue API-Antwort mit neuer Struktur.

> sehe ich in Home Assistent über die kia hyundai blulink

Welche wäre das?

andig, 2025-03-04:
Sieht aus als wäre EV jetzt hier hin gewandert:

```json
                "Green": {
                    "DrivingReady": 0,
                    "PowerConsumption": {
                        "Prediction": {
                            "Climate": 0
                        }
                    },
                    "BatteryManagement": {
                        "SoH": {
                            "Ratio": 100
                        },
                        "BatteryRemain": {
                            "Value": 309765.6,
                            "Ratio": 90
                        },
                        "BatteryConditioning": 0,
                        "BatteryPreCondition": {
                            "Status": 0,
                            "TemperatureLevel": 1
                        },
                        "BatteryCapacity": {
                            "Value": 359280
                        }
                    },
```

FSTNDonald, 2025-03-06:
Test und Ergebnisse (Plattform ist Ubuntu 24.04.2 LTS // evcc 0.200.5):

Das Problem besteht seit ca. 1. März, heute (6.03.) folgender Test:
3. evcc service gestoppt, evcc.yaml gelöscht und mit sudo evcc configure neu aufgebaut
4. dabei erfolgte eine Fehlermeldung vom config script beim Konfigurieren des Fahrzeugs (KIA Bluelink), siehe anbei.
5. Schleife zurück zum erneuten Konfigurieren des Fahrzeugs, diesmal lief das EVCC config script ohne Fehlermeldung durch
6. restliche Schritte (Sponsor Token, WB usw.) einrichten, alles ohne Fehler
7. evcc.yaml nach /etc kopieren und evcc service neu starten

Resultat ist, dass in der iPhone GUI das Fahrzeug zwar als verbunden angezeigt wird, aber weder das fahrzeugeigene Ladelimit, noch geladene Kilometer und Ladezustand angezeigt werden. Neben dem Namen des Fahrzeugs ist das durchgestrichene Cloud-Symbol.

Bitte Info, falls logs benötigt werden (und welche).

zu 4.:
[main  ] ERROR 2025/03/06 18:38:48 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: Post "https://prd.eu-ccapi.kia.com:8080/api/v1/user/language": net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)

winconlin, 2025-03-06:
> Sieht aus als wäre das hier eine komplett neue API-Antwort mit neuer Struktur.
> 
> > sehe ich in Home Assistent über die kia hyundai blulink
> 
> Welche wäre das?

Diese hier: 
https://github.com/Hyundai-Kia-Connect

closed 2025-03-12 by andig

winconlin, 2025-03-12:
Also completed sehe ich das jetzt nicht so...