evcc-io/evcc #19197: Forecast.Solar: horizon parameter not working correctly

jodabis, 2025-02-26:
### Describe the bug

When using Forecast.Solar with a horizon param, evcc builds the URL incorrectly by surrounding the horiozon with ''.

### Steps to reproduce

1. Create forecast-solar template
2. Enter horizon as documented
3. Forecast.Solar responds with Error 422 (Unprocessable Entity)

### Configuration details

```yaml
solar:
    - type: template
      template: forecast-solar
      horizon: '0,5,5,0'
      lat: xx.xx
      lon: yy.yy
      dec: zz
      az: aa
      kwp: b.b
```

### Log details

```text
evcc -c /etc/evcc.yaml --database /var/lib/evcc/evcc.db tariff solar -l trace
...
[solar-template-0] TRACE 2025/02/26 10:55:59 GET https://api.forecast.solar/estimate/xx.xx/yy.yy/zz/aa/b.b?time=utc&full=1&resolution=60&horizon='0,5,5,0'
[solar-template-0] TRACE 2025/02/26 10:56:00 {"result":"ERROR Invalid horizon value: '0 (comma separated list of numerics expected)","message":{"c ...
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.3

jodabis, 2025-02-26:
I don't know if a setup like this will actually work, though. However, you can add your three panels as a list in yaml:
`solar:`
`   - type: template #string1`
`     template: forecast-solar`
`     ...`
`   - type: template #string2`
`     template: forecast-solar`
`     ...`
`   - type: template #string3`
`     template: forecast-solar`
`     ...`

andig, 2025-02-27:
> horizon: '0,5,5,0'

Don't use quotes.

jodabis, 2025-02-28:
> > horizon: '0,5,5,0'
> 
> Don't use quotes.

As I said, it really does not matter how you add the parameter in your config. Single quotes are always added when the URL is built.

andig, 2025-02-28:
> As I said, it really does not matter how you add the parameter in your config.

Ah, didn't see this in your config. Closed in b4db9e8e0427e4c510be090a76cda5fa663d0fdd.

closed 2025-02-28 by andig