evcc-io/evcc #19198: Lastmanagement  with two loadpoints

hybe2022, 2025-02-26:
At my site I am allowed to use two charger with 22kW each, but load management/balancing needs to ensure that a maximum of 22kW is used from grid when both charger are used

I thought it is posssible to create cascading circuits to allow that in a way that the parent circuit has no real meter, but should inherit the sum of each meter from the charger below.

                 main (50A)
                       |
               charger (32A)
            /                        \
garage (32A)        carport (32A)

Would it be possible to allow a virtual circuit e.q. charger with the sum of the circuits/meter below? Is it possible to define max grid consumption for a circuit while with generated PV energy the consumption might be more? in that case max. 32A grid consumption for all charger while with PV energy both might consume more than 32A?

VolkerK62, 2025-02-26:
So the meter value for the charge-circuit should be grid-power minus home-power?

hybe2022, 2025-02-26:
The first point is that it is possible to create a hierarchy with a virtual meter that holds the sum of the meter below without having a real physical meter for that circuit. This is currently not allowed. evcc does not start if I configure the circuits as shown in the overview as it complains that charger has no meter.

hybe2022, 2025-02-26:
> So the meter value for the charge-circuit should be grid-power minus home-power?

I think this is already with the current implementation possible if I set charger to also 50A. 
The question is if there is also the possibility to reduce the load only for grid consumption for a sub circuit to a defined value, here 32A.

VolkerK62, 2025-02-26:
you can create a virtuell meter by using the known values
```
- name: charge_circuit
  type: custom
  power:
    source: http
    uri: http://localhost:7070/api/state
    jq: .result.grid.power - .result.homePower
```
So this meter will show the power, the chargers are using from grid.
But, it is only possible with power and not with currents.

hybe2022, 2025-02-26:
It is known that I am able to create a virtual meter in yaml configuration. Just moved from yaml to UI and thought UI/db configuration withou yaml is the future.
So this request was intended to improve evcc and to allow virtual circuits without any meter which would then also be possible without yaml workaround. 
With the virtual meter anyhow the issue with different max  values for grid and internal power consumption would not be targeted.

hybe2022, 2025-02-26:
A new issue now also occurred when adding LPC to hems section. Now I have to assign the load points to lpc circuit while I also need load balancing for the load points to achieve maximum grid consumption of those during parallel use.

It needs to be possible to assign circuits multiple times to load points