evcc-io/evcc #19208: Dyn. Stromtarif Tibber nicht mehr sichtbar in der Version 0.200.3

SmartVolker, 2025-02-26:
### Describe the bug

Ich habe bisher Tibber als dyn. Stromtarif in der evcc.yaml hinterlegt. Das hat funktioniert bis zur Version 0.200.2. Nach Update auf 0.200.3 ist es nicht mehr in der Vorhersage und Hausbatterie / Netzladen sichtbar.

### Steps to reproduce

1. Tibber in evcc.yaml konfiguriert in Version 0.200.2
2. Update auf 0.200.3
3. Nicht mehr sichtbar in Vorhersage, Hausbatterie/ Netzladen
...

### Configuration details

```yaml
Kann zur Verfügung gestellt werden
```

### Log details

```text
[main ] INFO 2025/02/26 15:47:53 evcc 0.200.3
[main ] INFO 2025/02/26 15:47:53 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/26 15:47:53 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/26 15:47:57 listening at :7070
[lp-1 ] DEBUG 2025/02/26 15:48:10 charge total import: 8769.420kWh
[site ] INFO 2025/02/26 15:48:12 site config:
[site ] INFO 2025/02/26 15:48:12 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/26 15:48:12 grid: power ✓ energy ✗ currents ✓
[site ] INFO 2025/02/26 15:48:12 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2025/02/26 15:48:12 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site ] INFO 2025/02/26 15:48:12 vehicles:
[site ] INFO 2025/02/26 15:48:12 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2025/02/26 15:48:12 loadpoint 1:
[lp-1 ] INFO 2025/02/26 15:48:12 mode: now
[lp-1 ] INFO 2025/02/26 15:48:12 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/02/26 15:48:12 meters: charge ✓
[lp-1 ] INFO 2025/02/26 15:48:12 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/02/26 15:48:12 set buffer soc: 50
[site ] DEBUG 2025/02/26 15:48:12 set buffer start soc: 0
[site ] DEBUG 2025/02/26 15:48:12 set priority soc: 50
[site ] DEBUG 2025/02/26 15:48:12 set battery discharge control: true
[site ] DEBUG 2025/02/26 15:48:12 set grid charge limit: 0.3
[lp-1 ] WARN 2025/02/26 15:48:12 ignoring deprecated phases: 3. please configure via UI
[lp-1 ] DEBUG 2025/02/26 15:48:12 set smart cost limit: 0.2
[lp-1 ] DEBUG 2025/02/26 15:48:12 phase timer inactive
[lp-1 ] DEBUG 2025/02/26 15:48:12 pv timer inactive
[site ] DEBUG 2025/02/26 15:48:13 ----
[lp-1 ] DEBUG 2025/02/26 15:48:13 charge power: 3W
[lp-1 ] DEBUG 2025/02/26 15:48:13 charge currents: [0.021 0 0]A
[site ] DEBUG 2025/02/26 15:48:13 grid power: -4W
[site ] DEBUG 2025/02/26 15:48:13 grid currents: [0 0 0]A
[site ] DEBUG 2025/02/26 15:48:13 pv 1 power: 0W
[main ] ERROR 2025/02/26 15:48:23 version check failed: Get "https://api.github.com/repos/evcc-io/evcc/releases/latest": context deadline exceeded (installed: 0.200.3)
[site ] DEBUG 2025/02/26 15:48:23 battery 1 power: 319W
[site ] DEBUG 2025/02/26 15:48:23 battery 1 soc: 6%
[site ] DEBUG 2025/02/26 15:48:23 site power: 415W
[lp-1 ] DEBUG 2025/02/26 15:48:23 charge total import: 8769.420kWh
[lp-1 ] DEBUG 2025/02/26 15:48:24 charger status: A
[lp-1 ] INFO 2025/02/26 15:48:24 car disconnected
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.3

andig, 2025-02-26:
Was sagt

    evcc tariff

gramss, 2025-02-26:
Hi,

bei mir geht in der Vorhersage der Tibber Preis noch auf der v.0.200.3. Tibber Pulse ist ebenso noch aktiv.

VG

VolkerK62, 2025-02-26:
https://github.com/evcc-io/evcc/discussions/19162#discussioncomment-12313281

in der 0.200.3 gibt es wohl ein Problem mit tariffs in der yaml.
Wenn im UI konfiguriert, dann geht es.

SmartVolker, 2025-02-26:
> Was sagt
> 
> ```
> evcc tariff
> ```

[main  ] INFO 2025/02/26 17:42:00 evcc 0.200.3
[main  ] INFO 2025/02/26 17:42:00 using config file: /etc/evcc.yaml
[db    ] INFO 2025/02/26 17:42:00 using sqlite database: /var/lib/evcc/evcc.db
solar:
From    To    Yield (W)

co2:
From                   To                     Price/CostFootprint (gCO2/kWh)
2025-02-26 15:00:00    2025-02-26 16:00:00    289.000
2025-02-26 16:00:00    2025-02-26 17:00:00    113.000
2025-02-26 17:00:00    2025-02-26 18:00:00    96.000
2025-02-26 18:00:00    2025-02-26 19:00:00    106.000
2025-02-26 19:00:00    2025-02-26 20:00:00    248.000
2025-02-26 20:00:00    2025-02-26 21:00:00    262.000
2025-02-26 21:00:00    2025-02-26 22:00:00    238.000
2025-02-26 22:00:00    2025-02-26 23:00:00    238.000
2025-02-26 23:00:00    2025-02-27 00:00:00    238.000
2025-02-27 00:00:00    2025-02-27 01:00:00    238.000
2025-02-27 01:00:00    2025-02-27 02:00:00    244.000
2025-02-27 02:00:00    2025-02-27 03:00:00    265.000
2025-02-27 03:00:00    2025-02-27 04:00:00    272.000
2025-02-27 04:00:00    2025-02-27 05:00:00    265.000
2025-02-27 05:00:00    2025-02-27 06:00:00    265.000
2025-02-27 06:00:00    2025-02-27 07:00:00    262.000
2025-02-27 07:00:00    2025-02-27 08:00:00    248.000
2025-02-27 08:00:00    2025-02-27 09:00:00    193.000
2025-02-27 09:00:00    2025-02-27 10:00:00    148.000
2025-02-27 10:00:00    2025-02-27 11:00:00    130.000
2025-02-27 11:00:00    2025-02-27 12:00:00    110.000
2025-02-27 12:00:00    2025-02-27 13:00:00    103.000
2025-02-27 13:00:00    2025-02-27 14:00:00    103.000
2025-02-27 14:00:00    2025-02-27 15:00:00    103.000
2025-02-27 15:00:00    2025-02-27 16:00:00    103.000
2025-02-27 16:00:00    2025-02-27 17:00:00    93.000
2025-02-27 17:00:00    2025-02-27 18:00:00    79.000
2025-02-27 18:00:00    2025-02-27 19:00:00    168.000
2025-02-27 19:00:00    2025-02-27 20:00:00    248.000
2025-02-27 20:00:00    2025-02-27 21:00:00    220.000
2025-02-27 21:00:00    2025-02-27 22:00:00    189.000
2025-02-27 22:00:00    2025-02-27 23:00:00    175.000
2025-02-27 23:00:00    2025-02-28 00:00:00    179.000
2025-02-28 00:00:00    2025-02-28 01:00:00    210.000
2025-02-28 01:00:00    2025-02-28 02:00:00    217.000
2025-02-28 02:00:00    2025-02-28 03:00:00    217.000
2025-02-28 03:00:00    2025-02-28 04:00:00    220.000
2025-02-28 04:00:00    2025-02-28 05:00:00    217.000
2025-02-28 05:00:00    2025-02-28 06:00:00    224.000
2025-02-28 06:00:00    2025-02-28 07:00:00    231.000
2025-02-28 07:00:00    2025-02-28 08:00:00    248.000
2025-02-28 08:00:00    2025-02-28 09:00:00    196.000
2025-02-28 09:00:00    2025-02-28 10:00:00    193.000
2025-02-28 10:00:00    2025-02-28 11:00:00    151.000
2025-02-28 11:00:00    2025-02-28 12:00:00    124.000
2025-02-28 12:00:00    2025-02-28 13:00:00    110.000
2025-02-28 13:00:00    2025-02-28 14:00:00    99.000
2025-02-28 14:00:00    2025-02-28 15:00:00    103.000
2025-02-28 15:00:00    2025-02-28 16:00:00    61.000
2025-02-28 16:00:00    2025-02-28 17:00:00    58.000
2025-02-28 17:00:00    2025-02-28 18:00:00    72.000
2025-02-28 18:00:00    2025-02-28 19:00:00    103.000
2025-02-28 19:00:00    2025-02-28 20:00:00    279.000
2025-02-28 20:00:00    2025-02-28 21:00:00    203.000
2025-02-28 21:00:00    2025-02-28 22:00:00    189.000
2025-02-28 22:00:00    2025-02-28 23:00:00    186.000
2025-02-28 23:00:00    2025-03-01 00:00:00    186.000
2025-03-01 00:00:00    2025-03-01 01:00:00    210.000
2025-03-01 01:00:00    2025-03-01 02:00:00    210.000
2025-03-01 02:00:00    2025-03-01 03:00:00    217.000
2025-03-01 03:00:00    2025-03-01 04:00:00    213.000
2025-03-01 04:00:00    2025-03-01 05:00:00    210.000
2025-03-01 05:00:00    2025-03-01 06:00:00    262.000
2025-03-01 06:00:00    2025-03-01 07:00:00    238.000
2025-03-01 07:00:00    2025-03-01 08:00:00    224.000
2025-03-01 08:00:00    2025-03-01 09:00:00    182.000
2025-03-01 09:00:00    2025-03-01 10:00:00    165.000
2025-03-01 10:00:00    2025-03-01 11:00:00    127.000
2025-03-01 11:00:00    2025-03-01 12:00:00    117.000
2025-03-01 12:00:00    2025-03-01 13:00:00    93.000
2025-03-01 13:00:00    2025-03-01 14:00:00    103.000
2025-03-01 14:00:00    2025-03-01 15:00:00    93.000
2025-03-01 15:00:00    2025-03-01 16:00:00    72.000
2025-03-01 16:00:00    2025-03-01 17:00:00    65.000
2025-03-01 17:00:00    2025-03-01 18:00:00    55.000
2025-03-01 18:00:00    2025-03-01 19:00:00    79.000
2025-03-01 19:00:00    2025-03-01 20:00:00    279.000
2025-03-01 20:00:00    2025-03-01 21:00:00    206.000
2025-03-01 21:00:00    2025-03-01 22:00:00    196.000
2025-03-01 22:00:00    2025-03-01 23:00:00    206.000
2025-03-01 23:00:00    2025-03-02 00:00:00    206.000
2025-03-02 00:00:00    2025-03-02 01:00:00    162.000
2025-03-02 01:00:00    2025-03-02 02:00:00    165.000
2025-03-02 02:00:00    2025-03-02 03:00:00    151.000
2025-03-02 03:00:00    2025-03-02 04:00:00    151.000
2025-03-02 04:00:00    2025-03-02 05:00:00    144.000
2025-03-02 05:00:00    2025-03-02 06:00:00    227.000
2025-03-02 06:00:00    2025-03-02 07:00:00    182.000
2025-03-02 07:00:00    2025-03-02 08:00:00    200.000
2025-03-02 08:00:00    2025-03-02 09:00:00    134.000
2025-03-02 09:00:00    2025-03-02 10:00:00    86.000
2025-03-02 10:00:00    2025-03-02 11:00:00    55.000
2025-03-02 11:00:00    2025-03-02 12:00:00    55.000
2025-03-02 12:00:00    2025-03-02 13:00:00    55.000
2025-03-02 13:00:00    2025-03-02 14:00:00    55.000
2025-03-02 14:00:00    2025-03-02 15:00:00    55.000
2025-03-02 15:00:00    2025-03-02 16:00:00    55.000
2025-03-02 16:00:00    2025-03-02 17:00:00    55.000
2025-03-02 17:00:00    2025-03-02 18:00:00    55.000
2025-03-02 18:00:00    2025-03-02 19:00:00    55.000
2025-03-02 19:00:00    2025-03-02 20:00:00    162.000
2025-03-02 20:00:00    2025-03-02 21:00:00    162.000
2025-03-02 21:00:00    2025-03-02 22:00:00    182.000
2025-03-02 22:00:00    2025-03-02 23:00:00    175.000
2025-03-02 23:00:00    2025-03-03 00:00:00    175.000

SmartVolker, 2025-02-26:
> [#19162 (reply in thread)](https://github.com/evcc-io/evcc/discussions/19162#discussioncomment-12313281)
> 
> in der 0.200.3 gibt es wohl ein Problem mit tariffs in der yaml. Wenn im UI konfiguriert, dann geht es.

Danke für den Hinweis zum Workaround. Habe ich durchgeführt. Damit hat es funktioniert und nun sehe ich meinen Tibber dyn. Stromtariff wieder. 

sudo evcc migrate
[main  ] INFO 2025/02/26 17:47:26 evcc 0.200.3
[main  ] INFO 2025/02/26 17:47:26 using config file: /etc/evcc.yaml
[db    ] INFO 2025/02/26 17:47:26 using sqlite database: /var/lib/evcc/evcc.db
[main  ] INFO 2025/02/26 17:47:26 migrating:
[main  ] INFO 2025/02/26 17:47:26 - global settings
[main  ] INFO 2025/02/26 17:47:26 - network
[main  ] INFO 2025/02/26 17:47:26 - mqtt
[main  ] INFO 2025/02/26 17:47:26 - influx
[main  ] INFO 2025/02/26 17:47:26 - hems
[main  ] INFO 2025/02/26 17:47:26 - eebus
[main  ] INFO 2025/02/26 17:47:26 - modbusproxy
[main  ] INFO 2025/02/26 17:47:26 - messaging
[main  ] INFO 2025/02/26 17:47:26 - tariffs
[main  ] INFO 2025/02/26 17:47:26 - circuits
[main  ] INFO 2025/02/26 17:47:26 - device configs

evcc tariff
[main  ] INFO 2025/02/26 17:48:44 evcc 0.200.3
[main  ] INFO 2025/02/26 17:48:44 using config file: /etc/evcc.yaml
[db    ] INFO 2025/02/26 17:48:44 using sqlite database: /var/lib/evcc/evcc.db
grid:
From                   To                     Price/Cost (EUR/kWh)
2025-02-26 00:00:00    2025-02-26 01:00:00    0.355
2025-02-26 01:00:00    2025-02-26 02:00:00    0.351
2025-02-26 02:00:00    2025-02-26 03:00:00    0.348
2025-02-26 03:00:00    2025-02-26 04:00:00    0.347
2025-02-26 04:00:00    2025-02-26 05:00:00    0.348
2025-02-26 05:00:00    2025-02-26 06:00:00    0.348
2025-02-26 06:00:00    2025-02-26 07:00:00    0.367
2025-02-26 07:00:00    2025-02-26 08:00:00    0.399
2025-02-26 08:00:00    2025-02-26 09:00:00    0.417
2025-02-26 09:00:00    2025-02-26 10:00:00    0.400
2025-02-26 10:00:00    2025-02-26 11:00:00    0.377
2025-02-26 11:00:00    2025-02-26 12:00:00    0.364
2025-02-26 12:00:00    2025-02-26 13:00:00    0.352
2025-02-26 13:00:00    2025-02-26 14:00:00    0.352
2025-02-26 14:00:00    2025-02-26 15:00:00    0.358
2025-02-26 15:00:00    2025-02-26 16:00:00    0.367
2025-02-26 16:00:00    2025-02-26 17:00:00    0.387
2025-02-26 17:00:00    2025-02-26 18:00:00    0.405
2025-02-26 18:00:00    2025-02-26 19:00:00    0.414
2025-02-26 19:00:00    2025-02-26 20:00:00    0.403
2025-02-26 20:00:00    2025-02-26 21:00:00    0.384
2025-02-26 21:00:00    2025-02-26 22:00:00    0.350
2025-02-26 22:00:00    2025-02-26 23:00:00    0.353
2025-02-26 23:00:00    2025-02-27 00:00:00    0.343
2025-02-27 00:00:00    2025-02-27 01:00:00    0.330
2025-02-27 01:00:00    2025-02-27 02:00:00    0.333
2025-02-27 02:00:00    2025-02-27 03:00:00    0.329
2025-02-27 03:00:00    2025-02-27 04:00:00    0.328
2025-02-27 04:00:00    2025-02-27 05:00:00    0.329
2025-02-27 05:00:00    2025-02-27 06:00:00    0.335
2025-02-27 06:00:00    2025-02-27 07:00:00    0.346
2025-02-27 07:00:00    2025-02-27 08:00:00    0.390
2025-02-27 08:00:00    2025-02-27 09:00:00    0.398
2025-02-27 09:00:00    2025-02-27 10:00:00    0.379
2025-02-27 10:00:00    2025-02-27 11:00:00    0.364
2025-02-27 11:00:00    2025-02-27 12:00:00    0.340
2025-02-27 12:00:00    2025-02-27 13:00:00    0.329
2025-02-27 13:00:00    2025-02-27 14:00:00    0.339
2025-02-27 14:00:00    2025-02-27 15:00:00    0.340
2025-02-27 15:00:00    2025-02-27 16:00:00    0.355
2025-02-27 16:00:00    2025-02-27 17:00:00    0.379
[... 156 lines cut ...]
2025-03-04 02:00:00    2025-03-04 03:00:00    0.133
2025-03-04 03:00:00    2025-03-04 04:00:00    0.133
2025-03-04 04:00:00    2025-03-04 05:00:00    0.133
2025-03-04 05:00:00    2025-03-04 06:00:00    0.133
2025-03-04 06:00:00    2025-03-04 07:00:00    0.133
2025-03-04 07:00:00    2025-03-04 08:00:00    0.133
2025-03-04 08:00:00    2025-03-04 09:00:00    0.133
2025-03-04 09:00:00    2025-03-04 10:00:00    0.133
2025-03-04 10:00:00    2025-03-04 11:00:00    0.133
2025-03-04 11:00:00    2025-03-04 12:00:00    0.133
2025-03-04 12:00:00    2025-03-04 13:00:00    0.133
2025-03-04 13:00:00    2025-03-04 14:00:00    0.133
2025-03-04 14:00:00    2025-03-04 15:00:00    0.133
2025-03-04 15:00:00    2025-03-04 16:00:00    0.133
2025-03-04 16:00:00    2025-03-04 17:00:00    0.133
2025-03-04 17:00:00    2025-03-04 18:00:00    0.133
2025-03-04 18:00:00    2025-03-04 19:00:00    0.133
2025-03-04 19:00:00    2025-03-04 20:00:00    0.133
2025-03-04 20:00:00    2025-03-04 21:00:00    0.133
2025-03-04 21:00:00    2025-03-04 22:00:00    0.133
2025-03-04 22:00:00    2025-03-04 23:00:00    0.133
2025-03-04 23:00:00    2025-03-05 00:00:00    0.133

solar:
From    To    Yield (W)

andig, 2025-02-26:
> in der 0.200.3 gibt es wohl ein Problem mit tariffs in der yaml.

Ich finde keins- gibts das im Nightly noch oder können wir hier zu machen?

VolkerK62, 2025-02-26:
Ich hatte das lediglich aus der Diskussion #19162 
Ob es im Nightly gelöst ist, kann ich nicht beurteilen (hab alles im UI).

biddns, 2025-02-26:
Hallo, habe das gleiche Problem, keine Daten mehr von Tibber sichtbar seit 0.200.3
Ist in der evcc.yaml so eingetragen, wie es in der Anleitung steht.

andig, 2025-02-26:
https://github.com/evcc-io/evcc/issues/19208#issuecomment-2685751390

bodomenke, 2025-02-28:
Habe das gleiche Problem seit 0.200.3, allerdings nur bei den Ladepunkten:
Bei Click auf Plan wird unmittelbar folgender Fehler ins Log geschrieben (reproduzierbar):
```
Request failed with status code 400.
GET https://evcc.meko.haus/api/loadpoints/2/plan/static/preview/soc/90/2025-03-01T06:30:00.000Z
```
Die Anzeige des Tibber-Tarifs bei der Hausbatterie funktioniert dagegen weiterhin.

andig, 2025-02-28:
Das ist https://github.com/evcc-io/evcc/pull/19245

bodomenke, 2025-02-28:
Bestens. Danke. Dann habe ich mal gleich meinen ersten Nightly für evcc a den Start gebracht und habe nun auch Multisite-Forecast Solar… sehr fein. 👍

naltatis, 2025-02-28:
@SmartVolker hattest du eine gemischte Konfiguration der Tarife? Also einige Tarife in evcc.yaml und einige in der UI? Hier haben wir in der Tat das Verhalten geändert. Vorher wurden yaml und db Konfigurationen für Tarife gemerged. Nun wird der Tarifblock in der evcc.yaml ignoriert, wenn es auch eine Tarifkonfiguration im UI gibt.

https://github.com/evcc-io/evcc/pull/19184

naltatis, 2025-02-28:
> keinen Preis in der Home Ansicht

Du meinst in dem Energieflussbereich oben? Wenn du nur feste Preise hast wird dort der CO2 Wert (sofern vorhanden) verwendet. Am Ladepunkt kannst du das ja jeweils einstellen.

SmartVolker, 2025-03-01:
> [@SmartVolker](https://github.com/SmartVolker) hattest du eine gemischte Konfiguration der Tarife? Also einige Tarife in evcc.yaml und einige in der UI? Hier haben wir in der Tat das Verhalten geändert. Vorher wurden yaml und db Konfigurationen für Tarife gemerged. Nun wird der Tarifblock in der evcc.yaml ignoriert, wenn es auch eine Tarifkonfiguration im UI gibt.
> 
> Siehe: [#19184](https://github.com/evcc-io/evcc/pull/19184)

Ja, gestartet bin ich mit Tibber config in YAML Datei. Danach habe ich Grünstromindex über das Config UI ergänzt. Das funktionierte beides parallel einwandfrei bis zum Upgrade auf 0.200.3. Dann wurde Tibber nicht mehr angezeigt nur noch der Grünstromindex. Daher wäre der Hinweis gut gewesen, dass man bei Upgrade auf 0.200.3 vorher ein Migrate durchführt. Das hat ja dann auch geholfen, sodass Tibber Tarif wieder sichtbar war. Leider wurde Grünstromindex rausgelöst und ich musste es neu anlegen nach der Migration auf die Datenbank.

closed 2025-03-13 by andig