evcc-io/evcc #19212: Sofarsolar im Passive Modus hat falsche Werte

maxirass, 2025-02-26:
### Describe the bug

Ich benutze evcc in Home Assistant.
Mein rs458 zu ETH Adapter ist über das Modbus Proxy Addon in Home Assistant angebunden.
Der Wechselrichter ist ein SofarSolar HYD20000-3PH.
Die Werte waren bisher immer richtig. Jetzt habe ich heute das erste Mal mit dem Passive Mode rum gespielt, welchen ich über die Sofarsolar-Modbus-Integration in HA steuere. 
Als ich den Passive Mode aktiviert habe wurden die Werte in evcc plötzlich falsch angezeigt.

Korrekter Batteryoutput: 0,8 kW
Angezeigt in evcc: 80W

Als ich den Modus zurück auf Self-Use (so wie davor die ganze Zeit) gestellt habe, haben sich die Werte auch nicht mehr normalisiert. 
Neustart von HA und der evcc-Integration haben nichts bewirkt.

Woran kann das liegen?

### Steps to reproduce

1. Passive-Mode in SofarSolar aktivieren

### Configuration details

```yaml
meters:
- name: grid
  type: template
  template: sofarsolar-g3
  usage: grid
  modbus: tcpip
  id: 1  
  host: 192.168.1.205
  port: 502
- name: pv
  type: template
  template: sofarsolar-g3
  usage: pv
  modbus: tcpip
  id: 1  
  host: 192.168.1.205
  port: 502
- name: battery
  type: template
  template: sofarsolar-g3
  usage: battery
  modbus: tcpip
  id: 1  
  host: 192.168.1.205 # HA IP
  port: 502

chargers:
  - type: template
    template: go-e-v3
    host: 192.168.1.230
    name: wallbox4

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.37 # EUR/kWh
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  disable:
     delay: 1m
     threshold: 0

site:
  title: Mein Zuhause
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery
  residualPower: 100
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/02/26 18:58:52 evcc 0.200.3
[main  ] INFO 2025/02/26 18:58:52 using config file: /config/evcc.yaml
[db    ] INFO 2025/02/26 18:58:52 using sqlite database: /data/evcc.db
[main  ] INFO 2025/02/26 18:58:52 listening at :7070
[site  ] INFO 2025/02/26 18:58:52 site config:
[site  ] INFO 2025/02/26 18:58:52   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/02/26 18:58:52     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/02/26 18:58:52     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/02/26 18:58:52     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[lp-1  ] INFO 2025/02/26 18:58:52 loadpoint 1:
[lp-1  ] INFO 2025/02/26 18:58:52   mode:        pv
[lp-1  ] INFO 2025/02/26 18:58:52   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/02/26 18:58:52   meters:      charge ✓
[lp-1  ] INFO 2025/02/26 18:58:52     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] WARN 2025/02/26 18:58:52 ignoring deprecated minCurrent: 6.000000. please configure via UI
[lp-1  ] WARN 2025/02/26 18:58:52 ignoring deprecated maxCurrent: 32.000000. please configure via UI
[site  ] WARN 2025/02/26 18:58:52 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2025/02/26 18:58:58 car disconnected
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.3

maxirass, 2025-02-26:
Das ist die Batteriekurve aus evcc:

[image]

und das die gleiche von SofarSolar ausgelesen:

[image]

StefanSchoof, 2025-02-26:
> I have verified that the issue is reproducible with the latest nightly build

Heißt das https://github.com/evcc-io/evcc/pull/19158 funktioniert nicht oder hast du es nicht im nightly probiert?

StefanSchoof, 2025-02-26:
Ja, das geht mittlerweile. Aber da ich kein HA nutze, weiß ich nicht wie.

closed 2025-02-26 by maxirass