evcc-io/evcc #19213: Planner rechnet mit anderer Charge Power als der Loadpoint

All-Ex-U, 2025-02-26:
### Describe the bug

Meine wallbox (schneider-evlink) ist in der Firmware für 1 Phase konfiguriert (da ich meist 1-phasig lade). Wenn ich elektrisch 3 Phasen einschalte, merkt die Firmware nicht, dass 3 Phasen aktiv sind und meldet 1 Phase mit 10,6 kW charge power an evcc:

[lp-1 ] DEBUG 2025/02/26 19:57:42 **charge power: 10664W**
[lp-1 ] DEBUG 2025/02/26 19:57:42 charge currents: [15.6 0 0]A
[lp-1 ] DEBUG 2025/02/26 19:57:44 charge voltages: [228 0 0]V
[lp-1 ] DEBUG 2025/02/26 19:57:44 **detected connected phases: 1p**
[lp-1 ] DEBUG 2025/02/26 19:57:44 detected active phases: 1p

Der Planner rechnet jetzt mit 3,6 kW, obwohl Charge Power = 10,6 kW. Dadurch dauert die Ladezeit in der Planung zu lange (hier 8 Std):

[lp-1 ] DEBUG 2025/02/26 19:57:44 plan: charge **8h9m47s** between 2025-02-26 20:00:00 +0100 CET until 2025-02-27 08:00:00 +0100 CET (power: **3680W**, avg cost: 216.057)

Am Loadpoint mit wird korrekt gerechnet:

    "loadpoints": [
      {
        "batteryBoost": false,
        "chargeCurrent": 16,
        "chargeCurrents": [15.5909996032715, 0, 0],
        "chargeDuration": 2152,
        "chargePower": 10611.3,
        "chargeRemainingDuration": 10858,
        "chargeRemainingEnergy": 32004.333,
        "chargeTotalImport": 4156.617,
        "chargeVoltages": [226.399993896484, 0, 0],

32 kWh zu laden bei 10,6 kW = 3 Std. = 10.800 Sek. => passt!

Wäre es möglich, dass der Planner ebenfalls "chargePower" als Berechnungsgrundlage nimmt? Dann würden auch bei Wallboxes, welche die falsche Anzahl aktiver Phasen melden, korrekte Pläne herauskommen...

### Steps to reproduce

1. siehe oben
2.
3.
...

### Configuration details

```yaml
siehe oben
```

### Log details

```text
siehe oben
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

closed 2025-02-26 by andig

All-Ex-U, 2025-02-26:
Alles klar, dank dir für die schnelle Antwort!

premultiply, 2025-02-27:
Die Wallbox muss jederzeit die netzseitig anliegenden Phasenspannungen messen und zurückliefern damit die Erkennung der Phasenanzahl und der daraus abgeleiteten max. Leistung automatisch erfolgt.