evcc-io/evcc #19216: SMA HEMS and lpc cannot be used/configured together

hybe2022, 2025-02-26:
### Describe the bug

I have a SMA Sunny Home Manager configured as HEMS to get grid meter data and to provide charger consumption via SEMP to Sunny portal. Now I would like to configure LPC to control consumption in HEMS section which is not possible.

### Steps to reproduce

1. SMA home manager configured in HEMS section
2. Try to add relay in HEMS section for power consumption limitation
3. Error due to second type is thrown
...

### Configuration details

```yaml
UI HEMS configuration:

type: sma
allowcontrol: false
deviceid: '000000000001'
vendorid: '12345678'
type: relay
maxPower: 14800 # single SteuVE
limit:
    source: http
    uri: http://localhost:8123/api/states/switch.waermepumpe
    headers:
    - content-type: application/json
    - authorization: Bearer 1234589
    jq: if .state == "Ein" then "true" else "false" end
    timeout: 3s #
```

### Log details

```text
yaml: unmarshal errors: line 5: mapping key "type" already defined at line 1
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

hybe2022, 2025-03-11:
That means a HEMS cannot be used in parallel to a 14a relay box and this will also not be planned?

hybe2022, 2025-04-06:
Why shall it no be possible to connect evcc with SMA home manager and to use/connect evcc with  a relay of the control box to control SteuVE?
This is a real limitation and I joined evcc as contributor as I thought that should be possible. Now I see my SMA system cannot be used with evcc and evcc is controlling the SteuVE as the SMA HEMS and relay cannot be configured together. 

Is there a real reasonb why this is not planned, while I see it more as an bug, as there is no reason not be be able using both together...

andig, 2025-04-06:
I agree. It would be nice to have both. Happy to take a PR.

andig, 2025-04-06:
There is no PR. If you need this it needs be developed.

hybe2022, 2025-04-06:
Is it documented that evcc is only capable to either connect to SMA system or to support lpc and the HEMS section is only single entry or maybe misused for lpc?

andig, 2025-04-06:
https://github.com/evcc-io/evcc/issues/19216#issuecomment-2781532738

hybe2022, 2025-04-06:
> [#19216 (comment)](https://github.com/evcc-io/evcc/issues/19216#issuecomment-2781532738)

Jetzt habe ich das neue Ticket erstellt, aber es wurde geschlossen.

premultiply, 2025-04-06:
Code schreibt sich leider nicht dadurch, dass man Tickets erstellt.

Jetzt braucht es nur noch eine konkreten Lösungsvorschlag - und der fällt nicht einfach vom Himmel.

andig, 2025-04-12:
Die Lösung könnte sein, ein HEMS und zusätzlich SMA zu erlauben, SMA also in ein eigenes Objekt zu verschieben. Es wirkt auch anders als die später dazu gekommen HEMS. Dann bräuchte es einen zusätzlichen Top-level Eintrag "sma" oder eine andere Bezeichnung dafür.

Alternativ: mehrere HEMS (=Liste), die sich aber ausschließen, z.B. kann nur ein HEMS auf die Circuits einwirken.

@naltatis wie wäre es mit der Alternative?

hybe2022, 2025-04-12:
Die einfachste Lösung wäre HEMS und SMA aufzutrennen, wobei ich nicht weiß, ob es die sinnvollste Lösung ist. 

Besser wäre sicher aus HEMS eine Liste zu machen. 
In der Zwischenzeit gibt es verschiedene HEMS mit unterschiedlichen Features und so kann es in Zukunft auch dazu kommen, dass andere User 2 oder sogar 3 HEMS einsetzen müssen um ihre Geräte bedienen zu können. Somit wäre die Alternative mit der Liste die, welche auch dies automatisch für die Zukunft abdecken würde die flexibelste und würde es erlauben mit evcc die Systeme zentral zu verbinden. Das wäre ein Pluspunkt von evcc...

hybe2022, 2025-06-01:
Die Steuerbox ist mittlerweile installiert, aber leider kann ich evcc nicht für das Loadmanagement damit verbinden, weil mein SMA Home Manager bereits über HEMS mit evcc verbunden ist. 
Habt ihr noch auf dem Schirm für LPC Konfiguration aus HEMS auszulagern oder zwei HEMS möglich zu machen?

andig, 2025-06-01:
@naltatis entweder Liste oder SMA separieren- Beides müssten wir im UI unterstützen.

hybe2022, 2025-06-01:
Genau, das waren die beiden Option, die ich bereits erwähnt hatte. 
Das Thema könnte auch sehr gut unter folgendes gehängt werden...

https://github.com/evcc-io/evcc/issues/14261

closed 2025-06-01 by andig

hybe2022, 2025-06-01:
Da dieses Thema nun als Duplikat von https://github.com/evcc-io/evcc/issues/14261 gesetzt wurde, wäre es auch gut, wenn es in der Liste der offenen Punkte dort auch auftauchen würde. 
Könntet ihr das bitte dort in der Liste des ersten Beitrags hinzufügen? 

Danke