evcc-io/evcc #19217: LG Therma V R290 Monobloc Heatpump - Modbus control

isarrider, 2025-02-26:
The LG Therma V R290 Monobloc is one of the more silent, efficient
(and in the eyes of myself) goodlooking heatpumps.

It can be controlled via some Modbus register, a good summary is here:
https://github.com/basti242/homeassistant_lg_therma_v_modbus/wiki/LG-Register-documentation

the official dokumentation is here (at the moment):
https://www.lg.com/de/support/produkt-hilfe/cs-HM093HFX.UB60/#manual-tab

I think support would be nice...

isarrider, 2025-02-26:
Linking this in here:
https://github.com/evcc-io/evcc/discussions/13651

andig, 2025-02-26:
Wanna try a PR now that we have https://github.com/evcc-io/evcc/pull/19008?

isarrider, 2025-02-26:
20 years ago I would have said yes, 10 years ago maybe...
Nowadays this will take me so much time that everybody will just be annoyed...

andig, 2025-02-26:
Hast du das Ding so dass du testen könntest?

isarrider, 2025-02-26:
Ist bestellt...

isarrider, 2025-03-07:
40009 macht ja nur das Trink-Wasser wärmer - sinnvoller wäre auch eine Erhöhung der Heiztemperatur...
Leistungsmessund können die neueren HP anscheinend, aber ich habe eh schon davor einen SDM gesetzt.
Danke aber erstmal...

premultiply, 2025-03-08:
> Es muss einfach Holding Register 9 (keine Ahnung warum das als 40009 bezeichnet wird, das scheint falsch zu sein)

Das ist so eine der vielen sinnlosen Dinge im Modbus-Umfeld die sich mal irgendwer in einer ganz finsteren Stunde ersonnen hat. Die Unüberlegung war den verschiedenen Registerzugriffsmodi verschiedene Register-Nummernbereiche zuzuordnen. Am liebsten noch mit +1-Offset und Swapped-Word-Encoding. 🤯

Wenn du einen sehr guten Eindruck haben willst wie man Modbus möglichst mit all diesen tollen Sachen völlig unbrauchbar implementiert solltest du dir unbedingt mal iobroker ansehen... 👹

andig, 2025-03-08:
Besonders lustig wird das dann wenn du in der Doku func code 3 mit 40000er Adressen siehst. Völlig sinnlos…

premultiply, 2025-03-08:
Frei nach dem Motto: Offsets kann man doch auch auf Functioncodes anwenden 🤠

tafx, 2025-03-08:
Direct communication of EVCC with LG Therma HP would be nice. The same registers are valid also for older LG models. Nevertheles I have solved the LG control by automation in HA and custom charger in EVCC. When EVCC switchs LG charger into enable state, HA automation changes LG heat pump parameter "energy state" to 5 (by modbus). You can simply define your elevated target temperatures for heating and DHW at that energy state 5 as you wish. When there are not enough PV surplus energy state is set back to 2 or 1. It is working nicely. But direct communication EVCC - HP would be more robust solution. So if anybody interested, I could test it on our LG Therma monobloc HP.

isarrider, 2025-03-14:
> Direct communication of EVCC with LG Therma HP would be nice. The same registers are valid also for older LG models. Nevertheles I have solved the LG control by automation in HA and custom charger [...]

You are using that?
https://github.com/basti242/homeassistant_lg_therma_v_modbus

maximilianadolf, 2025-08-11:
Ich habe die LG Therma V R290 über eine benutzerdefinierte Heizung (sg-ready, komplett) eingebunden. 
Modbus läuft über einen Waveshare RS485 TO ETH (B) und die LG ist im Menü unter Konnektivität > Energiezustand > Signaltyp > Modbus eingestellt.

> 

> See https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md#device-templates for testing templates.

Das mit den Templates habe ich noch nicht genau gefunden, wo ich das Template einfügen könnte. Ich habe EVCC aber auch als Addon in Home Assistant installiert.

```
## required attributes [type: sgready]

setmode: # set operation mode (1: reduced, 2: normal, 3 boost)
  source: map
  values: #Mapping zu Energiezuständen
    1: 7 #Energiesparen
    2: 2 #Normalbetrieb
    3: 6 #Boost
  set:
    source: modbus
    uri: 192.168.178.83:502
    id: 33
    rtu: false
    register:
      address: 9
      type: writesingle
      encoding: uint16

## optional attributes (read-only)

getmode: # operation mode (1: reduced, 2: normal, 3 boost)
  source: modbus
  id: 33
  uri: 192.168.178.83:502
  rtu: false
  register:
    address: 11
    type: input
    decode: int16
temp: # current temperature drinking water (°C)
  source: modbus
  id: 33
  uri: 192.168.178.83:502
  rtu: false
  scale: 0.1
  register:
    address: 5
    type: input
    decode: uint16
limittemp: # temperature limit (°C) no value known for LG Therma V
  source: const
  value: 65
power: # charge power in W, LG internal values are too high, suggest to use external device
  source: modbus
  id: 1
  uri: 192.168.178.83:502
  rtu: false
  register:
    address: 52
    type: input
    decode: float32

## optional attributes (writeable)

#setmaxpower: # update the maximum charging power
#  source: js
#  script: console.log(maxpower);

```

maximilianadolf, 2025-08-21:
Ich habe es jetzt in einer Windowsinstallation ausprobiert und mit LG Therma V ausprobiert. Ich hatte erst noch einige Fehler drin, aber so schaut es aktuell gut aus.
Ich habe aktuell die Modi 1, 2, und 3 auf Energiezustände 7, 2, und 6 gelegt, da man hier die Anpassungen für SG Ready individuell anpassen kann. Ich habe jetzt aber noch nicht gefunden, wie ich dem User die Wahl geben kann ("tempsource" hab ich einem anderen Template gefunden und das klappt, aber kann ich auch individuelle Parameter wie z.B. "reduce_setting" definieren?).
Ich habe leider jetzt noch nicht gefunden, wie man den angepassten Code einbringen kann, aber ich versuche mich mal am Wochenende einzulesen...
```
template: lg-therma
products:
  - brand: LG
    description:
      generic: Therma V R290 Monobloc (SG Ready)
group: heating
requirements:
  description:
    de: |
      Modbus TCP Verbindung zur LG Therma V Wärmepumpe.
      Funktioniert über SG Ready Energiezustände für intelligente Steuerung.
    en: |
      Modbus TCP connection to LG Therma V heat pump.
      Uses SG Ready energy states for intelligent control.
params:
  - name: modbus
    choice: ["tcpip"]
    id: 33
  - name: tempsource
    choice: ["warmwater", "flow_temperature"]
render: |
  type: sgready
  setmode:
    source: map
    values:
      1: 7 # Energiesparen
      2: 2 # Normalbetrieb
      3: 6 # Boost
    set:
      source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 9
        type: writeholdings
        encoding: uint16
  getmode:
    source: map
    values:
      7: 1 # Energiesparen
      2: 2 # Normalbetrieb
      6: 3 # Boost
    get:
      source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 9
        type: holding
        encoding: uint16        
  temp:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    scale: 0.1
    register:
      address:  {{ if eq .tempsource "warmwater" -}} 5 {{ else }} 2 {{- end }} # 5 Trinkwasser, 2 Wassereinlasstemperatur
      type: input
      decode: uint16
  power:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      address: 35
      type: input
      decode: uint16
```

andig, 2025-08-22:
@maximilianadolf du kannst Deine Anpassungen hier im PR in der Codesicht mit + direkte rein tippen so dass ich sie nur noch mergen muss.

andig, 2025-08-22:
Nvm, ich hatte nicht richtig hingeschaut. PR für Dein Template ist erstellt.

andig, 2025-08-22:
> aber kann ich auch individuelle Parameter wie beispielsweise "reduce_setting" definieren?).

Du kannst dich da frei austoben und weitere Parameter definieren.

andig, 2025-08-26:
Du hast die Arbeit gemacht ;)

isarrider, 2025-08-26:
Topp - Danke...
Ein paar Screenshots?

maximilianadolf, 2025-08-26:
> Topp - Danke... Ein paar Screenshots bitte ;)

[image]

[image]

[image]

Du brauchst halt eine RS485 Verbindung zur LG. Ich habe es über einen Waveshare ETH-RS485 verbunden.

3rhan1999, 2025-10-04:
Heyy habe das Problem, dass bei meiner Wärmepumpe die Inneneinheit scheinbar keine Registerausgabe für Stromverbrauch hat Firmware 3.07.3a. Aus dem grund will er mir auch die Connecten nicht prüfen, er bringt eine Fehlermeldung bezüglich dem "Power", welches scheinbar die HN1639 HC NK0 nicht ausgibt. Wurde nach langer Recherce, von werk angeblich ausgeblendet. Denkt ihr, dass ich hier ein Shelly 3em integrieren kann und hier das power anderweitig vergeben könnte?

maximilianadolf, 2025-10-05:
Der Shelly würde dir erst im Nachgang etwas bringen, wenn die Grundkonfiguration funktioniert.
Ich weiß nicht, ob ich einen Heater auch ohne "Power" definieren kann (bzw. diesen auf "N/A" setzen kann) und dann auch nur mit einem externen Energiezähler betreiben kann.
@andig ist das möglich?

@3rhan1999 kannst du deine Firmware updaten? Ich habe es mit 3.0.7.6a am laufen, habe aber auch nicht die Hydrobox.
Hast du in der LG die Energieüberwachung eingeblendet und bekommst hier Werte?

andig, 2025-10-05:
> Ich weiß nicht

Einfach ausprobieren? Ja!

maximilianadolf, 2026-04-05:
Cloud Gateway ist eigentlich nur für den Service interessant. Für den Benutzer gibt's da nur rudimentär Möglichkeiten das Heißwasser einzustellen. 
Für die Anbindung an EVCC oder Home Assistent ist immer eine Modbus Verbindung notwendig.