evcc-io/evcc #19219: Sofar: Support multiple storage units

cschlipf, 2025-02-26:
> Würde auch für die Variante analog zu anderen Implementierungen via `storageunit` plädieren.
> Die passenden Summenbildungen macht evcc an der `site` automatisch. 

 _Originally posted by @premultiply in [#18933](https://github.com/evcc-io/evcc/issues/18933#issuecomment-2686187915)_

Similar like for Solax: https://github.com/evcc-io/evcc/blob/714438d0581c11fe14fb7496747250cd9f659398/templates/definition/meter/solax.yaml#L18

cschlipf, 2025-02-27:
@premultiply Muss Dich leider doch noch mit einer Frage nerven.

Für die aktuelle Leistung (Power_Batx) und Soc (SOC_Batx) gibt es pro storageunit Register - so weit so gut. Für power und soc haben wir also individuelle Register.

ABER nicht für Energy. HIer gibt es nur ein Bat_Discharge_Total.

Es wäre nun ja sicher falsch für jede Storage Unit nun die Total Energy zu nehmen, wenn diese dann zusammengerechnet wird. Im `solax.yaml` sehe ich, dass es hier ähnlich ist und da wurde die Energy einfach auskommentiert. Das mache ich hier dann auch, richtig?

Schroediman, 2025-02-27:
Hallo, das gleiche Thema mit der Auflösung in 100 W Schritten trifft neben der Batterie Ladung / Endladung auch das Register zum Einlesen der solaren Leistung (Power_PV_Total) zu. Da dieses Register verwendet wird, kann steigt die aktuelle PV-Leistung nur in 100 W Schritten. Die Register der einzelnen Strings sind allerdings in 10 W Schritten aufgelöst. Die könnte man ähnlich den Batterien einzeln aus den Registern lesen und dann addieren. Ich mache das zur Zeit, in dem ich den Wechselrichter über eine Customintegration auslese , in der die beiden Strings des HYD einzeln gelesen und dann summiert werden. Also so:

name: HYD_pv
    type: custom
    power:
      source: calc
      add:
      - source: modbus
        uri: 192.168.1.xyz:502
        id: 1
        register:
          address: 0x586 # Power PV 1
          type: holding
          decode: int16
        scale: 10
      - source: modbus
        uri: 192.168.1.xyz:502
        id: 1
        register:
          address: 0x589 # Power PV 2
          type: holding
          decode: int16
        scale: 10
  

MfG

Schroediman

Jungs, das Projekt ist toll.

cschlipf, 2025-02-28:
Oh, guter Punkt. Das wäre nun allerdings ein Breaking Change für alle. ABER ich habe da eine Idee. Wie bei der Batterie ein parameter 'pvinputstring' (offen für einen besseren Namen). Default ist 0 - dann werden die Total Registers genommen. Nimmt man 1 und/oder 2 nur diese Strings.

Damit würde man Backward-Kompatibel bleiben, aber man könnte die einzelnen Strings ansteuern mit einer höheren Genauigkeit.

Nachteil: Jemand könnte auf die Idee kommen "string" 0, mit 1 und 2 zu kombinieren - dann würde Total zu den String Leistungen addiert werden, was natürlich falsch ist.

@andig @premultiply - was haltet ihr davon? Dasselbe könnte man auch für die Batterie machen um einen Breaking Change zu vermeiden: storageunit=0 - Total Values.

andig, 2025-02-28:
> Damit würde man Backward-Kompatibel bleiben, aber man könnte die einzelnen Strings ansteuern mit einer höheren Genauigkeit.

Die Änderung gibts erst ein paar Tage, können wir problemlos einfach wieder zurück drehen als Fehlerbehebung. Keine zusätzliche Komplexität.

cschlipf, 2025-02-28:
> > Damit würde man Backward-Kompatibel bleiben, aber man könnte die einzelnen Strings ansteuern mit einer höheren Genauigkeit.
> 
> Die Änderung gibts erst ein paar Tage, können wir problemlos einfach wieder zurück drehen als Fehlerbehebung. Keine zusätzliche Komplexität. Gleiche Lösung wie bei anderen WR auch…

Halt stop: Wir reden jetzt von was anderem. Nicht mehr von der Batterie, sondern auch davon das bei den PV Eingängen so zu machen. Während bei der Batterie wir nun nach langer Zeit den ersten Fall haben, wo eine zweite Batterie ein Problem macht, weil wir davor nur die erste Batterie unterstützt haben, wäre das beim PV Eingang eine andere Story - hier hat fast jeder alle beide Strings angeschlossen und wäre betroffen. Daher die Idee mit 0 = Total. 1,2 = individuelle Strings.

andig, 2025-02-28:
Ahh, I see. Strings hab ich ja nicht viele, max 2? Die können wir ja addieren. Sie müssen natürlich an PV hängen, nicht ab Battery.

cschlipf, 2025-02-28:
Neuere Modell kommen mit 3 Strings. Aber zurück zur Frage:

Sollen wir das machen? Und wenn ja: Breaking Change oder kein Breaking Change mit stringid=0 zu den Total Registern?

andig, 2025-02-28:
Separater PR, einfach machen. Keine stringid neu erfinden, gibts bei anderen WR auch nicht.

closed 2025-02-28 by andig

Schroediman, 2025-02-28:
Hallo, ich misch mich eben mal ein; laut Protokoll werden bis zu 16 PV-Strings  und maximal 8 Batterietürme unterstützt. Also 8 BDUs mit bis zu 4 einzelnen Batteriemodulen. Das ist auch der Grund warum so viele , wie ich auch nur EINEN Batterieturm besitzen. Der kann von  5 bis 20 kWh ausgebaut sein kann. Sprich, ich hab zwar 2 Batterien (2 * 5kWh), die sind in einem Batterieturm zusammengefasst und damit hab ich für das Protokoll nur eine Batterie. Das gilt für die BTS5K-Batterie Serie von Sofar.

Danke

Schroediman

cschlipf, 2025-02-28:
@Schroediman Man kann allerdings nur 2 BTS Türme an den aktuellen Wechselrichtern anschließen. Aber ich habe vorher meinen Kontakt bei Sofar gefragt. Die 8 Batterien machen im Fall eine Amass GTX-3000 Speichers Sinn, da hier die einzelnen Module einzeln angesprochen werden. Auch hier wundert es mich, dass sich noch niemand früher gemeldet hat.

Wegen BTS: Deswegen habe ich extra den storageunit Parameter ausführlicher dokumentiert, damit der Unterschied zwischen Batteriemodul und Batterieturm klar wird.