evcc-io/evcc #19223: [tariff]: Nachträgliches Initialisieren, wenn beim Starten von evcc nicht verfügbar

An-Wi, 2025-02-27:
**Is your feature request related to a problem? Please describe.**
Da es heute die **Solar**prognose betrifft und gestern der **CO2**-Grünstromindex betroffen war, ist mir Folgendes aufgefallen:
Ist beim Starten von evcc ein [tariff] nicht verfügbar oder es tritt ein Fehler beim Initialisieren auf, dann wird dieser Abruf auch später nicht nachgeholt.
<pre>
[gsi   ] ERROR 2025/02/26 08:12:58 Get "https://api.corrently.io/v2.0/gsi/prediction?zip=***": Post "https://console.corrently.io/v2.0/auth/requestToken": context deadline exceeded (Client.Timeout exceeded while awaiting headers) (Client.Timeout exceeded while awaiting headers)
[main  ] ERROR 2025/02/26 08:12:58 creating tariff co2 failed: cannot create tariff type 'template': cannot create tariff type 'grÃ¼nstromindex': Get "https://api.corrently.io/v2.0/gsi/prediction?zip=07552": Post "https://console.corrently.io/v2.0/auth/requestToken": context deadline exceeded (Client.Timeout exceeded while awaiting headers) (Client.Timeout exceeded while awaiting headers) 
</pre>
<pre>
[tariff] ERROR 2025/02/27 08:27:00 unexpected status: 401 (Unauthorized)
[main  ] ERROR 2025/02/27 08:27:00 creating tariff solar failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 401 (Unauthorized) 
</pre>

**Describe the solution you'd like**
Ein stündliches Nachinitialisieren der fehlenden Tarife wäre in diesem Fall sehr hilfreich.
Auch gibt es momtenan keine Warnmeldung zu diesem Problem. Ein "gelbes" Ausrufezeichen oben in der Hauptansicht wäre vielleicht sinnvoll.

andig, 2025-02-27:
Aktuell keine Priorität, wird aber irgendwann kommen...

closed 2025-02-27 by andig

An-Wi, 2025-02-27:
Für die **Solar**prognose muss ich meine Aussage zurücknehmen, denn es wurde vorhin nach genau 1h erneut versucht:
<pre>
[tariff] ERROR 2025/02/27 09:27:00 unexpected status: 401 (Unauthorized) 
</pre>
Auch das Log von heute Nacht weißt auf eine regelmäßigen Abruf der Solar-Daten hin, denn ab etwa 03:56 Uhr habe ich die Fehlermeldung mit `unexpected status: 401 (Unauthorized)` im Log (exakt im Stundentakt).

Damit stellt sich die Frage, warum das bei **CO2** gestern nicht genauso funktioniert hat?
Liegt es vielleicht daran, dass diese Daten nur 1x täglich abgerufen werden?

An-Wi, 2025-02-27:
Letzte Ergänzung:
Die **Solar**prognose mit meinem API-Key ist wieder abrufbar (habe das Problem mit dem Support von forecast.solar klären können).
Auch die Fehlermeldung tauchte `10:27:00` im Log nicht mehr auf.

Allerdings ist das Verhalten leider auch hier nun doch identisch zum CO2-Tarif => es werden weiterhin _keine_ Solarprognosen in evcc bereitgestellt (bis zum nächsten Neustart).