evcc-io/evcc #19226: Implementierung MID Zähler Wago 879-3000

MarioKobus, 2025-02-27:
Es wäre super wenn der Wago MID Zähler 89-3000 mit in die Meter Liste aufgenommen werden würde.
Es handelt sich dabei um einen Zähler mit Modbus-RTU Schnittstelle. 
Anbei die Dokumentation.

[m087930x0-00000000-0de.pdf](https://github.com/user-attachments/files/19004953/m087930x0-00000000-0de.pdf)

andig, 2025-02-27:
Gerne PR, idealerweise in https://github.com/volkszaehler/mbmd

tolot27, 2025-04-13:
Ich habe demnächst zwei 879-3040, die zusammen mit einem SDM120-M an einem 879-9000 betrieben werden. Da kann ich mir das mit anschauen.

guido4096, 2025-05-08:
@andig, is it ok that I take this? I have contacted Wago, they promised to send me a sample so I can test. The code is ready, if indeed they send me a sample, I will submit the code after testing.

andig, 2025-05-08:
Sure, glad if someone picks it up. It would still be nice for greater benefit to add this to https://github.com/volkszaehler/mbmd and add a template here.

guido4096, 2025-05-08:
> Sure, glad if someone picks it up. It would still be nice for greater benefit to add this to https://github.com/volkszaehler/mbmd and add a template here.

That is the intent, that is where I created the starting code. I'll create a pull request there later this week and then tolot27 can try to test? 

I have no rights to assign issues, could you reopen and assign this topic to me?

guido4096, 2025-05-09:
in mbmd: https://github.com/volkszaehler/mbmd/pull/363

guido4096, 2025-05-11:
> > [@andig](https://github.com/andig), is it ok that I take this? I have contacted Wago, they promised to send me a sample so I can test. The code is ready, if indeed they send me a sample, I will submit the code after testing.
> 
> I got my two devices yesterday and can test it, too.

@tolot27 , could you test these changes from here: https://github.com/guido4096/mbmd/tree/add-support-for-Wago-87930 ? I did not receive the Wago yet.

guido4096, 2025-05-11:
@tolot27 , the EVCC version from branch guido4096:add-support-for-wago-879-30xx should be fully functional for testing with the Wago . If you can build this version locally?

guido4096, 2025-05-11:
@andig pull request for EM24/EM24_E1 to support powers and voltages in all cases: https://github.com/evcc-io/evcc/pull/21173

guido4096, 2025-05-16:
@tolot27 if you could send me one of your devices? I can test, and then send it back?

tolot27, 2025-05-18:
> @tolot27 if you could send me one of your devices? I can test, and then send it back?

What's your address? You can send it via mail.

guido4096, 2025-05-18:
I don’t have your email address; can you drop me a mail at guido4096@gmail.com?

closed 2025-05-25 by andig