evcc-io/evcc #19228: SonnenCharger funkktioniert nach Sonnen Migration nicht mehr

bluemlh-pe, 2025-02-27:
### Describe the bug

Seitdem der SonnenCharger in die Sonnen APP migriert wurde funktioniert die Ansteuerung über evcc nicht mehr.
Das Hauptproblem ist, das Sonnen die Ansteuerung im Powermode immer überschreibt. Das stellt sich dann als EIn/AUS/EIN/AUS .. dar. Also keine Sonne evcc schaltet über Modbus aus, Sonnen APP schaltet wieder in den Power Modus und EIN.
VG Peter

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
nicht nötig
```

### Log details

```text
nicht nötig
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

mfuchs1984, 2025-02-27:
Die Wallbox darf nur von einem System gesteuert werden. Also entweder Sonnen APP oder EVCC, beides gleichzeitig geht nicht.

bluemlh-pe, 2025-02-27:
Das ist mir schon klar, aber wie unterbinde ich die Sonnen APP Ansteuerung? Hat bis Dienstag funktioniert (Update von Sonnen).

mfuchs1984, 2025-02-27:
Irgendwie muss es ja möglich sein, den charger aus der Sonne App / dem Sonnen System zu entfernen, Notfalls, indem du ihn aus dem Internet aussperrst.