evcc-io/evcc #19229: UI config: Add loadpoint ABB Terra (OCPP)

michai-27, 2025-02-27:
### Describe the bug

Bei dem Erstellen eines neuen Ladepunktes mit der Vorlage "ABB Terra AC (OCPP)" ist die Konfiguration sowie die Verbindungsaufnahme zur Wallbox möglich, kann aber nicht gespeichert werden. Es erscheint eine Fehlermeldung im Borwser "create failed", siehe Screenshots anbei.

[image]

[image]

### Steps to reproduce

1. Ladepunkt hinzufügen mittels Vorlage "ABB Terra AC (OCPP)"
2. Name "Garage" hinterlegen
3. Station ID mittels Seriennummer der Wallbox pflegen
4. Auf "prüfen" klicken
5. Überprüfen und speichern

### Configuration details

```yaml
Die Konfigurationen sind "nackt" mittels neuem Homeassistant Add-On hinzugefügt worden. Hier die Pfade:

/config/evcc.yaml
/data/evcc.db

Netzwerkkonfigurationen sind default.

Der OCPP Server ist in der TerraConfig App wie folgt hinterlegt: "http://192.168.27.5:8887"
```

### Log details

```text
[main ] INFO 2025/02/27 10:11:36 evcc 0.200.4 (714438d0)
[main ] INFO 2025/02/27 10:11:36 using config file: /config/evcc.yaml
[db ] INFO 2025/02/27 10:11:36 using sqlite database: /data/evcc.db
[main ] INFO 2025/02/27 10:11:37 listening at :7070
[site ] INFO 2025/02/27 10:11:37 site config:
[site ] INFO 2025/02/27 10:11:37 meters: grid ✗ pv ✗ battery ✗
[site ] INFO 2025/02/27 10:11:37 tariffs:
[site ] INFO 2025/02/27 10:11:37 grid: ✗
[site ] INFO 2025/02/27 10:11:37 feed-in: ✗
[site ] INFO 2025/02/27 10:11:37 co2: ✗
[site ] INFO 2025/02/27 10:11:37 solar: solar
[site ] WARN 2025/02/27 10:11:37 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:25:43 !! registering TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:25:43 waiting for chargepoint: 5m0s
[ocpp ] DEBUG 2025/02/27 10:25:48 charge point connected: TACW1140621G0XXX
[ocpp-1] DEBUG 2025/02/27 10:26:01 !! registering :1
[ocpp-1] DEBUG 2025/02/27 10:26:01 waiting for chargepoint: 5m0s
[ocpp ] DEBUG 2025/02/27 10:26:03 charge point disconnected: TACW1140621G0XXX
[ocpp-1] DEBUG 2025/02/27 10:26:05 !! registering :1
[ocpp ] DEBUG 2025/02/27 10:26:29 charge point connected: TACW1140621G0XXX
[ocpp ] ERROR 2025/02/27 10:26:30 ocpp message (5929564): NotSupported - Unsupported feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2025-02-27T09:25:59.000Z"}])
[ocpp ] ERROR 2025/02/27 10:26:30 ocpp message (8454120): NotSupported - Unsupported feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2025-02-27T09:26:30.000Z"}])
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:29:16 !! registering TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:29:16 !! connected TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:29:19 !! registering TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:29:19 !! connected TACW1140621G0XXX:1
[ocpp-1] DEBUG 2025/02/27 10:29:59 !! registering :1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:30:44 !! registering TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:30:44 !! connected TACW1140621G0XXX:1
[ocpp-1] DEBUG 2025/02/27 10:31:01 !! registering :1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:31:23 !! registering TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:31:23 !! connected TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:31:24 !! registering TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:31:24 !! connected TACW1140621G0XXX:1
[ocpp-1] DEBUG 2025/02/27 10:31:30 !! registering :1
[ocpp-1] DEBUG 2025/02/27 10:31:31 !! registering :1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:31:41 !! registering TACW1140621G0XXX:1
[TACW1140621G0XXX-1] DEBUG 2025/02/27 10:31:41 !! connected TACW1140621G0XXX:1
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc 0.200.4 (714438d0)

shortcut18, 2025-02-28:
Hänge mich hier mal an, das Problem habe ich auch mit meinem Fronius Wattpilot Firmware 41.13

Bis evcc 0.133 lief alles perfekt mit dem "alten" template fronius-wattpilot. Seit der Umstellung auf OCPP-only gibt es bei jedem Versuch im Besten Fall das obige "create Failed".

Getestet im Docker-Setup und auch als Standalone-Installation per DEB. Leider ergebnisgleich.

closed 2025-03-02 by andig

michai-27, 2025-03-04:
Die Konfiguration der Wallbox mittels OCPP Anbindung hat nun im ersten Schritt sauber funktioniert.

Nach dem erforderlichen Neustart von evcc erscheint folgende Meldung:

[image]

Hier der dazugehörigen Logs:
[main ] INFO 2025/03/04 08:01:00 evcc 0.200.5 (575d20cd)
[main ] INFO 2025/03/04 08:01:00 using config file: /config/evcc.yaml
[db ] INFO 2025/03/04 08:01:00 using sqlite database: /data/evcc.db
[main ] INFO 2025/03/04 08:01:00 listening at :7070
[TACW1140621G0511-1] DEBUG 2025/03/04 08:01:00 waiting for chargepoint: 5m0s
[ocpp ] DEBUG 2025/03/04 08:03:23 charge point connected: TACW1140621G0511
[ocpp ] DEBUG 2025/03/04 08:03:39 charge point disconnected: TACW1140621G0511
[main ] FATAL 2025/03/04 08:03:39 charger cannot create charger 'db:3': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main ] FATAL 2025/03/04 08:03:39 will attempt restart in: 15m0s
[ocpp ] DEBUG 2025/03/04 08:04:03 charge point connected: TACW1140621G0XXX
[ocpp ] ERROR 2025/03/04 08:04:05 ocpp message (5929564): NotSupported - Unsupported feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2025-03-04T07:03:33.000Z"}])
[ocpp ] ERROR 2025/03/04 08:04:05 ocpp message (8454120): NotSupported - Unsupported feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2025-03-04T07:04:05.000Z"}])

mfuchs1984, 2025-03-04:
@andig eine sinnvolle Ergänzung des loggings wäre m.E., diese Nachricht um die Information zu ergänzen, auf was gerade gewartet wird
`[main ] FATAL 2025/03/04 08:03:39 charger cannot create charger 'db:3': cannot create charger type 'template': cannot create charger type 'ocpp': timeout`

Bspw.
`[main ] FATAL 2025/03/04 08:03:39 charger cannot create charger 'db:3': cannot create charger type 'template': cannot create charger type 'ocpp': timeout waiting for BootNotification`

michai-27, 2025-03-04:
> Mach mal ein ocpp trace log vom Startvorgang, dann sehen wir, was fehlt.

Hier ein gesamtes TRACE Log:

`[main ] INFO 2025/03/04 08:38:25 evcc 0.200.5 (575d20cd)
[main ] INFO 2025/03/04 08:38:25 using config file: /config/evcc.yaml
[db ] INFO 2025/03/04 08:38:25 using sqlite database: /data/evcc.db
[db ] TRACE 2025/03/04 08:38:25 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db ] TRACE 2025/03/04 08:38:25 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2025/03/04 08:38:25 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db ] TRACE 2025/03/04 08:38:25 SELECT * FROM `settings` 3 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main ] INFO 2025/03/04 08:38:26 listening at :7070
[db ] TRACE 2025/03/04 08:38:26 SELECT * FROM `configs` WHERE `configs`.`class` = 5 1 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db ] TRACE 2025/03/04 08:38:26 SELECT * FROM `configs` WHERE `configs`.`class` = 1 2 <nil>
[TACW1140621G0XXX-1] DEBUG 2025/03/04 08:38:26 waiting for chargepoint: 5m0s
[ocpp ] DEBUG 2025/03/04 08:40:48 charge point connected: TACW1140621G0XXX
[ocpp ] TRACE 2025/03/04 08:40:48 send TACW1140621G0XXX: [2,"608890338","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp ] TRACE 2025/03/04 08:40:49 recv TACW1140621G0XXX: [2, "2247220", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp ] TRACE 2025/03/04 08:40:49 send TACW1140621G0XXX: [3,"2247220",{}]
[ocpp ] TRACE 2025/03/04 08:40:49 recv TACW1140621G0XXX: [2, "7440378", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp ] TRACE 2025/03/04 08:40:49 send TACW1140621G0XXX: [3,"7440378",{}]
[ocpp ] TRACE 2025/03/04 08:40:49 recv TACW1140621G0XXX: [3, "608890338", {"status": "Accepted"}]
[ocpp ] TRACE 2025/03/04 08:40:49 send TACW1140621G0XXX: [2,"4246919303","GetConfiguration",{}]
[ocpp ] DEBUG 2025/03/04 08:41:03 charge point disconnected: TACW1140621G0XXX
[main ] FATAL 2025/03/04 08:41:03 charger cannot create charger 'db:3': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main ] FATAL 2025/03/04 08:41:03 will attempt restart in: 15m0s
[main ] TRACE 2025/03/04 08:41:03 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[main ] TRACE 2025/03/04 08:41:04 {"url":"https://api.github.com/repos/evcc-io/evcc/releases/203162908","assets_url":"https://api.github.com/repos/evcc-io/evcc/releases/203162908/assets","upload_url":"https://uploads.github.com/repos/evcc-io/evcc/releases/203162908/assets{?name,label}","html_url":"https://github.com/evcc-io/evcc/releases/tag/0.200.5","id":203162908,"author":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"node_id":"RE_kwDODX4bUs4MHAUc","tag_name":"0.200.5","target_commitish":"master","name":"0.200.5","draft":false,"prerelease":false,"created_at":"2025-03-01T11:12:56Z","published_at":"2025-03-01T11:30:32Z","assets":[{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/233627782","id":233627782,"node_id":"RA_kwDODX4bUs4N7OCG","name":"checksums.txt","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscript [... cut ...]
[ocpp ] TRACE 2025/03/04 08:42:00 recv TACW1140621G0XXX: [2, "0521846", "MeterValues", {"connectorId": 1, "meterValue": [{"timestamp": "2025-03-04T07:42:00.000Z", "sampledValue": [{"value": "16318", "context": "Sample.Clock", "format": "Raw", "measurand": "Energy.Active.Import.Register", "location": "Inlet"}]}]}]
[ocpp ] TRACE 2025/03/04 08:42:00 send TACW1140621G0XXX: [3,"0521846",{}]
[ocpp ] TRACE 2025/03/04 08:42:31 recv TACW1140621G0XXX: [2, "4778614", "Heartbeat", {}]
[ocpp ] TRACE 2025/03/04 08:42:31 send TACW1140621G0XXX: [3,"4778614",{"currentTime":"2025-03-04T07:42:31Z"}]
`

mfuchs1984, 2025-03-04:
```
[ocpp ] TRACE 2025/03/04 08:40:49 send TACW1140621G0XXX: [2,"4246919303","GetConfiguration",{}] 
[ocpp ] DEBUG 2025/03/04 08:41:03 charge point disconnected : TACW1140621G0XXX 
```

Sieht für mich aus, als würde der charger die Verbindung terminieren, anstatt auf den GetConfiguration request zu antworten.

mfuchs1984, 2025-03-04:
Ist ein bekanntes Problem, factory reset der Wallbox hat bei anderen geholfen.
https://github.com/evcc-io/evcc/issues/15493#issuecomment-2308143746
https://github.com/evcc-io/evcc/issues/15533#issuecomment-2308195368

michai-27, 2025-03-04:
Danke für die Analyse. Werde ich versuchen und zeitnah berichten.

michai-27, 2025-03-04:
Hat nach dem Reset der Wallbox via TerraConfig reibungslos funktioniert. Danke euch!

[image]