evcc-io/evcc #19238: Nach Update auf 0.200.3 (mittlerweile 0.200.4) keine curcuits bzw. kein Lastmanagement mehr

neolight1302, 2025-02-27:
### Describe the bug

Nach o.g. Update startet evcc nicht mehr.

An der Konfiguration wurde nichts geändert..

Log:

[main ] INFO 2025/02/25 15:17:14 evcc 0.200.3
[main ] INFO 2025/02/25 15:17:14 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/25 15:17:14 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/25 15:17:15 listening at :7070
[main ] FATAL 2025/02/25 15:17:16 loadpoint [lp-1] circuit: not found: main
[main ] FATAL 2025/02/25 15:17:16 will attempt restart in: 15m0s

Eindeutig ein Problem von Version > 0.200.2
Nach Downgrade auf 0.200.2 läuft es...

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id50de85bc6d3333b9be7f35651936c1297e6bcf6d724d6e4af49bf319d63d02c9

interval: 5s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: shelly-3em 
  usage: grid  
  host: 192.168.178.99 
  name: grid1

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.92  
  name: wallbox5
- type: template
  template: go-e-v3 
  host: 192.168.178.108
  name: wallbox6

circuits:
  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: Hauptstromkreis # Anzeige in der UI (später)
    maxCurrent: 33 # 63A (optional)
    #maxPower: 30000 # 30kW (optional)
    meter: grid1 # optional

loadpoints:

- title: Garage
  charger: wallbox5
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  circuit: main
  enable:
    threshold: 0
    delay: 0.25m
  disable:
    threshold: 200 # maximum import power (W)
    delay: 0.25m

- title: Einfahrt
  charger: wallbox6
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  circuit: main
  enable:
    threshold: 0
    delay: 0.25m
  disable:
    threshold: 200 # maximum import power (W)
    delay: 0.25m
    

site:
  title: Waltraute
  meters:
    grid: grid1
  residualPower: -300
```

### Log details

```text
[main ] INFO 2025/02/25 15:17:14 evcc 0.200.3
[main ] INFO 2025/02/25 15:17:14 using config file: /etc/evcc.yaml
[db ] INFO 2025/02/25 15:17:14 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/02/25 15:17:15 listening at :7070
[main ] FATAL 2025/02/25 15:17:16 loadpoint [lp-1] circuit: not found: main
[main ] FATAL 2025/02/25 15:17:16 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.4

andig, 2025-02-27:
Can't confirm. Your yaml starts just fine:

```
[site  ] INFO 2025/02/27 17:27:37 site config:
[site  ] INFO 2025/02/27 17:27:37   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2025/02/27 17:27:37     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/02/27 17:27:37   tariffs:
[site  ] INFO 2025/02/27 17:27:37     grid:      ✗
[site  ] INFO 2025/02/27 17:27:37     feed-in:   ✗
[site  ] INFO 2025/02/27 17:27:37     co2:       ✗
[site  ] INFO 2025/02/27 17:27:37     solar:     solar
[lp-1  ] INFO 2025/02/27 17:27:37 loadpoint 1:
[lp-1  ] INFO 2025/02/27 17:27:37   mode:        pv
[lp-1  ] INFO 2025/02/27 17:27:37   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/02/27 17:27:37   meters:      charge ✓
[lp-1  ] INFO 2025/02/27 17:27:37     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2025/02/27 17:27:37 loadpoint 2:
[lp-2  ] INFO 2025/02/27 17:27:37   mode:        pv
[lp-2  ] INFO 2025/02/27 17:27:37   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-2  ] INFO 2025/02/27 17:27:37   meters:      charge ✓
[lp-2  ] INFO 2025/02/27 17:27:37     charge:    power ✓ energy ✓ currents ✓
```

Did you add any config settings via UI, potentially overriding your yaml config? See

    evcc config

neolight1302, 2025-02-27:
Yes, there are three vehicles config over UI.

[main  ] INFO 2025/02/27 18:48:51 evcc 0.200.2
[main  ] INFO 2025/02/27 18:48:51 using config file: /etc/evcc.yaml
[main  ] FATAL 2025/02/27 18:48:51
Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked with no explicit database path.
Running the same config with multiple databases can lead to expiring vehicle tokens.

If you want to use the existing service database run the following command:

sudo evcc --database /var/lib/evcc/evcc.db

If you want to create a new user-space database run the following command:

evcc --database ~/.evcc/evcc.db

If you know what you're doing, you can skip the database check with the --ignore-db flag.

naltatis, 2025-02-27:
@neolight1302 please try opening the "Loadmanagement" section in Config UI, delete the contents for the yaml editor, save and restart. Looks like you've leftovers there that take precedence.

neolight1302, 2025-02-27:
That was actually the problem. Now it's running again. thank you!

closed 2025-02-27 by neolight1302