evcc-io/evcc #19250: Foxess H3 Hybrid Wechselrichter

79Anni, 2025-02-28:
### Describe the bug

ich setze eine Foxxess H3 Wechselrichter, eine 15 kWp Pav-Anlage, eine 11 kWh Foxess Batterie und eine Stiebel Eltron Wärmepumpe ein. 
Beim Versuch das evcc Add On in mein Home Assistant auf einem Raspberry Pi einzusetzen erhalte ich folgende Fehlermeldungen. Die go-e Wallbox und das Fahrzeug ist im Add On ergänzt und wird ohne Fehlermeldung geführt. Die evcc.yaml im Verzeichnis /addon_configs/49686a9f_evcc/ ist leer!
Ergänzend möchte ich erwähnen das ich die Wechselrichter-Daten bereits erfolgreich mit dem Repository Foxess Modbus über einen Elfin EW11 abgreife.
Ich hoffe Ihr könnt mir helfen, da bereits seit einiger Zeit versuche das Problem im Alleingang zu lösen

### Steps to reproduce

1. Foxess H3 Wechselrichter erfolgreich mit dem Repository Foxess Modbus über einen Elfin EW11 eingebunden.
2. Sponsorship ist eingeholt und ergänzt
3. AddOn evcc für Home Assistant installiert
4. Wallbox eingeichtet => OK
5. Fahrzeug eingerichtet => OK
6. Netzanschluss eingerichtet => NOK siehe Log

### Configuration details

```yaml
evcc.yaml im Verzeichnis /addon_configs/49686a9f_evcc/ ist leer
```

### Log details

```text
[main ] INFO 2025/02/28 06:14:16 evcc 0.200.4
[main ] INFO 2025/02/28 06:14:16 using config file: /config/evcc.yaml
[db ] INFO 2025/02/28 06:14:16 using sqlite database: /data/evcc.db
[main ] INFO 2025/02/28 06:14:17 listening at :7070
[bmw ] DEBUG 2025/02/28 06:14:17 identity.Login - database token found
[bmw ] DEBUG 2025/02/28 06:14:17 identity.Login - database token invalid. Proceeding to login via user, password and captcha.
[main ] ERROR 2025/02/28 06:14:17 creating vehicle db:6 failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized)
[lp-1 ] DEBUG 2025/02/28 06:14:17 set title: go-e Wallbox
[lp-1 ] DEBUG 2025/02/28 06:14:17 set priority: 0
[lp-1 ] DEBUG 2025/02/28 06:14:17 set smart cost limit: <nil>
[lp-1 ] DEBUG 2025/02/28 06:14:17 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-1 ] DEBUG 2025/02/28 06:14:17 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/02/28 06:14:17 set session energy limit: 0
[lp-1 ] DEBUG 2025/02/28 06:14:17 set session soc limit: 0
[lp-1 ] DEBUG 2025/02/28 06:14:17 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0x4000e7787c}
[lp-1 ] DEBUG 2025/02/28 06:14:17 set default mode: pv
[lp-1 ] DEBUG 2025/02/28 06:14:17 set phases: 3
[lp-1 ] DEBUG 2025/02/28 06:14:17 charge total import: 6.000kWh
[site ] INFO 2025/02/28 06:14:17 site config:
[site ] INFO 2025/02/28 06:14:17 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/02/28 06:14:17 grid: power ✓ energy ✓ currents ✗
[site ] INFO 2025/02/28 06:14:17 pv 1: power ✓ energy ✗ currents ✗
[site ] INFO 2025/02/28 06:14:17 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site ] INFO 2025/02/28 06:14:17 vehicles:
[site ] INFO 2025/02/28 06:14:17 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site ] INFO 2025/02/28 06:14:17 tariffs:
[site ] INFO 2025/02/28 06:14:17 grid: ✗
[site ] INFO 2025/02/28 06:14:17 feed-in: ✗
[site ] INFO 2025/02/28 06:14:17 co2: ✗
[site ] INFO 2025/02/28 06:14:17 solar: solar
[lp-1 ] INFO 2025/02/28 06:14:17 loadpoint 1:
[lp-1 ] INFO 2025/02/28 06:14:17 mode: off
[lp-1 ] INFO 2025/02/28 06:14:17 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2025/02/28 06:14:17 meters: charge ✓
[lp-1 ] INFO 2025/02/28 06:14:17 charge: power ✓ energy ✓ currents ✓
[lp-1 ] DEBUG 2025/02/28 06:14:17 phase timer inactive
[lp-1 ] DEBUG 2025/02/28 06:14:17 pv timer inactive
[lp-1 ] INFO 2025/02/28 06:14:17 vehicle updated: unknown -> i4
[site ] WARN 2025/02/28 06:14:17 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2025/02/28 06:14:17 ----
[lp-1 ] DEBUG 2025/02/28 06:14:17 charge power: 0W
[lp-1 ] DEBUG 2025/02/28 06:14:17 charge currents: [0 0 0]A
[site ] ERROR 2025/02/28 06:14:19 pv 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site ] ERROR 2025/02/28 06:14:19 battery 1 power: read failed: modbus: exception '2' (illegal data address), function '3'
[site ] ERROR 2025/02/28 06:14:19 battery 1 soc: read failed: modbus: exception '2' (illegal data address), function '3'
[site ] ERROR 2025/02/28 06:14:19 grid power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site ] DEBUG 2025/02/28 06:14:29 ----
[lp-1 ] DEBUG 2025/02/28 06:14:29 charge power: 0W
[lp-1 ] DEBUG 2025/02/28 06:14:29 charge currents: [0 0 0]A
[site ] ERROR 2025/02/28 06:14:30 pv 1 power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
[site ] ERROR 2025/02/28 06:14:30 battery 1 power: read failed: modbus: exception '2' (illegal data address), function '3'
[site ] ERROR 2025/02/28 06:14:30 battery 1 soc: read failed: modbus: exception '2' (illegal data address), function '3'
[site ] ERROR 2025/02/28 06:14:30 grid power: add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.4

andig, 2025-02-28:
Was sagt

    evcc meter —log trace,db:error

79Anni, 2025-02-28:
In der Termial Befehlszeile "evcc meter —log trace,db:error" eingegeben, bringt folgende Inf0 "zsg: command not found: evcc"

christophritterbach, 2025-03-01:
Ich bin zwar kein Entwickler von evcc, habe aber in den letzten Tagen meinen Foxess H3 erfolgrech angebunden.
Ich habe den Wechselrichter direkt mit RS485 an meinen alten Raspberry angeschlossen. Es werden aber die gleichen Register angesprochen.
Ist denn die korrekte ID angegeben worden? Wenn im WR nichts verändert wurde, dann ist die ID des Gerätes (im Bus) 247.
Wenn Du den WR am HomeAssistant eingerichtet hast, wird auch dort die ID angegeben sein. Oder sie steht im Elfin EW11.
Und wenn Du dort nichts findest, dann schaue mal auf dem Display des Gerätes. Das ist eben nur elende Tipperei auf dem kleinen Display.
Zum o.a. Befehl: evcc könnte in /usr/bin liegen.
Also `/usr/bin/evcc meter`, besser wäre aber `sudo /usr/bin/evcc meter`.
Das führt bei meiner Version, v0.200.4 leider auch zu einem Fehler, ich muss die Datenbank angeben.
`/usr/bin/evcc  --database /var/lib/evcc/evcc.db meter`
und als Ausgabe
```
Foxess_ECS2900
--------------
Power:        5W
Soc:          16%
Controllable: true

Netz
----
Power:  789W
Energy: 7788.7kWh

Foxess_H3
---------
Power: 0W
```

79Anni, 2025-03-01:
Hi, die ID ist 247. Ich greife ja bereits mit Foxss Modbus darauf zu und die Daten werden korrekt angezeigt. Nur bei evcc klappt es nicht. Ich werde deine Angeben prüfen, bin aber jetzt für 2 Wochen dienstlich unterwegs, melde mich dann aber mit den Ergebnissen zurück. Danke

Yahoo Mail - E-Mail vereinfacht

79Anni, 2025-03-17:
Sorry, war einige Zeit beruflich unterwegs und konnte mich nicht weiter mit dem Problem beschäftigen. 

Leider weiß ich gar nicht weiter. Das problem ist nicht gelöst!

@andig schrieb: was sagt evcc meter —log trace,db:error

Ich habe diesen Befehl in die Window Eingabeaufforderung wie auch in HA Terminal eingegeben aber. Bekomme aber immer die Antwort "command not found"

@andig kannst Du mir bitte erklären was ich zu tun habe. Bin leider ein rookie und benötige entwas mehr hilfe.
Danke