evcc-io/evcc #19252: Solcast korrekt und Vorhersage abgeschnitten für Übermorgen

otrsz, 2025-02-28:
### Describe the bug

Unter "Vorhersage" wird der Tag nach dem Tag abgeschnitten, hier im Beispiel werden nur 152 wh angezeigt, weil der ganze Tag nicht berücksichtigt wird.

[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Solar: Typ: Vorlage Vorlage: Solcast Standort: 361d-5a83-dc4xxxx Token: SazEZmuZrhOqcTi0rqVH7evS_xxxxx Intervall: 3h
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.4

andig, 2025-02-28:
@naltatis lösen wir mit https://github.com/evcc-io/evcc/pull/18867?

andig, 2025-02-28:
@otrsz Logfile fehlt. Es ist unklar, woher Du >48h Forecast im API haben willst.

closed 2025-02-28 by andig

naltatis, 2025-02-28:
Ja, das Thema wird mit https://github.com/evcc-io/evcc/pull/18867 gelöst. Die Ansicht bleibt vorerst auf max 48h. Liegen aber mehr Daten vor werden diese mit dem PR komplett betrachtet.