evcc-io/evcc #19253: Sofar: Improve PV Power accuracy using individual strings

cschlipf, 2025-02-28:
> Hallo, das gleiche Thema mit der Auflösung in 100 W Schritten trifft neben der Batterie Ladung / Endladung auch das Register zum Einlesen der solaren Leistung (Power_PV_Total) zu. Da dieses Register verwendet wird, kann steigt die aktuelle PV-Leistung nur in 100 W Schritten. Die Register der einzelnen Strings sind allerdings in 10 W Schritten aufgelöst. Die könnte man ähnlich den Batterien einzeln aus den Registern lesen und dann addieren. Ich mache das zur Zeit, in dem ich den Wechselrichter über eine Customintegration auslese , in der die beiden Strings des HYD einzeln gelesen und dann summiert werden. Also so:
> 
> name: HYD_pv
>     type: custom
>     power:
>       source: calc
>       add:
>       - source: modbus
>         uri: 192.168.1.xyz:502
>         id: 1
>         register:
>           address: 0x586 # Power PV 1
>           type: holding
>           decode: int16
>         scale: 10
>       - source: modbus
>         uri: 192.168.1.xyz:502
>         id: 1
>         register:
>           address: 0x589 # Power PV 2
>           type: holding
>           decode: int16
>         scale: 10
>   
> 
> MfG
> 
> Schroediman
> 
> Jungs, das Projekt ist toll. 

 _Originally posted by @Schroediman in [#19219](https://github.com/evcc-io/evcc/issues/19219#issuecomment-2689264526)_

@cschlipf will take care of creating a PR

cschlipf, 2025-02-28:
@Schroediman

Wieviele PV Eingänge sollten wir denn supporten? Das Protokoll scheint hier mit 16 sehr großzügig zu sein, aber aktuell hat soweit ich weiß, kein Consumer Wechselrichter von Sofar mehr als 3 Eingänge.

Ich würde sagen, wenn wir 4 Strings supporten sind wir auf der sicheren Seite. Was meinst Du?

cschlipf, 2025-02-28:
Das Calc ist nicht notwendig. Bei mehreren Meters wird automatisch addiert.

Schroediman, 2025-02-28:
@cschlipf 
Bei Sofar scheint man ja mit 4 PV Strängen in dem (semi)professionellen Bereich bis 50 kW zu liegen, da gibt's dann noch einen Wechselrichter, der kann bis 80 kW und hat 6 MPPTs. Ich denke so 4 - 6  PV-Stränge werden wohl für unsere Anwendung ausreichen.

MfG

Schroediman

cschlipf, 2025-02-28:
Ich glaub bei 80kW braucht man kein EVCC mehr für Überschussladen :)

Ich belasse es jetzt mal bei 4 Strängen. Erweitern ist leicht aber ich glaube nicht, dass wir hier jemand sehen werden, der soviel auf dem Dach hat.

closed 2025-02-28 by andig

Schroediman, 2025-02-28:
Moin könntet ihr die Adressen für Batterie  noch einmal gegenlesen?

Die Power_Bat 1 scheint nicht zu passen laut Protokoll die 0x606.

    {{- if eq .storageunit "1" }}
      address: 0x0607 # Power_Bat1
      {{- else if eq .storageunit "2" }}
      address: 0x060D # Power_Bat2
      {{- else if eq .storageunit "3" }}
      address: 0x0614 # Power_Bat3
      {{- else if eq .storageunit "4" }}
      address: 0x061B # Power_Bat4
      {{- else if eq .storageunit "5" }}
      address: 0x0622 # Power_Bat5
      {{- else if eq .storageunit "6" }}
      address: 0x0629 # Power_Bat6
      {{- else if eq .storageunit "7" }}
      address: 0x0630 # Power_Bat7
      {{- else if eq .storageunit "8" }}
      address: 0x0637 # Power_Bat8
      {{- end }}
      type: holding
      decode: int16
    scale: -10
  soc:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      {{- if eq .storageunit "1" }}
      address: 0x0608 # SOC_Bat1
      {{- else if eq .storageunit "2" }}
      address: 0x060F # SOC_Bat2
      {{- else if eq .storageunit "3" }}
      address: 0x0616 # SOC_Bat3
      {{- else if eq .storageunit "4" }}
      address: 0x061D # SOC_Bat4
      {{- else if eq .storageunit "5" }}
      address: 0x0624 # SOC_Bat5
      {{- else if eq .storageunit "6" }}
      address: 0x062B # SOC_Bat6
      {{- else if eq .storageunit "7" }}
      address: 0x0632 # SOC_Bat7
      {{- else if eq .storageunit "8" }}
      address: 0x0639 # SOC_Bat8

cschlipf, 2025-02-28:
Danke Dir. Die anderen register scheinen zu stimmen, aber bei PowerBat1 hat sich echt ein Fehler eingeschlichen. PR #19265 fixt das.

PR #19258 (2025-02-28): Sofar: Improve PV Power accuracy by summing up strings
By summing up the PV Power from individual strings we improve the accuracy from 100W to 10W

Closes #19253