evcc-io/evcc #19268: Add automated SolarEdge ModBus "StorageControl" Setting

Ph1975, 2025-02-28:
**Is your feature request related to a problem? Please describe.**
SolarEdge Infrastructure to be controlled by evcc require certain ModBus Register Settings on the device. There is no "standard" Tool out there which can easily perform this Modbus setting but EVCC has everything it needs to simply apply the right modbus setting (i guess!).

https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/
Page 18

**Describe the solution you'd like**
Please automate the BatterySetting to set "Storage Control Mode  4" (as described in your documentation) in the configuration tab of evcc, so that this can be easily enabled without the need to become a modbus professional.
OR: supply maybe a simple python script which can do the job.

**Describe alternatives you've considered**
Installation of OpenHab (which seems to have a python extension which can do this) is way too heavyweight.

**Additional context**

Ph1975, 2025-02-28:
Maybe i just found my way through this..... 
I would be very happy if someone could confirm this as a workign solution.

Writing the register and then reading it again seems to have set it to "4".
Not sure this is everything which needs to be done in that matter.

`from pymodbus.client import ModbusTcpClient
from pymodbus.exceptions import ModbusException
from pymodbus.pdu import ExceptionResponse
from pymodbus import FramerType
from pyModbusTCP import utils

# evcc modbus proxy for SolarEdge
HOST = "192.168.50.207"
PORT = 5021

def get_data_type(format: str):
    """Return the ModbusTcpClient.DATATYPE according to the format"""
    for data_type in ModbusTcpClient.DATATYPE:
        if data_type.value[0] == format:
            return data_type

ModbusTcpClient = ModbusTcpClient(
    host=HOST,
    port=PORT,
    # Common optional parameters:
    framer=FramerType.SOCKET,
    timeout=5,
)

ModbusTcpClient.connect()

 
 # READ
for addr, format, factor, comment, unit in ( 
    # data_type according to ModbusClientMixin.DATATYPE.value[0]
    (40000, "s", 1, "C_SunSpec_ID", ""), 
    (40004, "s", 1, "C_Manufacturer", ""), 
    (57348, "H", 1, "Storage Control Mode", ""),  # E004 (same as F704)
    (63236, "H", 1, "Storage Control Mode", ""),  #F704
    (63237, "H", 1, "Storage AC Charge Policy", ""),  #F705
    (63238, "f", 1, "Storage AC Charge Limit", ""), #F706
    (63240, "f", 1, "Storage Backup Reserved Setting", "%"), #F708
    (63242, "H", 1, "Storage Charge/Discharge Default Mode", ""), #F70A
    (63243, "H", 1, "Remote Control Command Timeout", "sec"), #F70B
    (63245, "H", 1, "Remote Control Command Mode", ""), #F70D
    (63246, "f", 1, "Remote Control Charge Limit", ""), #F70E
    
    (61762, "i", 1, "AdvancedPwrControlEn", ""), #F142
    (61700, "i", 1, "ReactivePwrConfig", ""), #F104
    
):
    data_type = get_data_type(format)
    if (addr == 40000):
        count = 2
    elif (addr == 40004):
        count = 16
    else:
        count = data_type.value[1]
    
    
    var_type = data_type.name
    
    rr = ModbusTcpClient.read_holding_registers(address=addr, count=count, slave=1)
    value = ModbusTcpClient.convert_from_registers(rr.registers, data_type)
    print(comment + " -> " + str(value) + " " + str(unit))
    
    
# WRITE

for addr, format, factor, comment, payload in ( 
     (57348, "H", 1, "Storage Control Mode", 4),
#    (61762, "i", 1, "AdvancedPwrControlEn", 1), #F142
#    (61700, "i", 1, "ReactivePwrConfig", 4), #F104
#    (61696, "h", 1, "CommitPowerControl", 1), #F100
):
    data_type = get_data_type(format)
    count = data_type.value[1]
    try:
        rr = ModbusTcpClient.write_register(address=addr, value=payload)
    except ModbusException as exc:
        print(f"Received ModbusException({exc}) from library")
    
    if rr.isError():
        print(f"Received exception ({rr})")
`

andig, 2025-02-28:
You can use any light-weight modbus too, like mbrtu or QModbusMaster to do this.

closed 2025-02-28 by andig