evcc-io/evcc #19271: OCPP: Charge Power Timeout, Mennekes Smart T (ACU)

benesolar, 2025-02-28:
### Describe the bug

Ladestation überträgt lt. Log ca. alle 60 Sekunden (+- 7Sekunden) `Power.Active.Import` Werte.
(`"connectorId": 1` = Loadpoint 2)
Diese Werte werden nur übertragen wenn am Anschluss ein Auto steckt.

Oft kommt ca. 30 Sekunden später: `charge power: timeout `

Sollten hier nicht das `meterinterval: 60s` abgewartet werden?

Beispiel Auszug Log:
```
[lp-2 ] ERROR 2025/02/28 20:42:35 charge power: timeout
[lp-2 ] ERROR 2025/02/28 20:42:45 charge power: timeout 
```

```
[ocpp ] TRACE 2025/02/28 20:42:03 recv Ladestation: [2, "171b37e8-f60c-11ef-8f0a-000a1484c32e", "MeterValues", {"meterValue": [{"timestamp": "2025-02-28T19:42:01.000Z", "sampledValue": [{"measurand": "Power.Active.Import", "unit": "W", "value": "11"}, {"phase": "L1", "measurand": "Power.Active.Import", "unit": "W", "value": "0"}, {"phase": "L2", "measurand": "Power.Active.Import", "unit": "W", "value": "11"}, {"phase": "L3", "measurand": "Power.Active.Import", "unit": "W", "value": "0"}]}], "connectorId": 1, "transactionId": 1740755170}]
[ocpp ] TRACE 2025/02/28 20:43:10 recv Ladestation: [2, "3eb89624-f60c-11ef-8f0a-000a1484c32e", "MeterValues", {"meterValue": [{"timestamp": "2025-02-28T19:43:07.000Z", "sampledValue": [{"measurand": "Power.Active.Import", "unit": "W", "value": "11"}, {"phase": "L1", "measurand": "Power.Active.Import", "unit": "W", "value": "0"}, {"phase": "L2", "measurand": "Power.Active.Import", "unit": "W", "value": "11"}, {"phase": "L3", "measurand": "Power.Active.Import", "unit": "W", "value": "0"}]}], "connectorId": 1, "transactionId": 1740755170}]
```

### Steps to reproduce

Auto angesteckt, Stromverbrauch durch laufenden Ladevorgang oder Eigenverbrauch der Station.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: /var/lib/evcc/evcc.db
telemetry: true

# log settings
log: error
levels:
  core: error
  lp-1: error
  lp-2: error
  lp-3: error
  cache: error
  db: error
  ocpp: error

# meter definitions
meters:
  - name: grid
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.1.248
  - name: pv1
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.1.248
  - name: pv2
    type: template
    template: sunspec-inverter
    usage: pv
    modbus: tcpip
    id: 1
[... 216 lines cut ...]
      chats:
      - -xxx # list of chat ids
```

### Log details

```text
see comment
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.4 (b2fb1aa4)

benesolar, 2025-02-28:
[evcc-20250228-210633-trace.log](https://github.com/user-attachments/files/19032808/evcc-20250228-210633-trace.log)

mfuchs1984, 2025-02-28:
Mit `meterinterval: 60s` sagt evcc der Wallbox, sie soll die Messwerte nur alle 60 Sekunden bereitstellen. Das timeout in evcc liegt jedoch bei 30 Sekunden.

andig, 2025-02-28:
Wieso ist das überhaupt gesetzt?

benesolar, 2025-02-28:
Die Ladestation akzeptiert keinen kürzeren Zeitraum.
Trotzdem weg damit?

mfuchs1984, 2025-02-28:
Ne, dann würde ich es lassen und mit den Warnungen leben. Auf die TriggerMessage reagiert sie ja nicht wirklich.

benesolar, 2025-02-28:
Ich hatte den Fehler ca. 60x die Stunde.

Den Punkt deaktiviert und Neustart Station und evcc.
Fehler nur 1x nach Start?

[evcc-20250228-223901-trace.log](https://github.com/user-attachments/files/19033484/evcc-20250228-223901-trace.log)

benesolar, 2025-03-01:
Fehler kommt nun nur mehr selten:

[image]

Kann gerne geschlossen werden wenn es für euch so passt.

andig, 2025-03-02:
Wenn keine Transaktion läuft passiert das:

```go
	// zero value on timeout when no transaction is running
	if conn.isMeterTimeout() {
		if conn.txnId != 0 {
			return 0, api.ErrTimeout
		}

		return 0, nil
	}
```

Das MeterTimeout dafür ist 30s. Das kann nicht klappen wenn deine Box nur alle 60s Werte raus rückt. Wahrscheinlich wäre es hier sinnvoll, das ganze `meterinterval` abzuwarten?

mfuchs1984, 2025-03-02:
Plus ein wenig Toleranz.

closed 2025-03-02 by andig