evcc-io/evcc #19272: migrate Befehl sollte den Pfad der aktiven Datenbank prüfen

HolgerMiara, 2025-02-28:
**Is your feature request related to a problem? Please describe.**
Der Befehls `evcc migrate` schreibt die Daten aus der evcc.yaml ungefragt in eine Datenbank unter dem Pfad `/root/.evcc/evcc.db`. Meine aktive Datenbank liegt aber seit der Version 0.200.1 in `/config/evcc/evcc.db` bzw. in `/addon_configs/49686a9f_evcc/evcc/evcc.db`.

**Describe the solution you'd like**
Ich denke, dass der `migrate` Befehl in einem der nächsten Releases den Pfad der aktiven Datenbank prüfen und nutzen sollte.

heatpumper, 2025-03-01:
Hallo Holger, wie kannst du das prüfen? Bei mir funktioniert der Migrate Befehl nicht, vielleicht hilft mir das.

HolgerMiara, 2025-03-01:
Moin Andi, markier diese Antwort gerne als off-topic, aber hier eine schnelle Anleitung für @heatpumper 

1. Installier das Addon [Advanced SSH & Web Terminal](https://github.com/hassio-addons/addon-ssh) (**nicht** das offizielle Addon "Terminal & SSH" aus dem offiziellen add-on store). 
2. Deaktivier den Schiebeschalter "Protection mode" und starte das Addon.  
3. Folge der Anleitung [Wie kann ich das evcc CLI (im Homeassistent Addon) nutzen?](https://docs.evcc.io/docs/installation/home-assistant#wie-kann-ich-das-evcc-cli-nutzen) um Zugriff auf evcc per Kommandozeile zu bekommen.

closed 2025-04-02 by github-actions[bot]