evcc-io/evcc #19273: Nach Migration des Modbus proxy von YAML --> Datenbank lässt evcc nicht mehr starten (HomeAssistant Addon)

HolgerMiara, 2025-02-28:
### Describe the bug

Folgender Modbusproxy funktionierte einwandfrei in der evcc.yaml; Drittgeräte konnten über den Proxy auf den Wechselrichter zugreifen. 
```yaml
modbusproxy:
  - port: 502
    uri: 192.168.188.100:502
``` 

Nach Ausführen des Befehls `evcc migrate` startete evcc nicht mehr mit folgender Fehlermeldung
[image]

Der `migrate` Befehl hat aus meinen drei YAML Zeilen folgenden Eintrag in der Datenbank generiert, mit dem evcc _nicht_ mehr startet:
[image]

### Steps to reproduce

1. siehe oben
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2025/02/28 22:56:57 evcc 0.200.4
[main  ] INFO 2025/02/28 22:56:57 using config file: /config/evcc.yaml
[db    ] INFO 2025/02/28 22:56:57 using sqlite database: /config/evcc/evcc.db
[circuit-main] DEBUG 2025/02/28 22:57:00 validation of max power disabled
[circuit-carport] DEBUG 2025/02/28 22:57:00 validation of max power disabled
[solcast] ERROR 2025/02/28 22:57:01 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/02/28 22:57:01 creating tariff solar failed: cannot create tariff type 'template': cannot create tariff type 'solcast': unexpected status: 429 (Too Many Requests)
[lp-1  ] DEBUG 2025/02/28 22:57:01 charge total import: 15493.090kWh
[lp-2  ] DEBUG 2025/02/28 22:57:01 charge total import: 12291.289kWh

Konfiguration (/config/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

database:
  type: sqlite
  dsn: /config/evcc/evcc.db

  # unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

telemetry: true

modbusproxy:
  - port: 502
    uri: 192.168.188.100:502

meters:
  - type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.188.100
    port: 502
    usage: grid
    modbus: tcpip
    timeout: 15s
    storageunit: 1
    name: grid1
  - type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.188.100
    port: 502
    usage: pv
    modbus: tcpip
    timeout: 15s
    storageunit: 1
    name: pv1
  - type: template
    template: huawei-sun2000
    id: 1
    host: 192.168.188.100
    port: 502
    usage: battery
    modbus: tcpip
    timeout: 15s
    storageunit: 1
    name: battery1

chargers:
  - type: template
    template: easee
    user: *****
    password: *****
    charger: EH**KNJ3
    timeout: 20s
    authorize: true
    name: wallbox1
  - type: template
    template: easee
    user: *****
    password: *****
    charger: EH**ENH2
    timeout: 20s
    authorize: true
    name: wallbox2

vehicles:
  - type: template
    template: vw
    user: *****
    password: *****
    vin: *****
    title: Ingolf
    icon: car
    capacity: 58
    phases: 3
    mode: pv
    minCurrent: 6
    maxCurrent: 16
    cache: 15m
    timeout: 10s
    name: car_id3
  - type: template
    template: vw
    user: *****
    password: *****
    vin: *****
    title: Multivan Bo
    icon: shuttle
    capacity: 10.4
    phases: 1
    mode: pv
    minCurrent: 6
    maxCurrent: 16
    cache: 15m
    timeout: 10s
    name: car_mv

circuits:
  - name: main
    title: Hovedsikring
    maxCurrent: 25
    meter: grid1
  - name: carport
    title: Bilopladning
    maxCurrent: 20
    parent: main

loadpoints:
  - title: Carport left
    charger: wallbox1
    vehicle: car_mv
    mode: pv
    circuit: carport
  - title: Carport right
    charger: wallbox2
    vehicle: car_id3
    mode: pv
    circuit: carport

site:
  title: Miara hyblen
  meters:
    grid: grid1
    pv:
      - pv1
    battery:
      - battery1
  residualPower: -500

tariffs:
  currency: DKK
  grid:
    type: energinet
    region: dk2 # or dk1
    charges: 0.39 # my case
    tax: 0.25 # 25% VAT
    # https://radiuselnet.dk/elnetkunder/elprisen/
    formula: |
      gridfee := 0.2929     // high load
      if ts.Hour() < 6 {
        gridfee = 0.0976    // low load
      } else if ts.Hour() >= 17 && ts.Hour() < 21 {
        gridfee = 0.8788    // peak load
      }
      ( price + charges + gridfee ) * ( 1 + tax )

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: energinet # Energinet using the price in DKK
    region: DK* 
    charges: -0.01 # optional, additional charges per kWh
    tax: 0.0 # optional, additional tax

  co2:
    type: template
    template: electricitymaps-free # https://app.electricitymaps.com/map
    token: *****
    zone: DK-DK*

  solar:
    type: template
    template: solcast
    site: 32a0-2ef8-fcc6-5358 # Ressource ID deiner Anlage
    token: *****

Version: `0.200.4`
```

### Log details

```text
Sorry, ich bin gerade glücklich, dass evcc wieder läuft und habe den Fehler heute Abend nicht noch mal reproduziert um das Logfile anzuhängen. :dog:
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.4

andig, 2025-03-01:
@HolgerMiara danke für das Issue. Logfile brauchts nicht, das sollte sich auch so reproduzieren lassen.

andig, 2025-03-01:
@naltatis da nimmt migrate wohl das falsche struct

closed 2025-03-01 by naltatis