evcc-io/evcc #19275: Solcast Vorhersage wird nach Einrichtung in Config UI nicht angezeigt

marleyan333, 2025-02-28:
### Describe the bug

Trotz Einrichtung der Solcast-Vorhersage über die Cofig-UI wird keine PV-Vorhersage angezeigt.Netztarif und Feedin funktionieren, nur die PV-Prognose nicht.Bei Konfig über die evcc.yaml startet evcc gar nicht mehr.

### Steps to reproduce

Genau sohabe ich alles in Config-UI unter Tarife hinterlegt:

currency: EUR # (default EUR)

solar:
  type: template
  template: solcast
  site: 0d04-eb62-fc1e-82d8 # Ressource ID deiner Anlage
  token: https://api.solcast.com.au/rooftop_sites/0d04-eb62-fc1e-82d8/forecas>
  interval: 3h # optional

grid:
  type: fixed
  price: 0.2955 # EUR/kWh (default)
  zones:
    - days: Mo-So
      hours: 0-5
      price: 0.1609 # EUR/kWh

feedin: # price for feeding solar energy to the grid
  type: fixed
  price: 0.10 # EUR/kWh

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 10s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.41  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.41  
  port: 8080  
  name: pv2

chargers:
- type: template
  template: pmcc 
  ski: xxxxxxx 
  ip: 192.168.178.152  
  name: wallbox5

vehicles:
- type: template
[... 143 lines cut ...]
Feb 28 23:01:15 raspberrypi evcc[30588]: [site  ] DEBUG 2025/02/28 23:01:15 battery 1 power: -5W
Feb 28 23:01:15 raspberrypi evcc[30588]: [site  ] DEBUG 2025/02/28 23:01:15 battery 1 soc: 0%
Feb 28 23:01:15 raspberrypi evcc[30588]: [site  ] DEBUG 2025/02/28 23:01:15 site power: 4603W
Feb 28 23:01:15 raspberrypi evcc[30588]: [lp-1  ] DEBUG 2025/02/28 23:01:15 charger status: A
Feb 28 23:01:15 raspberrypi evcc[30588]: [lp-1  ] INFO 2025/02/28 23:01:15 car disconnected
Feb 28 23:01:15 raspberrypi evcc[30588]: [lp-1  ] DEBUG 2025/02/28 23:01:15 set battery boost: false
Feb 28 23:01:15 raspberrypi evcc[30588]: [lp-1  ] DEBUG 2025/02/28 23:01:15 set session soc limit: 0
Feb 28 23:01:15 raspberrypi evcc[30588]: [lp-1  ] DEBUG 2025/02/28 23:01:15 set session energy limit: 0
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.4

marleyan333, 2025-02-28:
Das logfile oben ist leider abgeschnitten, hier habe ich was zu solcast gefunden: 

Feb 28 23:25:38 raspberrypi evcc[31072]: [solcast] ERROR 2025/02/28 23:25:38 unexpected status: 401 (Unauthorized)
Feb 28 23:25:38 raspberrypi evcc[31072]: [main  ] ERROR 2025/02/28 23:25:38 creating tariff solar failed: cannot create tariff type 'template': cannot create tariff type 'solcast': unexpected status: 401 (Unauthorized)
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38 site config:
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38   meters:      grid ✓ pv ✓ battery ✓
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38     grid:      power ✓ energy ✗ currents ✗
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38     pv 1:      power ✓ energy ✗ currents ✗
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38   vehicles:
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38   tariffs:
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38     grid:      priceforecast
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38     feed-in:   pricestatic
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38     co2:       ✗
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] INFO 2025/02/28 23:25:38     solar:     TariffType(0)
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] INFO 2025/02/28 23:25:38 loadpoint 1:
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] INFO 2025/02/28 23:25:38   mode:        pv
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] INFO 2025/02/28 23:25:38   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] INFO 2025/02/28 23:25:38   meters:      charge ✓
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] INFO 2025/02/28 23:25:38     charge:    power ✓ energy ✗ currents ✓
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 set buffer start soc: 100
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 set battery discharge control: true
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 set residual power: 100
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 set smart cost limit: 0.2
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 phase timer inactive
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 pv timer inactive
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] INFO 2025/02/28 23:25:38 vehicle updated: unknown -> Taycan
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] WARN 2025/02/28 23:25:38 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 ----
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 charge power: 0W
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 charge currents: [0 0 0]A
Feb 28 23:25:38 raspberrypi evcc[31072]: [eebus ] DEBUG 2025/02/28 23:25:38 incoming connection request from bb74c3d74f3dcf4e839fda7e0b7b60d95acda3a3
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 pv 1 power: 0W
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 grid power: 4398W
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 battery 1 power: -6W
Feb 28 23:25:38 raspberrypi evcc[31072]: [eebus ] DEBUG 2025/02/28 23:25:38 ski bb74c3d74f3dcf4e839fda7e0b7b60d95acda3a3 connected
Feb 28 23:25:38 raspberrypi evcc[31072]: [eebus ] DEBUG 2025/02/28 23:25:38 Send:  read 1 NodeManagementDetailedDiscoveryData
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 battery 1 soc: 0%
Feb 28 23:25:38 raspberrypi evcc[31072]: [site  ] DEBUG 2025/02/28 23:25:38 site power: 4492W
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 charger status: A
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] INFO 2025/02/28 23:25:38 car disconnected
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 set battery boost: false
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 set session soc limit: 0
Feb 28 23:25:38 raspberrypi evcc[31072]: [lp-1  ] DEBUG 2025/02/28 23:25:38 set session energy limit: 0

marleyan333, 2025-02-28:
Ich glaube nicht nur ich wäre froh, wenn jemand von den Profis eine narrensichere Anleitung zum Thema PV-Vorhersage in die Dokumentation aufnehmen würde. Im Speziellen wie und wo man was konfigurieren muss.

VolkerK62, 2025-03-01:
Hast du bei `token` einen Link eingetragen?
Da musst du den Token einsetzen (copy & paste)

marleyan333, 2025-03-01:
Ja klar, der Link ist eingetragen.

OlliLu, 2025-03-01:
Anmelden, hier Deine. Api Key kopieren und unter Token eintragen. Speichern, neustarten, läuft. Die Begriffsbezeichnung Token ist etwas missverständlich.

https://toolkit.solcast.com.au/account/api-key

andig, 2025-03-01:
> 401 (Unauthorized)

Ist ja eindeutig. Was sagt

    evcc tariff

andig, 2025-03-01:
> Ich glaube nicht nur ich wäre froh, wenn jemand von den Profis eine narrensichere Anleitung zum Thema PV-Vorhersage in die Dokumentation aufnehmen würde. Im Speziellen wie und wo man was konfigurieren muss.

Nur Geduld, das Feature ist ja noch neu und wird weiter aufpoliert.

naltatis, 2025-03-01:
@marleyan333 ich hab deinen Code oben mal formatiert (` ``` ` davor und dahinter).

Sieht so aus als ob du die falschen Werte in die Felder geschrieben hast. Token ist der API Key und Site wird in der Solcast Oberfläche "Ressource ID" genannt. Siehe Kommentare in der Doku: https://docs.evcc.io/docs/tariffs#solcast

Ich mach das hier mal zu, sollte es das nicht sein schreib gerne noch mal.

closed 2025-03-01 by naltatis

marleyan333, 2025-03-01:
> Anmelden, hier Deinen Api Key kopieren und unter Token eintragen. Speichern, neustarten, läuft. Die Begriffsbezeichnung Token ist etwas missverständlich.
> 
> https://toolkit.solcast.com.au/account/api-key

Vielen Dank, das wusste ich nicht, habe da einfach die API-Adresse eingegeben.

marleyan333, 2025-03-01:
> > 401 (Unauthorized)
> 
> Ist ja eindeutig. Was sagt
> 
> ```
> evcc tariff
> ```

Sagt Folgendes:

[main  ] INFO 2025/03/01 15:16:20 evcc 0.200.5
[main  ] INFO 2025/03/01 15:16:20 using config file: /etc/evcc.yaml
[main  ] FATAL 2025/03/01 15:16:20 
Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked with no explicit database path.
Running the same config with multiple databases can lead to expiring vehicle tokens.

If you want to use the existing service database run the following command:

sudo evcc --database /var/lib/evcc/evcc.db

If you want to create a new user-space database run the following command:

evcc --database ~/.evcc/evcc.db

If you know what you're doing, you can skip the database check with the --ignore-db flag.

Auf diese Fehlermeldung bin ich auch schon mehrfach gestoßen. Ich verstehe das Problem nicht und die Lösungen nach denen ich hier gesucht habe,brachten mich nicht weiter...

marleyan333, 2025-03-01:
> [@marleyan333](https://github.com/marleyan333) ich hab deinen Code oben mal formatiert (`` ``` `` davor und dahinter).
>
> Sieht so aus als ob du die falschen Werte in die Felder geschrieben hast. [...]

Danke für das Umschreiben des Codes! abe das so in die Config UI eingetragen. Jetzt startet evcc zwar, aber die Vorhersage wird mir nach wie vor leider nicht angezeigt....langsam bin ich am Verzweifeln. Habe schon zigfach alles formatiert und neu installiert und trotzdem tritt immer irgend ein Problem auf.

marleyan333, 2025-03-01:
> Sagt Folgendes:
>
> [main ] INFO 2025/03/01 15:16:20 evcc 0.200.5 [main ] INFO 2025/03/01 15:16:20 using config file: /etc/evcc.yaml [main ] FATAL 2025/03/01 15:16:20 Found systemd service database at [...]

Die selbe Fehlermeldung erhalte ich auch bei "sudo systemctl status evcc" und den entsprechenden Befehlen für -meter, -vehicle, und -charger. Was hat das zu bedeuten und wie kann ich es beheben?

VolkerK62, 2025-03-01:
ergänze in der evcc.yaml diesen Eintrag
``` 
database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db
```
Bin mir nicht sicher, ob die "Position" wichtig ist.
Ich habe es gleich nach der network Sektion

marleyan333, 2025-03-01:
Danke @VolkerK62 , das Problem ist damit behoben, wieso werden diese Zeilen nicht automatisch eingefügt bei der Installation?

Bei evcc tariff kommt jetzt diese Fehlermeldung:

[main  ] INFO 2025/03/01 16:42:55 evcc 0.200.5
[main  ] INFO 2025/03/01 16:42:55 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/01 16:42:55 using sqlite database: /var/lib/evcc/evcc.db
[eebus ] INFO 2025/03/01 16:42:55 Local SKI: 7b9441b6df1709f6e258d3eaff3e79774ac5fec7
[eebus ] DEBUG 2025/03/01 16:42:55 starting websocket server on :4712
[eebus ] DEBUG 2025/03/01 16:42:55 mdns: announce
[eebus ] DEBUG 2025/03/01 16:42:55 mdns: using avahi
[eebus ] ERROR 2025/03/01 16:42:55 websocket server error: listen tcp :4712: bind: address already in use
[solcast] ERROR 2025/03/01 16:42:55 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/03/01 16:42:55 creating tariff solar failed: cannot create tariff type 'template': cannot create tariff type 'solcast': unexpected status: 429 (Too Many Requests)

VolkerK62, 2025-03-01:
Da hast du mittlerweile das Tageslimit der Anfragen (ich glaube 10 oder 12) überschritten.
Versuch es morgen wieder

pthauer, 2025-03-01:
Also ich habe den Database Eintrag nun auch in meiner ivcc.yaml eingefügt - auch nach network. Aber bei mir kommt immer noch der Fatal Error. Egal ob ich solcast oder forecast.solar konfiguriere. Noch irgendwelche Tipps

marleyan333, 2025-03-02:
> Da hast du mittlerweile das Tageslimit der Anfragen (ich glaube 10 oder 12) überschritten. Versuch es morgen wieder

Es geht aber leider auch heute nicht…