evcc-io/evcc #19287: Cupra Formentor Hybrid (Facelift) wird nicht erkannt

bsteimer, 2025-03-01:
### Describe the bug

Hallo,
ich habe heute meinen neuen Cupra Formentor (Facelift) versucht mit EVCC zu verbinden, doch leider scheitere ich daran. Mit dem alten Modell hat es mit der selben Konfiguration bis vor noch nicht all zu langer Zeit noch getan.

Wallbox ist eine Easee.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: seat
  title: Cupra Formentor
  user: b.*********@gmail.com
  password: ****
  capacity: 19.7
  mode: pv
  vin: V******************3
  name: ev4
```

### Log details

```text
[seat ] TRACE 2025/03/01 12:58:34 GET https://mal-3a.prd.eu.dp.vwg-connect.com/api/usermanagement/users/v2/users/****/vehicles
[seat ] TRACE 2025/03/01 12:58:34 {"userVehicles":{"userId":"****","vehicle": [{"homeRegionId":"ICTO-****","content":"***"}]}}
[seat ] TRACE 2025/03/01 12:58:34 GET https://mal-1a.prd.ece.vwg-connect.com/api/cs/vds/v1/vehicles/***/homeRegion
[seat ] TRACE 2025/03/01 12:58:34 {"homeRegion":{"baseUri":{"systemId":"ICTO-****","content":"https://mal-3a.prd.eu.dp.vwg-connect.com/api"}}}
[seat ] TRACE 2025/03/01 12:58:36 GET https://fal-3a.prd.eu.dp.vwg-connect.com/fs-car/bs/batterycharge/v1/VW/ES/vehicles/***/charger
[seat ] TRACE 2025/03/01 12:58:36 {"error":{"errorCode":"batterycharge.bs.internal","description":"technical error"}}
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.4

zachelnet, 2025-03-01:
@bsteimer nimm bitte mal das Cupra Template, 

grüße Martin

andig, 2025-03-01:
@zachelnet danke für den Hinweis. Im Gegensatz zu Skoda haben wir bei Seat noch beide Templates. Das Seat Tempalte ist als Gen3 ausgewiesen: https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/seat.yaml#L9. Wenn das jetzt funktioniert müssten wir den Formentor wohl auch bei Gen4 (cupra) hinzu fügen?

bsteimer, 2025-03-01:
Vielen Dank @zachelnet und @andig ihr seid die Besten! Das hat funktioniert, ich hatte das zwar vorher auch schon probiert, aber ich musste ihn tatsächlich neu anstecken damit aus dem Gastfahrzeug der Formentor wird, daher dachte ich das wäre nicht die Lösung. Aber es ist die Lösung.

closed 2025-03-01 by bsteimer