evcc-io/evcc #19293: Experimental UI feature - Reconfiguration/Deletion of charger not possible

paasherbert, 2025-03-01:
### Describe the bug

i added a Sungrow AC011E-01 Charger. I have a Waveshare RS485 to RJ45 Ethernet Adapter. Therefore i tried this configuration:
 
Modbus: 248
IP Adress: 192.168.4.34 (this is the correct Address of the Adapter)
Port: 502
Modbus Protocol: RTU  

The validate check return a error. 
Then i wanted to try with Protocol TCP, but it is not possible.

cannot create charger type 'template': cannot create charger type 'sungrow': connection already registered with different protocol: 192.168.4.34:502

There is also no button/possibility to delete wrong/not working charger. No button. I did not found a list where i manually can see/delete the charger.

The evcc.yaml is still empty.

### Steps to reproduce

1.Add a charger
2.Validation fails
3.Change configuration
4. Error message "cannot create charger type 'sungrow': connection already registered with different protocol"

### Configuration details

```yaml
no
```

### Log details

```text
nothing was traced
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-03-01:
Missing version

paasherbert, 2025-03-01:
Current version: 0.200.5

paasherbert, 2025-03-01:
issue is closed? Do i need to create a new issue now?

andig, 2025-03-01:
What happens here is that after a failed test, the connection is already created in the in connection pool and not removed again. We'll need to find a way to clear out stale connections. Not an urgent issue- you can finish Config after restart.

closed 2025-03-01 by andig