evcc-io/evcc #19295: UI shows charging from grid (Netzladen) after changing tariff from dynamic to fixed

dl6nm, 2025-03-01:
### Describe the bug

I have changed my tariff from a dynamic one (Tibber) to a fixed one. The UI shows, that it will charge the battery, if the price is below 32,4 ct. There is no way in the UI to change this behavior, as it was with the Tibber tariff.

[image]

### Steps to reproduce

1. Expect having a dynamic tariff (e.g. Tibber)
2. Change the price limit for charging the PV battery to an value like "32,4 ct"
3. In the config change the tariff from Tibber to a fixed one, e.g. to 35 ct

### Configuration details

```yaml
tariffs:
  ##
  ## Tibber
  ##
  #grid:
  #  type: template
  #  template: tibber
  #  token: MY_TOKEN

  #
  # Stromanbieter
  #
  grid:
    type: fixed
    price: 0.3532 # EUR/kWh (default)
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.5

dl6nm, 2025-03-01:
In addition to this, the same problem applies also the the vehicle.

detlefheese, 2025-03-02:
[image]

Ich habe ab morgen früh auch einen Zeitzonenbasierten Tarif. Ich habe Übergangsweise einen Fixtarif in die YAML geschrieben. 
Bei mir also der selbe Effekt ... Variable Tibber-Daten sind noch zu sehen.

bastian1607, 2025-03-02:
Tatsächlich dachte ich, dass es eine Vorbereitung wäre für eine Möglichkeit den Preis durch PV Forecast zu drücken.
Zudem habe ich 5 Stunden am Tag einen vergünstigten Tarif. Also die Möglichkeit nutze ich fürs Auto schon. 

Was ich sagen will: hier müsste es eine Unterscheidung zwischen fix, fix mit günstigen Zonen und flex/variabel geben.

naltatis, 2025-03-02:
@dl6nm The price limits (load points, battery) are currently not removed when switching from a dynamic/zoned to a fixed-price tariff.

Workaround 1:

Delete limits via REST API (see [docs](https://docs.evcc.io/docs/integrations/rest-api))

```sh
# specific loadpoint
DELETE /loadpoints/{id}/smartcostlimit
# all load points at once
DELETE /smartcostlimit
# battery 
DELETE /batterygridchargelimit
```

Workaround 2:

Delete limits via UI before switching to static. You can also temporary switch back to a static+zones tariff, delete the limits, and remove the zones.

naltatis, 2025-03-02:
> Was ich sagen will: hier müsste es eine Unterscheidung zwischen fix, fix mit günstigen Zonen und flex/variabel geben.

@bastian1607 Den Punkt verstehe ich noch nicht. Was das Preislimit angeht behandelt wir Fixed+Zones und Dynamisch aktuell genau gleich. Bei beiden Fällen wird ein Limit angeboten. Nur wenn es keinen Tarif gibt oder diese keine Dynamik hat (fixed ohne zones) gibt es diese Option nicht.

naltatis, 2025-03-02:
@detlefheese das was du beschreibst sie nach einem anderen Problem aus. Wenn deine Tibber Daten noch zu sehen sind ist auch irgendwo Tibber noch konfiguriert.

detlefheese, 2025-03-02:
> [@detlefheese](https://github.com/detlefheese) das was du beschreibst sie nach einem anderen Problem aus. Wenn deine Tibber Daten noch zu sehen sind ist auch irgendwo Tibber noch konfiguriert.

ich habe in der Konfigurationseinstellungen auch noch dieses drin ... aber nicht in YAML ... 

[image]

@naltatis Muss ich die YAML leer machen und mal alles in die Datenbank packen? Oder was ist der sinnvolle Weg?

naltatis, 2025-03-02:
@detlefheese Ja, sollte auf jeden fall nicht an zwei Stellen konfiguriert sein. Entweder im UI oder in der evcc.yaml. Bei älteren Versionen wurden die Einstellungen "gemerged". In der aktuellen Version ist es so, dass der `tariffs` Wert aus der evcc.yaml ignoriert wird, sobald Tarife in der UI gepflegt sind.

bastian1607, 2025-03-02:
> Bei beiden Fällen wird ein Limit angeboten. Nur wenn es keinen Tarif gibt oder diese keine Dynamik hat (fixed ohne zones) gibt es diese Option nicht.

Dann gibt es ja eine Trennung. Darauf wollte ich hinaus. Ich bin ja nur von variabel auf fix mit Zonen umgestiegen.

naltatis, 2025-03-02:
Ich sehe gerade keinen eleganten Weg im Code, wie wir dieses Verhalten automatisiert lösen können. Also bei Wechsel von Dynamisch auf Statisch, oder CO2 auf Preis die am LP/Site gespeicherten Limits entweder automatisch löschen oder außer Kraft zu setzen. Sobald wir rein in Config UI sind könnten wir das zum Zeitpunkt des Anlegen/Ändern/Löschen eines Tarifs prüfen und durchführen.

Für den Übergang würde ich daher eine Löschen Option via UI anbieten. Heißt da wo vorher der Auswahldialog und das Diagramm war würde eine Meldung mit Button zum Limit löschen erscheinen. Nicht super elegant, aber besser als die Workarounds oben.

closed 2025-03-03 by andig