evcc-io/evcc #19306: Lastmanagement: Einphasiges Laden auf verschiedenen Phasen

TheMr-S, 2025-03-02:
Moin,
folgende Problematik hintert mich aktuell daran, Lastmanagement richtig verwenden zu können:

Wir haben zwei Wallboxen (GoE), eine davon ist einphasig, die andere dreiphasig angeschlossen. Beide hängen am selben circuit mit 16A Absicherung pro Phase.

Damit beide Säulen zeitweise wenigstens einphasig gleichzeitig laden können, hängt die einphasige WB an Phase 2. Wenn ich das richtig sehe, kann ich diesen Fakt in der Konfiguration bei EVCC nicht abbilden - oder sehe ich das falsch? Man würde also erwarten, dass EVCC aktuell das einphasige gleichzeitig Laden beider WBs mit Volldampf nicht zulassen würde, wenn man für den circuit 16A als max einstellt.

Ein weiterer Feature-Wunsch (aber das könnte ggf. auch zu speziell sein) wäre, dass die dreiphasige WB auch selbst erkennt, dass Phase 2 "blockiert" ist (die einphasige WB hat bei uns höhere Prio) und entsprechend nur einphasig lädt, auch wenn sie auf "automatischer Wechsel" steht.

Vielen Dank und LG!

premultiply, 2025-03-02:
Wenn du den circuit (nur) auf Leistung beschränkst könnte/sollte das klappen wenn die Boxen entsprechend phasenrotiert angeschlossen sind.

TheMr-S, 2025-03-02:
Auf welche Leistung muss ich denn beschränken? 3,6 oder 7,2 oder ganz was anderes? Alleiniges 11kW-laden der einen WB soll weiterhin möglich sein, wenn der andere Ladepunkt nicht verwendet wird.

andig, 2025-03-02:
Leider ist Umwandlung in Discussion auf Github kaputt.

closed 2025-03-02 by andig