evcc-io/evcc #19307: Smart charge on dynamic tariffs

timpetersson81, 2025-03-02:
Awesome job guys! I just love EVCC!

Does EVCC work in such a way that it compares the prices of electricity now and the amount that the sun produces, but the prices until the charging plan is over, and then charging starts if the sun does not produce the full amount of electricity, so that it is only solar charging, but that it is actually cheaper to utilize the solar energy that is available right now than it is to have to buy the electricity later at a different price?

Ex. Price for electricity the current hour is 1, and minimum charge is 4kwh
Cheapste hours are the price 0.5
When sun produces 3kwh pr hour the saved amount in this hour is 3, and payed price is 1. if i charged only from grid later, the price would be 2. So charging now is cheaper that waiting for the cheapest hour.

This would in my cases require an input of export price API input, as it is not the same as import, but allso dynamic. So it allso compares the price for export when calculating the smart thing to do :)

andig, 2025-03-02:
Please use discussions for questions.

andig, 2025-03-02:
Leider ist Umwandlung in Discussion auf Github kaputt

closed 2025-03-02 by andig