evcc-io/evcc #19308: Total price in API

timpetersson81, 2025-03-02:
I would love to have the totalt charge price in the API :)

andig, 2025-03-02:
Unclear what youre asking

naltatis, 2025-03-02:
The `/api/state` API offers everything you need. It's the bases for what we show in the UI. Session price are here:
https://demo.evcc.io/api/state?jq=.loadpoints[0].sessionPrice

timpetersson81, 2025-03-02:
> The `/api/state` API offers everything you need. It's the bases for what we show in the UI. Session price are here: [https://demo.evcc.io/api/state?jq=.loadpoints[0].sessionPrice](https://demo.evcc.io/api/state?jq=.loadpoints%5B0%5D.sessionPrice)

I was asking for a total price of all sessions, like a sum og price in tabel sessions (I don't know the structure of the database). I can do something with sessions, but I have to call the API 12 times for each month to get the data.

naltatis, 2025-03-02:
@timpetersson81 you can call the api once without year/month filter to get all data at once. We dont offer precomputed aggregates right now. You'd have tool loop through the records in your program.

timpetersson81, 2025-03-03:
> [@timpetersson81](https://github.com/timpetersson81) you can call the api once without year/month filter to get all data at once. We dont offer precomputed aggregates right now. You'd have tool loop through the records in your program.

Thank you, I got that working :D I wonder if there is a price on household usage in both kwh and price somewhere in the API ? I have looked around, and can seem to find that :)