evcc-io/evcc #19309: Charge from grid in API

timpetersson81, 2025-03-02:
I would love to have the total charge from grid in the API, where the charge from PV is not in the calculation. Currently the only input I see from the sun is in %, but being on dynamic tariff makes that a hard caculation (Im on home assistant, an see the input there)

andig, 2025-03-02:
Unclear what you‘re asking

naltatis, 2025-03-02:
The api exports the energy and the solar percentage of the current session. Should be straight forward to calculate the grid energy from there. Or what do I miss?

closed 2025-03-02 by andig

timpetersson81, 2025-03-02:
> The api exports the energy and the solar percentage of the current session. Should be straight forward to calculate the grid energy from there. Or what do I miss?

I have dynamic energy prices, so they are not the same from hour to hour, so I don't se how I can calculate that from a totalt number og charged kwh, and a procent of that. Perhaps im missing something