evcc-io/evcc #19326: ID3 nicht ladbar über Min + PV

xnitro1, 2025-03-02:
### Describe the bug

Hallo Zusammen,

Ich habe EVCC schon länger laufen, habe jedoch heute mal wieder das Update auf die neueste Version gemacht ( v0.200.5 ).
Am Anfang lief es recht holprig, dass mein ID3 nicht  erkannt wurde. Nach ein paar mal Ladestecker rein und wieder raus sowie EVCC Reboot wurde das Auto dann erkannt.

Jedoch kann ich nicht über Min + PV laden. Der Status des Fahrzeuges ist die ganze Zeit : 

Ladebereit. Warte auf Fahrzeug …

Jedoch passiert nichts. Wenn ich auf "Schnell" stelle, dann lädt er mit den vollen 11 KW, jedoch wird es auch nicht als Ladepunkt erkannt, sondern als "Verbrauch".

### Steps to reproduce

1. Ladestecker rein
2. Auto lädt nicht im Min + PV Modus
3. Modus auf "Schnell" stellen. Auto lädt, jedoch als Haushaltsverbrauch erkant
4. Umstelle nauf Min + PV Modus, Laden wird beendet
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: bea18ab33ece5160475629290a537e13ae6a08f0f859b0ab8b9efb893b906c75

interval: 30s # control cycle interval

meters:

- type: template
  template: growatt-hybrid
  id: 1
  usage: grid
  modbus: tcpip
  host: 192.168.50.5
  port: 5021
  name: grid1

- type: template
  template: growatt-hybrid
  id: 1
  usage: pv
  modbus: tcpip
  host: 192.168.50.5
  port: 5021
  name: pv2

- type: template
  template: growatt-hybrid
  id: 1
  usage: battery
  modbus: tcpip
  host: 192.168.50.5
  port: 5021
  capacity: 6.9
  name: battery3

chargers:
[... 571 lines cut ...]
[site  ] DEBUG 2025/03/02 12:05:17 battery 1 soc: 50%
[site  ] DEBUG 2025/03/02 12:05:17 site power: 5610W
[lp-1  ] DEBUG 2025/03/02 12:05:17 charger status: B
[eebus ] DEBUG 2025/03/02 12:05:19 Send: d:_i:47859_Elli-Wallbox-2137A1MUJ4 notify 898 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/03/02 12:05:21 Send: d:_i:47859_Elli-Wallbox-2137A1MUJ4 notify 899 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/03/02 12:05:23 Send: d:_i:47859_Elli-Wallbox-2137A1MUJ4 notify 900 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/03/02 12:05:25 Send: d:_i:47859_Elli-Wallbox-2137A1MUJ4 notify 901 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/03/02 12:05:27 Send: d:_i:47859_Elli-Wallbox-2137A1MUJ4 notify 902 DeviceDiagnosisHeartbeatData
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

 v0.200.5

xnitro1, 2025-03-02:
> Gehts wenn du auf alter Version zurück gehst? Welche?

Vielen Dank für die Rückmeldung erstmal.

Ich wollte gerade deinen Ratschlag versuchen, habe aber vorher bestimmt jetzt zum 10. mal den Rasperry mit EVCC neugestartet und wirklich jetzt gerade scheint es wieder zu funktionieren. Ich habe jedoch keinerlei Änderungen an irgendwas vorgenommen, daher kann ich mir auch nicht erklären warum es die letzten 2 Stunden nicht wollte und jetzt urplötzlich doch wieder.

Ich hoffe es bleibt jetzt auch so. Dennoch vielen Dank für die Rückmeldung und sorry für die Zeitverschwendung :(

closed 2025-03-02 by xnitro1

andig, 2025-03-02:
Die Elli bleibt leider Elli...

StefanSchoof, 2025-03-02:
Bei der Elli nach einem Update auch gerne auf der Wallbox Seite einmal kontrollieren, ob die eine Verbindung zu evcc anzeigt und wenn nicht hilft da ein Reboot der Wallbox über die VW App.

xnitro1, 2025-03-14:
> Die Elli bleibt leider Elli...

Hallo Andig, kannst du bitte den gesamten Thread löschen? Leider habe ich übersehen, dass in der Konfiguration mein Password steht...Danke an denjenigen der damit gestern Abend Schindluder betrieben hat. Password ist geändert, aber bitte trotzdem den Thread löschen aufgrund der Versionhistory