evcc-io/evcc #19339: Smart charge from grid and PV for optimal price

timpetersson81, 2025-03-02:
Look at both export price, import price and production to calculate the cheapest way to reach the plan target.

Explained with some graphics
[image]
[image]

naltatis, 2025-03-02:
I've trouble understanding exactly what's going on in the graphic. Can you please add more context on whats going on?
Are there times where it's cheaper to pull electrify from the grid in Denmark than using it directly? Do you propose a change to the PV mode or just for the planning algorithm? Currently the planer does not take PV production into account at all. This might change with upcoming forecast features.

timpetersson81, 2025-03-02:
No, it is never cheeper to use the grid, but evcc first starts charging when the full pv Production is avalibel, and my cars want 4kwh. So when PV produces 3,9 I dont charge the car, but export to grid instead.

When i export to grid, and the buy it later to charge the car, because the plan requires it, then I Got less for the power, than I have to pay buying it back, due to transport fees and fees to the goverment.

That means it is sometimes cheaper to buy the missing kwh from grid istead off not charging.

If I charge to min+sun, then it just run at minimum, also into peak hours.

Is was Thinking it could be part of the sun, as this allready supports the plan, but it could allso be as new button. 

I hope it makes sence?

naltatis, 2025-03-03:
Ok understood. So it's mainly about a smarter enable/disable behavior based on the real prices and not only time/threshold based as right now. I've added this aspect to our planner improvement collection issue.

closed 2025-03-03 by naltatis