evcc-io/evcc #19342: Hyundai Bluelink support in the US

marcusb, 2025-03-02:
### Describe the bug

evcc is unable to connect to the my Bluelink account. I believe this may be because I am in the United States, and Bluelink has different APIs in the US and EU. I am able to connect to Bluelink with [hyundai_kia_connect_api](https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api).

The API endpoint for US is `api.owners.kia.com`, and not `prod.eu-ccapi.hyundai.com`.

### Steps to reproduce

1. Configure a Huyndai vehicle with a US-based Bluelink account
2. Run `evcc vehicle --log trace,db:error`

### Configuration details

```yaml
vehicles:
- type: template
  template: hyundai 
  user: xxx@yyy
  password: ***
  vin: ***
  title: My Hyundai
  icon: car  
  capacity: 77.4  
  phases: 1  
  mode: now  
  minCurrent: 32  
  maxCurrent: 32  
  cache: 5m  
  language: en  
  name: ev1
```

### Log details

```text
[main  ] INFO 2025/03/02 17:12:27 evcc 0.200.5                                                                                                                                                                                 
[main  ] INFO 2025/03/02 17:12:27 using config file: /app/evcc.yaml                                                                                                                                                            [hyundai] TRACE 2025/03/02 17:12:27 POST https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/notifications/register                                                                                                               
[hyundai] TRACE 2025/03/02 17:12:28 {"pushRegId":"999D0655AF022F5FD0907E489094B9A9FD12C9C6708350E0D82EFD25B3E792F4","pushType":"GCM","uuid":"ae652427-3b2d-419f-8e0e-8038f8aa1f70"}                                            --                                                                                                                                                                                                                             
{"retCode":"S","resCode":"0000","resMsg":{"deviceId":"baf3328a-5605-4e75-a471-6d7ae169d8a6"},"msgId":"ee686bb0-f780-11ef-8714-0bbf412dc7c2"}                                                                                   
[hyundai] TRACE 2025/03/02 17:12:28 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/user/oauth2/authorize?response_type=code&state=test&client_id=6d477c38-3ca4-4cf3-9557-2a1929a94654&redirect_uri=https://prd.eu-ccapi.hyund
ai.com:8080/api/v1/user/oauth2/redirect                                                                                                                                                                                        
[hyundai] TRACE 2025/03/02 17:12:29 <a href="https://prd.eu-ccapi.hyundai.com:8080/web/v1/user/authorize?cache=reset">Found</a>.                                                                                               
[hyundai] TRACE 2025/03/02 17:12:29 GET https://prd.eu-ccapi.hyundai.com:8080/web/v1/user/authorize?cache=reset 
[hyundai] TRACE 2025/03/02 17:12:29 <!doctype html><html lang="en"><head><meta charset="utf-8"><meta name="viewport" content="width=device-width,initial-scale=1,shrink-to-fit=no,maximum-scale=1,user-scalable=0"><meta name="
theme-color" content="#ffffff"><link href="https://fonts.googleapis.com/css?family=Nanum+Gothic:400,700,800&amp;subset=korean" rel="stylesheet"><link href="https://fonts.googleapis.com/icon?family=Material+Icons" rel="style
sheet"><link rel="manifest" href="/web/v1/user/manifest.json"><link rel="icon" href="/web/v1/user/favicon_hyundai.ico" type="image/x-icon"><!--[if lt IE 10]><script src="/web/v1/user/fetch.js"></script><![endif]--><title></
title><script defer="defer" src="/web/v1/user/static/js/main.5c84d3e1.js"></script><link href="/web/v1/user/static/css/main.7c0207e2.css" rel="stylesheet"></head><body><noscript>You need to enable JavaScript to run this app
...
25-b9f1-13fd5d3f3cfa&amp;kc_locale=nl">Nederlands</a></li>
                                <li class="kc-dropdown-item"><a href="/auth/realms/euhyundaiidm/login-
[hyundai] TRACE 2025/03/02 17:12:30 POST https://eu-account.hyundai.com/auth/realms/euhyundaiidm/login-actions/authenticate?session_code=bkwmSeY36miDvx-0zITt5hYKxeCzC3T-YBW4-jOO8vE&execution=f09c7a3a-cb33-4625-b9f1-13fd5d3f
3cfa&client_id=64621b96-0f0d-11ec-82a8-0242ac130003&tab_id=OTrt4AyYnPM
[hyundai] TRACE 2025/03/02 17:12:31 credentialId=&password=***&rememberMe=on&username=***
--
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"  "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
<html xmlns="http://www.w3.org/1999/xhtml" class="login-pf">

<head>
...
25-b9f1-13fd5d3f3cfa&amp;kc_locale=hu">Magyar</a></li>
                                <li class="kc-dropdown-item"><a href="/auth
[hyundai] TRACE 2025/03/02 17:12:31 POST https://prd.eu-ccapi.hyundai.com:8080/api/v1/user/signin
[hyundai] TRACE 2025/03/02 17:12:31 {"email":"***","password":"***"}
--
{"errId":"df69f59c-a691-40f5-ac45-2bf6fbef3b9f","errCode":"4003","errMsg":"Invalid values"}
[main  ] ERROR 2025/03/02 17:12:31 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'hyundai': login failed: unexpected status: 400 (Bad Request): Invalid values (4003)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.200.5 (3ade50fc)

andig, 2025-03-03:
Please send testing credentials to info@evcc.io

marcusb, 2025-03-04:
> Please send testing credentials to [info@evcc.io](mailto:info@evcc.io)

That would be a bit of a privacy problem, but I may try to hack something when I get some spare cycles. Looking at the codebase, I think it will be best to create a new `bluelink_us` module, because the APIs seem to be entirely different.

andig, 2025-03-04:
Great thank you! Lets close here as there‘s nothing the core team can so right now.

closed 2025-03-04 by andig

marcusb, 2025-12-18:
@andig This PR is ready for review.