evcc-io/evcc #19344: openWB series2 Duo über experimentelle GUI einbinden

Ion3, 2025-03-02:
### Describe the bug

Salü

Ich habe über den YouTube-Kanal [smart home & more](https://youtu.be/KSUfEhRCW3Y?si=uK-k_4P3ax3qeO_D) von eurer neuen GUI erfahren und dachte, ich probiere es mal aus. Ich besitze eine ([selbst gebaute](https://forum.openwb.de/viewtopic.php?p=22540#p22540)) openWB series2 Duo mit zwei Ladepunkten, die über einen Raspi angesteuert werden. Das funktioniert auch seit 4 Jahren ohne Probleme. Nun habe ich beim Wallbox hinzufügen folgenden Status erhalten:

phaseCurrents: outdated

Ich habe es unter den eingeblendeten erweiterten Einstellungen sowohl mit 1 als auch mit 2 Ladepunkten versucht.

openWB läuft in der aktuellen Version 1.9.303.0 (einen Update auf Software 2.x hab ich mich bisher nicht getraut. Never touch a running system.)

Das evcc Add-on in Home Assistant habe ich in den Versionen 0.200.4 und 0.200.5 ausprobiert.

Im Protokoll des Add-ons konnte ich auch nichts hilfreiches finden. Hier die letzten zwei Zeilen:

`[site  ] WARN 2025/03/02 17:14:31 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[openwb] INFO 2025/03/02 17:20:15 connecting evcc-705177188 at tcp://10.117.40.21:1883`

Könnt ihr mir sagen, wo hier das Problem ist?

Herzlichen Dank,
Timo

### Steps to reproduce

1. Home Assistant Add-on evcc starten
2. Konfiguration beginnen
3. Experimentell Modus "an"
4. Ladepunkt hinzufügen auswählen
5. Titel hinzufügen "BMW i3"
6. Hersteller auswählen "openWB series2"
7. IP-Adresse eintragen
8. Erweiterte Einstellungen anzeigen oder auch nicht (gleicher Effekt)
9. Ladepunkt 1 oder 2 auswählen (gleicher Effekt)
10. Auf prüfen klicken
11. Status: "fehlgeschlagen" Fehlermeldung: "phaseCurrents: outdated"

### Configuration details

```yaml
GUI used.
```

### Log details

```text
[site  ] WARN 2025/03/02 17:14:31 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[openwb] INFO 2025/03/02 17:20:15 connecting evcc-705177188 at tcp://10.117.40.21:1883
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

naltatis, 2025-03-02:
This should be fixed with the just merge https://github.com/evcc-io/evcc/pull/19267.
Please try the nightly tomorrow. If this does not work feel free to comment and we'll reopen.
\\cc @andig

closed 2025-03-02 by naltatis

andig, 2025-03-02:
Mir ist noch unklar was hier passiert und warum das nicht gehen sollte. Es ist auch nicht klar, warum ein fehlender Messwert überhaupt ein Problem sein sollte. (@naltatis es sei denn man könne aufgrund dessen nicht speichern…)

Ion3, 2025-03-09:
I am on Versionen 0.200.8 of the Home Assistant Addon and still have the same issue:

`Status: fehlgeschlagen

phaseCurrents: outdated`

Ion3, 2025-03-17:
Closed as duplicate of
[#19229](https://github.com/evcc-io/evcc/issues/19229) kann ich nicht nachvollziehen. Die Fehlermeldung und das Verhalten in [#19229](https://github.com/evcc-io/evcc/issues/19229) ist doch ein völlig Anderes.

Ion3, 2025-03-20:
Hallo Andi,

Ich bin jetzt auf Version 0.201.1. Die Fehlermeldung ist immer noch die gleiche:

```
Status: "fehlgeschlagen" 
Fehlermeldung: "phaseCurrents: outdated"
```

Die Fehlermeldung ist wie das Log leider nicht sehr aussagekräftig:

```
[main  ] INFO 2025/03/18 14:44:16 new version available: 0.201.0
[main  ] INFO 2025/03/18 20:44:16 new version available: 0.201.0
[main  ] INFO 2025/03/19 02:44:16 new version available: 0.201.0
[main  ] INFO 2025/03/19 08:44:16 new version available: 0.201.0
[main  ] INFO 2025/03/19 14:44:16 new version available: 0.201.0
[main  ] INFO 2025/03/19 20:44:16 new version available: 0.201.0
[main  ] INFO 2025/03/20 02:44:16 new version available: 0.201.0
[main  ] INFO 2025/03/20 08:44:16 new version available: 0.201.1
[main  ] INFO 2025/03/20 14:44:16 new version available: 0.201.1
[main  ] INFO 2025/03/20 20:44:16 new version available: 0.201.1
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/20 21:37:11 evcc 0.200.8
[main  ] INFO 2025/03/20 21:37:11 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/20 21:37:11 using sqlite database: /data/evcc.db
[main  ] INFO 2025/03/20 21:37:13 listening at :7070
[site  ] INFO 2025/03/20 21:37:13 site config:
[site  ] INFO 2025/03/20 21:37:13   meters:      grid ✗ pv ✗ battery ✗
[site  ] INFO 2025/03/20 21:37:13   tariffs:
[site  ] INFO 2025/03/20 21:37:13     grid:      ✗
[site  ] INFO 2025/03/20 21:37:13     feed-in:   ✗
[site  ] INFO 2025/03/20 21:37:13     co2:       ✗
[site  ] INFO 2025/03/20 21:37:13     solar:     solar
[main  ] INFO 2025/03/20 21:37:22 new version available: 0.201.1
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/03/20 21:49:40 evcc 0.201.1
[main  ] INFO 2025/03/20 21:49:40 using config file: /config/evcc.yaml
[db    ] INFO 2025/03/20 21:49:40 using sqlite database: /data/evcc.db
[main  ] INFO 2025/03/20 21:49:47 listening at :7070
[site  ] INFO 2025/03/20 21:49:47 site config:
[site  ] INFO 2025/03/20 21:49:47   meters:      grid ✗ pv ✗ battery ✗
[site  ] INFO 2025/03/20 21:49:47   tariffs:
[site  ] INFO 2025/03/20 21:49:47     grid:      ✗
[site  ] INFO 2025/03/20 21:49:47     feed-in:   ✗
[site  ] INFO 2025/03/20 21:49:47     co2:       ✗
[site  ] INFO 2025/03/20 21:49:47     solar:     ✓
[openwb] INFO 2025/03/20 21:54:31 connecting evcc-745064558 at tcp://10.117.40.21:1883
```

Wo ist denn vielleicht geschrieben, was "_phaseCurrents: outdated_" bedeutet?

Danke für deine Hilfe.

Liebe Grüsse,
Timo

andig, 2025-03-20:
> https://github.com/evcc-io/evcc/issues/19344#issuecomment-2731831002

Ion3, 2025-03-20:
???

Ion3, 2025-03-20:
Ich habe keine Möglichkeit gefunden ein Nightly über Home Assistant zu installieren.

Ion3, 2025-03-20:
Hilft das?

```
[main ] INFO 2025/03/20 21:49:40 evcc 0.201.1
[main ] INFO 2025/03/20 21:49:40 using config file: /config/evcc.yaml
[db ] INFO 2025/03/20 21:49:40 using sqlite database: /data/evcc.db
[main ] INFO 2025/03/20 21:49:47 listening at :7070
[site ] INFO 2025/03/20 21:49:47 site config:
[site ] INFO 2025/03/20 21:49:47 meters: grid ✗ pv ✗ battery ✗
[site ] INFO 2025/03/20 21:49:47 tariffs:
[site ] INFO 2025/03/20 21:49:47 grid: ✗
[site ] INFO 2025/03/20 21:49:47 feed-in: ✗
[site ] INFO 2025/03/20 21:49:47 co2: ✗
[site ] INFO 2025/03/20 21:49:47 solar: ✓
[openwb] INFO 2025/03/20 21:54:31 connecting evcc-745064558 at tcp://10.117.40.21:1883
[openwb] DEBUG 2025/03/20 21:54:31 tcp://10.117.40.21:1883 connected
[openwb] DEBUG 2025/03/20 21:54:31 tcp://10.117.40.21:1883 subscribe openWB/system/Timestamp
```