evcc-io/evcc #19345: AC Thor: Hinzufügen von Limit-Temperatur

docolli, 2025-03-02:
**Is your feature request related to a problem? Please describe.**
Ich habe heute lange gebraucht zu verstehen, warum mein AC Thor gesteuerter Heizstab (über evcc) nicht lädt. Bis ich gesehen habe, dass die WP meinen Puffer bereits vorher auf 60°C erwärmt hatte und in den Einstellungen des AC THor ebenfalls (dummerweise) als Limit 60°C eingestellt sind.

[image]

[image]

**Describe the solution you'd like**
Könnte man (analog zum SOC Limit, welches IM angeschlossenen Autos eingestellt ist) das im AC Thor eingestellte Temperaturlimit im Ladepunkt auch im Balken (und als Text) mit anzeigen? Also so wie hier der Balken und Text klar zeigt, das mehr als 90% nicht gehen, auch wenn ich hier Ladung bis 95% gewählt habe.

[image]

Der Wert lässt sich über das Modbus Register 1002 auslesen und wenn ich den Code von mypv.go richtige lese, ist bereits eine Funktion eingebaut, die diesen Wert auslesen sollte: https://github.com/evcc-io/evcc/blob/3ade50fcd1c84f531801df24b9c1558652a15b57/charger/mypv.go#L253

[image]

Im Trace Log ist aber keine Abfrage von Register 1002 zu finden. Scheint also entweder nicht zu funktionieren, oder die Abfrage erfolgt aus anderen Gründen nicht.

andig, 2025-03-03:
Könntest du bitte info@evcc.io mal Remotezugang einrichten? Ssh key siehe mein GH Profil. Alternativ Vpn.

andig, 2025-03-03:
...und bitte mal

    evcc charger

andig, 2025-03-03:
Limit can only be read from, but not written to, charger.

docolli, 2025-03-03:
Ich will ja nur, dass evcc das Limit vom AC Thor liest und anzeigt. Analog wie vom Auto das SOC Limit gelesen und angezeigt wird. 😉 Schreiben muss nicht sein!

Mail / Remotezugang bzw. Ausgabe von `evcc charger` kommt heute Abend.

docolli, 2025-03-03:
> ...und bitte mal
> 
> ```
> evcc charger
> ```

Here it is:

```sh
systemctl stop evcc
evcc -c /etc/evcc.yaml charger

[main  ] INFO 2025/03/03 17:42:52 evcc 0.200.5
[main  ] INFO 2025/03/03 17:42:52 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/03 17:42:52 using sqlite database: /var/lib/evcc/evcc.db
[mqtt  ] INFO 2025/03/03 17:42:52 connecting 653555047 at tcp://192.168.1.119:1883
myPV_ac-thor
------------
Power:         0W
Temp:          69°C
Charge status: B
Enabled:       false
Max Temp:      70°C
Features:      [IntegratedDevice Heating]
```

Max Temp korreliert mit dem eigestellten Wert. Hochgestellt auf 75°C:

[image]

```sh
systemctl stop evcc
evcc -c /etc/evcc.yaml charger

[main  ] INFO 2025/03/03 17:48:34 evcc 0.200.5
[main  ] INFO 2025/03/03 17:48:34 using config file: /etc/evcc.yaml
[db    ] INFO 2025/03/03 17:48:34 using sqlite database: /var/lib/evcc/evcc.db
[mqtt  ] INFO 2025/03/03 17:48:35 connecting 653555047 at tcp://192.168.1.119:1883
myPV_ac-thor
------------
Power:         0W
Temp:          69°C
Charge status: B
Enabled:       false
Max Temp:      75°C
Features:      [IntegratedDevice Heating]

```

Abfrage funktioniert grundsätzlich schon, aber wann frägt evcc das SOC Limit vom Auto ab? Vermutlich einmalig nach dem Erkennen des Einsteckens. Das funktioniert hier aber leider nicht, der Heizstab ist ja immer eingesteckt.

andig, 2025-03-03:
Da ist es doch?

> Max Temp:      75°C

Was willst Du denn noch? Ich verstehe Deine Frage nicht...

 > aber wann frägt evcc das SOC Limit vom Auto ab

Bei einem integrateddevice gar nicht- kann ja kein Auto dran hängen.

docolli, 2025-03-03:
Ja, aber der Wert wird nicht weiter verwendet und angezeigt wie hier im Beispiel der LimitSOC 90% vom Tesla:

[image]

Ist vielleicht auch eher eine Frage an @naltatis ?

andig, 2025-03-03:
Mhhm, das sollte er /cc @naltatis

docolli, 2025-03-03:
Das könnte dann so aussehen (kurz Affinity Photo genutzt), Beispiel mit 80°C Limit und 85°C in evcc eingestellt:

[image]

docolli, 2025-03-06:
Eben 0.200.6 getestet, sieht gut aus! Vielen Dank!:

[image]

Einzig das Tooltip zeigt noch "%" und "Fahrzeuglimit" an.

[image]