evcc-io/evcc #19351: OCPP: Energy meter value not parsed/identified by EVCC

tafx, 2025-03-02:
### Describe the bug

EVCC does not recognize the value of energy meter from wallbox connected by OCPP, although the value "Energy.Active.Import.Register" is present in the message received from wallbox.
The wallbox is some cheap china product Sino Energy PEVC2107.
In the log is the message: "[lp-2 ] ERROR 2025/03/02 20:35:55 charge total import: not available ".

Different OCPP charger config parameters were tested (e.g. metervalues), cf. config file without success.
2 chargers of the same type are used and configured with the same issue.

### Steps to reproduce

1. Start EVCC with config below and wallbox conncected.
2. Look at log.
3. Alternatively plug EV, initialize charging and see that charged energy is not reported during the session.

[evcc-20250302-201859-trace.log](https://github.com/user-attachments/files/19044548/evcc-20250302-201859-trace.log)

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070
interval: 10s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

meters:
  - type: custom
    name: grid1
    power:
      source: mqtt
      topic: evcc/tometers/grid
      timeout: 30s
      scale: 1

  - name: pv1
    type: template
    template: deye-hybrid-3p
    usage: pv
    # RS485 via TCP/IP (Modbus RTU)
    modbus: tcpip
    id: 1
    host: xxx # Hostname
    port: xxx # Port

  - name: battery1
    type: template
    template: deye-hybrid-3p
    usage: battery
    # RS485 via TCP/IP (Modbus RTU)
    modbus: tcpip
    id: 1
    host: xxx # Hostname
    port: xxx # Port

  - type: custom
    name: Bojler
    power:
      source: mqtt
      topic: evcc/tometers/bojler_power
      timeout: 30s
      scale: 1
[... 280 lines cut ...]
  user: xxx
  password: xxx
```

### Log details

```text
Log attached as a file.
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

premultiply, 2025-03-02:
Please show the output of `evcc charger --log trace,db:error --diagnose`.

Stop the evcc service first before running this command.

mfuchs1984, 2025-03-02:
The charging station rejects the measurand activation

```
[ocpp  ] TRACE 2025/03/02 20:14:23 send 222: [2,"496852054","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2025/03/02 20:14:24 recv 222: [3,"496852054",{"status":"Rejected"}]
```

but still sends the data.

In case of Current.Import, it accepts the configuration.

tafx, 2025-03-03:
Thanks for looking at it. Should I try to change config somehow? It was my guess that the wallbox has nonstandard responses. Is there any workaround?

andig, 2025-03-03:
You can set `metervalues` imho and evcc will trust those as „working“. Might also help to open a vendor ticket to get this fixed.

mfuchs1984, 2025-03-03:
Was thinking this as well, but at least in the provided config, they are configured 
`metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import,Voltage # optional
`

tafx, 2025-03-03:
Yes for the case of wallbox with id 111. The second wallbox (id 222) has minimal config. Results seem to me to be the same.
I already tried to set metervalues but without any effect. To be honest I dont know what each of config parameters mean as I found no reference to them.

mfuchs1984, 2025-03-03:
Oh, could you please temporarily remove the second charger and create new logs? It is not easy to get an overview when having to fund our for each message to which charger it belongs.

tafx, 2025-03-04:
Here is the new log with just one charger. 

[evcc-20250304-110350-trace.txt](https://github.com/user-attachments/files/19068198/evcc-20250304-110350-trace.txt)

mfuchs1984, 2025-03-04:
Ich glaube ich habe das Problem gefunden: 

Der charger meldet eine Phase bei `Energy.Active.Import.Register`
`{"value":"824952","context":"Trigger","format":"Raw","measurand":"Energy.Active.Import.Register","phase":"L1","location":"Body","unit":"Wh"}`
was irgendwie ziemlicher Quatsch ist.

Hier wird dann die Phase an den Typ angehängt
https://github.com/evcc-io/evcc/blob/f26786ca9d85134c009654fd745b680f37eb08b6/charger/ocpp/connector_core.go#L53

Was dann hier wiederum dazu führt, dass der Wert nicht gefunden wird
https://github.com/evcc-io/evcc/blob/f26786ca9d85134c009654fd745b680f37eb08b6/charger/ocpp/connector.go#L319

mfuchs1984, 2025-03-04:
While Ocpp specs are very vague about this
>The phase field is not applicable to all Measurands.

it is quite obvious that it does not make sense any sense in case of this specific measurand.

mfuchs1984, 2025-03-04:
Still, seems to be easy to solve in evcc. Just fall back to phase measurements as it is done here
https://github.com/evcc-io/evcc/blob/f26786ca9d85134c009654fd745b680f37eb08b6/charger/ocpp/connector.go#L274

tafx, 2025-03-04:
Thanks to both.
Understood. The problem is "phase":"L1" in the part specifiying total energy charged. So I try to find if there is any possibility to restrict content of the ocpp response. Probably there is no option.
I will try ask the vendor but they did not responded in the past regarding modbus register description so the only way will be to replace the wallboxes or add external meters.

mfuchs1984, 2025-03-04:
Created the pull request #19405, I think it might make this charger work while not hurting anyone else.

closed 2025-03-04 by andig

mfuchs1984, 2025-03-04:
@tafx you can try again with the next nightly build.

mfuchs1984, 2025-03-04:
I think it is possible, see https://docs.evcc.io/en/docs/installation/home-assistant#installation

But for quick testing, it might be simpler to just run the nightly on your normal computer with the same config as in home assistant (maybe without vehicle configuration).

Anyhow, you have to wait for the next nightly build.

The other option is to wait for the next release.

tafx, 2025-03-08:
Here is report: EVCC have the values from both wallboxes, but with wrong unit. In the part of the message the unit is correct  {"value":"826199","context":"Trigger","format":"Raw","measurand":"Energy.Active.Import.Register","phase":"L1","location":"Body","unit":"Wh"}.
But EVCC did not account and reads the value as kWh. Could it be solved in EVCC please?

andig, 2025-03-08:
Excellent, here we go: https://github.com/evcc-io/evcc/pull/19590

tafx, 2025-03-11:
Unit conversion is working. Thanks.
Now I have new problem - after car connection the charging is immediately starting no matter wich mode (off, pv, min+pv, fast) is set in EVCC. It seems to me that charger ignores switch off command. The control of power - 4.1-11kW works.
Should I set new issue? The log file is attached.

[evcc-20250310-184649-trace.log.docx](https://github.com/user-attachments/files/19182716/evcc-20250310-184649-trace.log.docx)

mfuchs1984, 2025-03-11:
Hmm, according to the log, the charge currents are reduced to 0 when evcc stops charging.

Please create a new proper log file in txt format, starting with evcc startup, then connect a car and switch between charging modes. Give itv2 minutes between switching modes.

tafx, 2025-03-15:
So I got time to make an experiment as you required:
approx time 10:58 EVCC restart
11:00 all chargers turned to OFF
11:01 car plugged in, LP2 mode OFF, wallbox started charging with minimal powar 4,1 kW although mode OFF set
11:03 LP2 mode set to Solar, other chargers still OFF
11:05 LP2 mode still Solar, LP3 set to Fast in order to force the change charge power of LP2, power changed as i should
11:07 LP2 mode still Solar, LP3 set to OFF
11:09 LP2 mode se to OFF, wallbox decreased power to 4,1 kW but continued with charging and did not switch off

Corresponding filtered and full log files are enclosed.
I appretiate any hel/advice. Even the throwing wallbox to the bin :)

[evcc-20250315-111154-trace_full.log](https://github.com/user-attachments/files/19260704/evcc-20250315-111154-trace_full.log)
[evcc-20250315-111038-trace_filtered ocpp_lp2.log](https://github.com/user-attachments/files/19260705/evcc-20250315-111038-trace_filtered.ocpp_lp2.log)

mfuchs1984, 2025-03-15:
Seems that it switches off for a short time after receiving the charging profile with limit 0, but then continues charging with 6 A. 
This might be another Wallbox that would supposedly work when stopping the transaction.
Do you have home assistant?

tafx, 2025-03-15:
Yes I have HA. EVCC is running as addon in HA.

mfuchs1984, 2025-03-15:
Then you could try this
https://github.com/evcc-io/evcc/discussions/18414

(In case the integration 0.8.x does not work, install 0.7.0)

tafx, 2025-03-15:
Thank for the hint. I will try.