evcc-io/evcc #19352: charging started for no reason by itself

tomaszrondio, 2025-03-02:
### Describe the bug

Any ideas why it started charging?

api/state response

```
{
  "result": {
    "auth": {

    },
    "aux": [],
    "battery": [],
    "batteryDischargeControl": false,
    "batteryGridChargeActive": false,
    "batteryMode": "unknown",
    "bufferSoc": 0,
    "bufferStartSoc": 0,
    "circuits": {
      "main": {
        "power": 3799.687,
        "current": 16.688,
        "maxPower": 4600,
        "maxCurrent": 20
      }
    },
    "currency": "PLN",
    "eebus": false,
    "ext": [],
    "fatal": null,
    "forecast": {
      "grid": [
        {
          "start": "2025-03-02T00:00:00+01:00",
          "end": "2025-03-02T01:00:00+01:00",
          "price": 0.75013
        },
        {
          "start": "2025-03-02T01:00:00+01:00",
          "end": "2025-03-02T02:00:00+01:00",
          "price": 0.71962
        },
        {
          "start": "2025-03-02T02:00:00+01:00",
          "end": "2025-03-02T03:00:00+01:00",
          "price": 0.69077
        },
        {
          "start": "2025-03-02T03:00:00+01:00",
          "end": "2025-03-02T04:00:00+01:00",
          "price": 0.67264
        },
        {
          "start": "2025-03-02T04:00:00+01:00",
          "end": "2025-03-02T05:00:00+01:00",
          "price": 0.67264
        },
        {
          "start": "2025-03-02T05:00:00+01:00",
          "end": "2025-03-02T06:00:00+01:00",
          "price": 0.66403
        },
        {
          "start": "2025-03-02T06:00:00+01:00",
          "end": "2025-03-02T07:00:00+01:00",
          "price": 0.67264
        },
        {
          "start": "2025-03-02T07:00:00+01:00",
[... 495 lines cut ...]
[lp-1 ] DEBUG 2025/03/02 21:00:53 soc estimated: 75.27% (vehicle: 75.00%)
[lp-1 ] DEBUG 2025/03/02 21:00:53 vehicle soc: 75%
[lp-1 ] DEBUG 2025/03/02 21:00:53 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/03/02 21:00:53 vehicle range: 397km
[lp-1 ] DEBUG 2025/03/02 21:00:53 smart cost active: 0.77
[site ] DEBUG 2025/03/02 21:01:08 ----
[lp-1 ] DEBUG 2025/03/02 21:01:08 charge power: 3515W
[lp-1 ] DEBUG 2025/03/02 21:01:08 charge currents: [16 0.5 0.2]A
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

StefanSchoof, 2025-03-02:
```
        "smartCostActive": true,
        "smartCostLimit": 0.8,
```

```
        {
          "start": "2025-03-02T20:00:00+01:00",
          "end": "2025-03-02T21:00:00+01:00",
          "price": 0.82687
        },
        {
          "start": "2025-03-02T21:00:00+01:00",
          "end": "2025-03-02T22:00:00+01:00",
          "price": 0.77214
        },
```

There is an active smart coat limit and at 21:00 the price has gone below your configured limit.

tomaszrondio, 2025-03-02:
the only reason I came up with is that energy price went from 0.8x to 0.7x which made it below price limit. If that is the case then we need a better documentation explaining the logic behind charging.

tomaszrondio, 2025-03-02:
ok then this is acutally a cool feature I didn't know about. 
So there is:
1. plan - it plans to charge cheapest possible
2. limit - will charge every time price is below limit? Would be nice but that rises the question why it was not charging before. I had whole day of prices below the limit and car connected.

mfuchs1984, 2025-03-02:
There is documentation of the feature, see https://docs.evcc.io/en/docs/features/dynamic-prices#cheap-grid-charging

StefanSchoof, 2025-03-02:
Also there should be an icon with the reason why the charging is happening.

tomaszrondio, 2025-03-02:
thanks a lot

closed 2025-03-02 by tomaszrondio