evcc-io/evcc #19354: Error when adding OCPP charger - create failed

luboshl, 2025-03-02:
### Describe the bug

If I try to add OCPP charger, i get "create failed" error even when validation of OCPP charger succeeded.

### Steps to reproduce

1. Clear installation, docker image, version 0.200.5
2. Add sponsorhip token to evcc.yaml
3. Start configuration in UI
4. Configure network to 172.16.1.4:7070
5. Add charger
6. Set name
7. Select "OCPP 1.6J comptabile charger..." manufacturer
8. Left OCPP-Server URL with prefilled value "ws://172.16.1.4:8887"
9. Set Station ID to "EZ8-NCG-20K"
10. Click to Validate - overall status changed to green "successful", status in detail to "connected" with some other green info
11. Click to Save button -> error "create failed" is displayed in browser's dialog

### Configuration details

```yaml
sponsortoken: eyJhb********
```

### Log details

```text
/
Logs
Search

TRACE
All areas
[main ] INFO 2025/03/02 23:20:12 evcc 0.200.5
[main ] INFO 2025/03/02 23:20:12 using config file: /etc/evcc.yaml
[db ] INFO 2025/03/02 23:20:12 using sqlite database: /root/.evcc/evcc.db
[db ] TRACE 2025/03/02 23:20:12 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT * FROM `settings` 3 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main ] INFO 2025/03/02 23:20:12 listening at :7070
[db ] TRACE 2025/03/02 23:20:12 SELECT * FROM `configs` WHERE `configs`.`class` = 5 0 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT * FROM `configs` WHERE `configs`.`class` = 3 0 <nil>
[db ] TRACE 2025/03/02 23:20:12 SELECT * FROM `configs` WHERE `configs`.`class` = 5 0 <nil>
[site ] INFO 2025/03/02 23:20:12 site config:
[site ] INFO 2025/03/02 23:20:12 meters: grid ✗ pv ✗ battery ✗
[site ] INFO 2025/03/02 23:20:12 tariffs:
[site ] INFO 2025/03/02 23:20:12 grid: ✗
[site ] INFO 2025/03/02 23:20:12 feed-in: ✗
[site ] INFO 2025/03/02 23:20:12 co2: ✗
[site ] INFO 2025/03/02 23:20:12 solar: solar
[main ] TRACE 2025/03/02 23:20:12 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[site ] WARN 2025/03/02 23:20:12 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[main ] TRACE 2025/03/02 23:20:12 {"url":"https://api.github.com/repos/evcc-io/evcc/releases/203162908","assets_url":"https://api.github.com/repos/evcc-io/evcc/releases/203162908/assets","upload_url":"https://uploads.github.com/repos/evcc-io/evcc/releases/203162908/assets{?name,label}","html_url":"https://github.com/evcc-io/evcc/releases/tag/0.200.5","id":203162908,"author":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"node_id":"RE_kwDODX4bUs4MHAUc","tag_name":"0.200.5","target_commitish":"master","name":"0.200.5","draft":false,"prerelease":false,"created_at":"2025-03-01T11:12:56Z","published_at":"2025-03-01T11:30:32Z","assets":[{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/233627782","id":233627782,"node_id":"RA_kwDODX4bUs4N7OCG","name":"checksums.txt","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"content_type":"text/plain; charset=utf-8","state":"uploaded","size":754,"download_count":2,"created_at":"2025-03-01T11:42:54Z","updated_at":"2025-03-01T11:42:54Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.200.5/checksums.txt"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/233627779","id":233627779,"node_id":"RA_kwDODX4bUs4N7OCD","name":"evcc_0.200.5_amd64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"https://api.github.com/users/andig/received_events","type":"User","user_view_type":"public","site_admin":false},"content_type":"application/vnd.debian.binary-package","state":"uploaded","size":23377990,"download_count":14,"created_at":"2025-03-01T11:42:52Z","updated_at":"2025-03-01T11:42:54Z","browser_download_url":"https://github.com/evcc-io/evcc/releases/download/0.200.5/evcc_0.200.5_amd64.deb"},{"url":"https://api.github.com/repos/evcc-io/evcc/releases/assets/233627777","id":233627777,"node_id":"RA_kwDODX4bUs4N7OCB","name":"evcc_0.200.5_arm64.deb","label":"","uploader":{"login":"andig","id":184815,"node_id":"MDQ6VXNlcjE4NDgxNQ==","avatar_url":"https://avatars.githubusercontent.com/u/184815?v=4","gravatar_id":"","url":"https://api.github.com/users/andig","html_url":"https://github.com/andig","followers_url":"https://api.github.com/users/andig/followers","following_url":"https://api.github.com/users/andig/following{/other_user}","gists_url":"https://api.github.com/users/andig/gists{/gist_id}","starred_url":"https://api.github.com/users/andig/starred{/owner}{/repo}","subscriptions_url":"https://api.github.com/users/andig/subscriptions","organizations_url":"https://api.github.com/users/andig/orgs","repos_url":"https://api.github.com/users/andig/repos","events_url":"https://api.github.com/users/andig/events{/privacy}","received_events_url":"h [... cut ...] .Active.Import","unit":"W"}]}]}]
[ocpp ] TRACE 2025/03/02 23:24:18 send EZ8-NCG-20K: [3,"1316680",{}]
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

mfuchs1984, 2025-03-03:
Does evcc start and run? The log looks good.

https://github.com/evcc-io/evcc/issues/19229 seems to be similar.

andig, 2025-03-03:
Please try todays nightly.

michai-27, 2025-03-04:
After the last nightly build I can create a charger, but after a restart evcc is not able to start due an error:

[UI config: Add loadpoint ABB Terra (OCPP) #19229 ](https://github.com/evcc-io/evcc/issues/19229)

I suspect that these topics belong together. Otherwise, please let me know if this issue belongs somewhere else or is a completely new one.

andig, 2025-03-04:
See other issue- factory reset recommended.