evcc-io/evcc #19355: Tesla fleet API access fails with 412 because of incorrect region

MarcusWichelmann, 2025-03-02:
### Describe the bug

I configured the Tesla fleet API access as described [here](https://docs.evcc.io/blog/2025/01/20/tesla-api-update), but instead of myteslamate.com I used https://github.com/MikeBishop/fleetapi-tokens on my own server.

I could successfully retrieve an `accessToken` and `refreshToken` and configured evcc as follows:

```
vehicles:
  - name: tesla-model3
    title: Tesla Model 3
    icon: car
    type: template
    template: tesla-command
    clientId: 2d08a8cc-...
    accessToken: eyJhbGciOiJSUzI...
    refreshToken: EU_63a030315...
    vin: LRW...
    capacity: 60
    phases: 3
    minCurrent: 6
    maxCurrent: 16
    mode: pv
    priority: 10
```

With trace-logging enabled, I can see this in the evcc log:

```
[tesla ] TRACE 2025/03/02 23:07:38 GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
[tesla ] TRACE 2025/03/02 23:07:39 {"error":"Account *** must be registered in the current region https://fleet-api.prd.na.vn.cloud.tesla.com, please see https://developer.tesla.com/docs/fleet-api/endpoints/partner-endpoints#register"}
[main  ] ERROR 2025/03/02 23:07:39 creating vehicle tesla-model3 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': 412 Precondition Failed
```

It looks like evcc tries to call the north american API endpoint to detect the region (EU) of my token, but this fails, because my token is not valid for the north american region.

This can be seen when I manually call the `/users/region` endpoint from different regions:

```
curl --header 'Content-Type: application/json' --header "Authorization: Bearer <access-token>" https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/users/region
{"response":{"region":"eu","fleet_api_base_url":"https://fleet-api.prd.eu.vn.cloud.tesla.com"}
```

```
curl --header 'Content-Type: application/json' --header "Authorization: Bearer <access-token>" https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/users/region
{"error":"Account 2d08a8cc-... must be registered in the current region https://fleet-api.prd.na.vn.cloud.tesla.com, please see https://developer.tesla.com/docs/fleet-api/endpoints/partner-endpoints#register"}
```

My [`fleetapi-tokens` server](https://github.com/MikeBishop/fleetapi-tokens) is configured like this:

```
      NODE_ENV: production
      DOMAIN: <my domain>
      ALLOWED_USERS: <my mail address>
      AUDIENCE: https://fleet-api.prd.eu.vn.cloud.tesla.com
      LOCALE: de_DE
      CLIENT_ID: 2d08a8cc-...
      CLIENT_SECRET: ta-secret....
```

The `fleetapi-tokens` application registers itself using this endpoint: https://developer.tesla.com/docs/fleet-api/endpoints/partner-endpoints#register
Code: https://github.com/MikeBishop/fleetapi-tokens/blob/540ca19de1fdf13935da791fd3e2f4e40556965e/tesla-tokens.js#L83

Because my audience is `https://fleet-api.prd.eu.vn.cloud.tesla.com`, my application is registered in EU only. This seems to be an issue for the way evcc auto-detects the region: https://github.com/evcc-io/evcc/blob/3ade50fcd1c84f531801df24b9c1558652a15b57/vehicle/tesla.go#L80

Is there a way to disable the auto-detection of my region in evcc and configure the EU URL manually?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
...
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

MarcusWichelmann, 2025-03-02:
As a workaround, it helps to start the `fleetapi-tokens` docker once with the north american `AUDIENCE`. This way the application gets registered for that region too and from that point on, the region-detection in evcc works fine.

An option to manually set the region evcc uses, would still be great.

andig, 2025-03-03:
I‘m not sure that‘s what happens. When calling the region endpoint, a server CANNOT know what region the user/account is in- it‘s only trying to find that out! Imho the region endpoint should either answer requests from any region (and I seem to recall that it does) or it might help to add more audiences to your token if that‘s possible. I‘d suggest to migrate this issue to test vehicle commands.

andig, 2025-03-03:
> Is there a way to disable the auto-detection of my region in evcc and configure the EU URL manually?

Let me check that, it would still be an (ugly) workaround.

MarcusWichelmann, 2025-03-03:
> Imho the region endpoint should either answer requests from any region

This would definitely make sense, but in my manual testing, this wasn't the case unfortunately.
Note that the issue is the application region here. When the application is registered in the US as well, it **is** able to tell me the region of EU user accounts.

Registering the application for both regions helped though. For `myteslamate.com`, this is probably the case, therefore this was never an issue.

MarcusWichelmann, 2025-03-03:
> Let me check that, it would still be an (ugly) workaround.

Yeah, I agree it's not the nicest solution. It's on you to decide whether you want that.

At least this issue serves as documentation for others that may stumble over that.

closed 2025-03-03 by andig

lupinve, 2025-03-17:
Hi, I'm experiencing an issue when trying to add my vehicle.

The error message I receive is:
[Tesla] TRACE 2025/03/17 17:15:34 {"error":"Account *** must be registered in the current region https://fleet-api.prd.na.vn.cloud.tesla.com/, please see [https://developer.tesla.com/docs/fleet-api/endpoints/partner-endpoints#register"}](https://developer.tesla.com/docs/fleet-api/endpoints/partner-endpoints#register%22%7D)

Could you please help me understand how to adjust the region?

loeffelpan, 2025-03-22:
Same here.

@andig converted #19886 to 19890 without any comment. Seems to be the same issue when creating tokens with myteslamate as described in the docs.

andig, 2025-03-23:
Incorrect region has been discussed before. Afair your tesla client needs be in the same region as your vehicle. That doesn‘t make sense on the Tesla side of the api and may warrant help from Tesla support.