evcc-io/evcc #19357: Tibber Tarif connection refused

EvilNievel, 2025-03-03:
### Describe the bug

I have sometimes the problem that the price from Tibber not update.
EVVCC run on Homeassistant und in the Terminal i can Ping www.google.de  so i think the Internet is ok.

This Error i have on evcc

 planner: tariff not available: cannot create tariff type 'tibber': Message: Post "https://api.tibber.com/v1-beta/gql": dial tcp 34.248.178.69:443: connect: connection refused, Locations: [], Extensions: map[code:request_error], Path: []

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    #type: fixed
    #price: 0.294 # EUR/kWh
   # zones:
     # - days: Mon-Fri
     #   hours: 2-5
     #   price: 0.2 # EUR/kWh
     # - days: Sat,Sun
    #    price: 0.15 # EUR/kWh

    # or variable tariffs
     type: tibber
     token: "myToken" # access token
```

### Log details

```text
planner: tariff not available: cannot create tariff type 'tibber': Message: Post "https://api.tibber.com/v1-beta/gql": dial tcp 34.248.178.69:443: connect: connection refused, Locations: [], Extensions: map[code:request_error], Path: []
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

StefanSchoof, 2025-03-03:
The Tibber server does reject your connection. This could happen if tibber has server problems, tibber blocks your IP or your TCP Internet connection does not work (ping uses icmp).

andig, 2025-03-03:
You‘ll have to contact Tibber. Nithing evcc can do.

closed 2025-03-03 by andig