evcc-io/evcc #19373: Hybrid inverter - PV power

djfanatix, 2025-03-03:
### Describe the bug

I noticed today that the following is happening:

I have a hybrid inverter with 4000w ac power and about 7000wp solar panels (dc) power

I see EVCC is using DC power to calculate pv surplus, so in the end EVCC is importing power from the grid when there is a lot of sun.

So I had 6kw PV power, but max 4 kw inverter, so EVCC is importing 2kw to charge at 6kw

I have max AC power configured in hybrid battery an hybrid inverter

It should be:
1. PV mode: max AC power - consumption = PV Surpluss
2. PV mode + cheap tariff: same as above + circuit max
3. Fast: Same as above

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
How to paste with new UI config?

It is a hybrid solaredge
```

### Log details

```text
/
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

mfuchs1984, 2025-03-03:
Do you use the maxacpower option in the PV meter configuration? It is made for this scenario (but there seems to be an issue with it, currently under investigation).

mfuchs1984, 2025-03-03:
Some discussions here are ongoing if it is implemented correctly, see latest comments in https://github.com/evcc-io/evcc/discussions/18943