evcc-io/evcc #19374: Hyundai Bluelink SOC funktioniert nach Update von EVCC auf 0.200.5 nicht mehr

tha0o0, 2025-03-03:
### Describe the bug

Bis heute vormittag hat EVCC bei meinem Hyundai erfolgreich alle Werte von Bluelink empfangen. Nun habe ich simpel upgedatd von Version 0.200.2 auf 0.200.5 und seitdem habe ich einen Fehler

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
evcc.yaml
```

### Log details

```text
[lp-2 ] DEBUG 2025/03/03 16:00:13 charge total import: 3084.869kWh
[shelly] TRACE 2025/03/03 16:00:13 POST http://192.168.178.102/rpc/Shelly.GetStatus
[shelly] TRACE 2025/03/03 16:00:13 {"id":0,"on":false,"src":"evcc","method":"Shelly.GetStatus"} -- {"ble":{},"cloud":{"connected":true},"input:0":{"id":0,"state":false},"mqtt":{"connected":false},"script:1":{"id":1,"running":true,"mem_used":560,"mem_peak":3136,"mem_free":24626},"switch:0":{"id":0, "source":"HTTP_in", "output":true, "apower":516.2, "voltage":230.9, "current":2.427, "aenergy":{"total":3084869.496,"by_minute":[8666.280,8648.918,8659.109],"minute_ts":1741014000},"temperature":{"tC":44.4, "tF":111.9}},"sys":{"mac":"B48A0A20E174","restart_required":false,"time":"16:00","unixtime":1741014019,"uptime":7492228,"ram_size":267344,"ram_free":116528,"fs_size":393216,"fs_free":98304,"cfg_rev":18,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{"beta":{"version":"1.5.1-beta1"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.178.102","status":"got ip","ssid":"Karsten und Steffi","rssi":-64},"ws":{"connected":false}}
[lp-2 ] DEBUG 2025/03/03 16:00:13 charger status: C
[shelly] TRACE 2025/03/03 16:00:13 POST http://192.168.178.102/rpc/Switch.GetStatus
[shelly] TRACE 2025/03/03 16:00:13 {"id":0,"on":false,"src":"evcc","method":"Switch.GetStatus"} -- {"id":0, "source":"HTTP_in", "output":true, "apower":516.2, "voltage":230.9, "current":2.427, "aenergy":{"total":3084869.496,"by_minute":[8666.280,8648.918,8659.109],"minute_ts":1741014000},"temperature":{"tC":44.4, "tF":111.9}}
[site ] DEBUG 2025/03/03 16:00:14 ----
[shelly] TRACE 2025/03/03 16:00:14 POST http://192.168.178.102/rpc/Shelly.GetStatus
[go-e ] TRACE 2025/03/03 16:00:14 GET http://192.168.178.96/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[shelly] TRACE 2025/03/03 16:00:14 GET http://192.168.178.123/status
[shelly] TRACE 2025/03/03 16:00:14 GET http://192.168.178.122/status
[shelly] TRACE 2025/03/03 16:00:14 {"wifi_sta":{"connected":true,"ssid":"Karsten und Steffi","ip":"192.168.178.122","rssi":-77},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"16:00","unixtime":1741014014,"serial":7042,"has_update":false,"mac":"84CCA89F6B03","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"source":"http"}],"meters":[{"power":0.00,"overpower":0.00,"is_valid":true,"timestamp":1741017614,"counters":[0.000, 0.000, 0.000],"total":0}],"inputs":[{"input":0,"event":"","event_cnt":0}],"temperature":43.55,"overtemperature":false,"tmp":{"tC":43.55,"tF":110.39, "is_valid":true},"temperature_status":"Normal","ext_sensors":{},"ext_temperature":{},"ext_humidity":{},"update":{"status":"idle","has_update":false,"new_version":"20230913-113709/v1.14.0-gcb84623","old_version":"20230913-113709/v1.14.0-gcb84623","beta_version":"20231107-164450/v1.14.1-rc1-g0617c15"},"ram_total":42960,"ram_free":29852,"fs_size":233681,"fs_free":148843,"uptime":318859}
[lp-4 ] DEBUG 2025/03/03 16:00:14 charge power: 0W
[shelly] TRACE 2025/03/03 16:00:14 {"wifi_sta":{"connected":true,"ssid":"Karsten und Steffi","ip":"192.168.178.123","rssi":-78},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"16:00","unixtime":1741014014,"serial":43142,"has_update":false,"mac":"84CCA89FD8A5","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"source":"input"}],"meters":[{"power":2.45,"overpower":0.00,"is_valid":true,"timestamp":1741017615,"counters":[2.304, 2.214, 2.231],"total":71718}],"inputs":[{"input":0,"event":"","event_cnt":0}],"temperature":46.79,"overtemperature":false,"tmp":{"tC":46.79,"tF":116.22, "is_valid":true},"temperature_status":"Normal","ext_sensors":{},"ext_temperature":{},"ext_humidity":{},"update":{"status":"idle","has_update":false,"new_version":"20230913-113709/v1.14.0-gcb84623","old_version":"20230913-113709/v1.14.0-gcb84623","beta_version":"20231107-164450/v1.14.1-rc1-g0617c15"},"ram_total":42960,"ram_free":30380,"fs_size":233681,"fs_free":148843,"uptime":1952909}
[lp-3 ] DEBUG 2025/03/03 16:00:14 charge power: 0W
[go-e ] TRACE 2025/03/03 16:00:14 {"alw":true,"eto":4122317,"car":4,"wh":32254.84028,"trx":null,"nrg":[233.1199951,231.5700073,234.6699982,2.480000019,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
[lp-1 ] DEBUG 2025/03/03 16:00:14 charge power: 0W
[lp-1 ] DEBUG 2025/03/03 16:00:14 charge currents: [0 0 0]A
[shelly] TRACE 2025/03/03 16:00:15 {"id":0,"on":false,"src":"evcc","method":"Shelly.GetStatus"} -- {"ble":{},"cloud":{"connected":true},"input:0":{"id":0,"state":false},"mqtt":{"connected":false},"script:1":{"id":1,"running":true,"mem_used":560,"mem_peak":3136,"mem_free":24626},"switch:0":{"id":0, "source":"HTTP_in", "output":true, "apower":522.4, "voltage":230.9, "current":2.431, "aenergy":{"total":3084869.641,"by_minute":[8666.280,8648.918,8659.109],"minute_ts":1741014000},"temperature":{"tC":44.4, "tF":112.0}},"sys":{"mac":"B48A0A20E174","restart_required":false,"time":"16:00","unixtime":1741014021,"uptime":7492229,"ram_size":267332,"ram_free":116384,"fs_size":393216,"fs_free":98304,"cfg_rev":18,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{"beta":{"version":"1.5.1-beta1"}},"reset_reason":3},"wifi":{"sta_ip":"192.168.178.102","status":"got ip","ssid":"Karsten und Steffi","rssi":-65},"ws":{"connected":false}}
[lp-2 ] DEBUG 2025/03/03 16:00:15 charge power: 522W
[shelly] TRACE 2025/03/03 16:00:15 POST http://192.168.178.68/rpc/EM.GetStatus
[sunspec] TRACE 2025/03/03 16:00:15 modbus: send 2a 28 00 00 00 06 01 03 9c 87 00 32
[sunspec] TRACE 2025/03/03 16:00:15 modbus: send a8 43 00 00 00 06 01 03 9d 49 00 08
[my_pv3] TRACE 2025/03/03 16:00:15 GET http://192.168.178.119/api/livedata/status
[shelly] TRACE 2025/03/03 16:00:15 {"id":0,"on":false,"src":"evcc","method":"EM.GetStatus"} -- {"id":0,"a_current":2.667,"a_voltage":234.3,"a_act_power":-461.4,"a_aprt_power":624.5,"a_pf":0.74,"a_freq":50.0,"b_current":1.980,"b_voltage":234.5,"b_act_power":-268.8,"b_aprt_power":464.3,"b_pf":0.57,"b_freq":50.0,"c_current":3.534,"c_voltage":236.3,"c_act_power":-773.2,"c_aprt_power":834.1,"c_pf":0.93,"c_freq":50.0,"n_current":null,"total_current":8.181,"total_act_power":-1503.436,"total_aprt_power":1922.891, "user_calibrated_phase":[]}
[site ] DEBUG 2025/03/03 16:00:15 grid power: -1503W
[shelly] TRACE 2025/03/03 16:00:15 POST http://192.168.178.68/rpc/EM.GetStatus
[sunspec] TRACE 2025/03/03 16:00:15 modbus: recv 2a 28 00 00 00 67 01 03 64 03 7d 01 30 01 26 01 26 ff fe 0f e2 0f d6 0f f5 09 39 09 28 09 26 ff ff 50 a0 ff ff 13 8d ff fe 52 1f ff ff f0 66 ff ff d9 a7 ff fe 04 e9 42 e8 00 00 6d 26 ff fc 1d 4b ff ff 51 da ff ff 80 00 0f 65 80 00 80 00 ff fe 00 04 00 00 ff ff ff ff ff ff ff ff 00 00 00 00 ff ff ff ff ff ff ff ff 00 00 00 00
[sunspec] TRACE 2025/03/03 16:00:15 model 103 block 0 point W: 2064
[site ] DEBUG 2025/03/03 16:00:15 pv 1 power: 2064W
[sunspec] TRACE 2025/03/03 16:00:15 modbus: send 2a 29 00 00 00 06 01 03 9c 87 00 32
[shelly] TRACE 2025/03/03 16:00:15 {"id":0,"on":false,"src":"evcc","method":"EM.GetStatus"} -- {"id":0,"a_current":2.667,"a_voltage":234.3,"a_act_power":-461.4,"a_aprt_power":624.5,"a_pf":0.74,"a_freq":50.0,"b_current":1.980,"b_voltage":234.5,"b_act_power":-268.8,"b_aprt_power":464.3,"b_pf":0.57,"b_freq":50.0,"c_current":3.534,"c_voltage":236.3,"c_act_power":-773.2,"c_aprt_power":834.1,"c_pf":0.93,"c_freq":50.0,"n_current":null,"total_current":8.181,"total_act_power":-1503.436,"total_aprt_power":1922.891, "user_calibrated_phase":[]}
[site ] DEBUG 2025/03/03 16:00:15 grid powers: [-461 -269 -773]W
[... cut ...]
[shelly] TRACE 2025/03/03 16:00:20 {"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"source":"http"}
[lp-4 ] DEBUG 2025/03/03 16:00:20 pv charge current: 6.63A = 0A + 6.63A (-1525W @ 1p)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

andig, 2025-03-03:
Geht die alte Version noch? Was sagt

    evcc vehicle

VolkerK62, 2025-03-03:
Kann ich nicht bestätigen. Funktioniert bei mir ohne Probleme.

tha0o0, 2025-03-03:
Danke Volker, aufgrund deines tips hab ich einfach mal in der UI das Fahzeug durchgeklickt und prüfen lassen jetzt gehts wieder scheinbar

closed 2025-03-03 by tha0o0

tha0o0, 2025-03-03:
Den hatte ich zweimal durchgeführt aber ohne Erfolg