evcc-io/evcc #19378: Loadpoint Priorität und preisgesteuertes Laden

VolkerK62, 2025-03-03:
Ja, shelly bleibt.
Auch wenn ich disable.treshold von PV auf -15000 ändere schaltet es sich nicht aus.
[evcc-20250222-135435-debug.log](https://github.com/user-attachments/files/18922736/evcc-20250222-135435-debug.log)

_Originally posted by @longtom99 in https://github.com/evcc-io/evcc/discussions/19061#discussioncomment-12285680_

VolkerK62, 2025-03-03:
LP1 (Auto) lädt preisgesteuert (hat Prio 0)
LP2 (WP) ist im PV-Modus (hat Prio 1) und wird nicht abgeschaltet.
```
[site  ] DEBUG 2025/02/22 13:52:47 ----
[lp-1  ] DEBUG 2025/02/22 13:52:47 charge power: 10783W
[lp-2  ] DEBUG 2025/02/22 13:52:47 charge power: 541W
[lp-1  ] DEBUG 2025/02/22 13:52:47 charge currents: [15.6 15.7 15.6]A
[site  ] DEBUG 2025/02/22 13:52:47 ext 1 power: 541W
[site  ] DEBUG 2025/02/22 13:52:48 pv 1 power: 3778W
[site  ] DEBUG 2025/02/22 13:52:49 grid power: 8683W
[site  ] DEBUG 2025/02/22 13:52:49 site power: 8683W
[lp-1  ] DEBUG 2025/02/22 13:52:49 detected active phases: 3p
[lp-1  ] DEBUG 2025/02/22 13:52:49 charge total import: 5599.433kWh
[lp-1  ] DEBUG 2025/02/22 13:52:49 charger status: C
[lp-1  ] DEBUG 2025/02/22 13:52:49 soc estimated: 71.27% (vehicle: 71.00%)
[lp-1  ] DEBUG 2025/02/22 13:52:49 vehicle soc: 71%
[lp-1  ] DEBUG 2025/02/22 13:52:49 vehicle soc limit: 80%
[lp-1  ] DEBUG 2025/02/22 13:52:49 vehicle range: 252km
[lp-1  ] DEBUG 2025/02/22 13:52:49 smart cost active: 0.08
[site  ] DEBUG 2025/02/22 13:53:17 ----
[lp-1  ] DEBUG 2025/02/22 13:53:17 charge power: 10779W
[lp-2  ] DEBUG 2025/02/22 13:53:17 charge power: 541W
[lp-1  ] DEBUG 2025/02/22 13:53:17 charge currents: [15.6 15.7 15.6]A
[main  ] DEBUG 2025/02/22 13:53:17 lp Warmwasser at prio 1 gets additional 10779W from Parkplatz at prio 0, total 10779W
[site  ] DEBUG 2025/02/22 13:53:17 ext 1 power: 541W
[site  ] DEBUG 2025/02/22 13:53:18 pv 1 power: 3792W
[site  ] DEBUG 2025/02/22 13:53:19 grid power: 8536W
[site  ] DEBUG 2025/02/22 13:53:19 site power: -2243W (including 10779W prioritized power)
[lp-2  ] DEBUG 2025/02/22 13:53:19 charge total import: 742.045kWh
[lp-2  ] DEBUG 2025/02/22 13:53:19 charger status: C
[lp-2  ] DEBUG 2025/02/22 13:53:19 vehicle soc: 58%
[lp-2  ] DEBUG 2025/02/22 13:53:19 pv charge current: 12.8A = 3A + 9.75A (-2243W @ 1p)
[site  ] DEBUG 2025/02/22 13:53:47 ----
```
Erwartung wäre, dass LP2 in dem Fall abgeschalten würde, da preisgesteuertes Laden im Prinzip mit "schnell" Laden gleichzusetzen ist.

andig, 2025-03-03:
Ich verstehe die Logik nicht. LP2 hat höhere Prio- warum sollte gerade der abschalten?

VolkerK62, 2025-03-03:
Eigentlich genau die gleichen Bedingungen wie bei der Batteriesperre.

closed 2025-03-05 by andig