evcc-io/evcc #19387: Error when adding EVSwitch Charger

minhphuong0401, 2025-03-04:
### Describe the bug

Unable to add EVSwitch Charger https://www.evswitchstore.com.au/pages/ev-charger-range.
Error is "The system time on the Charge Point was changed"

### Steps to reproduce

1. Clear installation,  version 0.200.5
2. Add sponsorhip token to evcc.yaml
3. Add chargers to evcc.yaml

`
chargers:
  - name: my_charger
    type: template
    template: ocpp
    stationid: SN10052412277356
`
4. Run `evcc charger --log trace --diagnose`

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 5b6d0cc3191dc845299129db0dc53001fe494416b6080daa15949a9a099e5369

interval: 30s # control cycle interval

meters:
- type: template
  template: sungrow-inverter 
  id: 1  
  host: 192.168.68.117  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-inverter 
  id: 1  
  host: 192.168.68.117  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
    
tariffs:
  currency: AUD # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.24 # EUR/kWh
    zones:
[... 167 lines cut ...]
                TransactionMessageAttempts (r/w): 0
                TransactionMessageRetryInterval (r/w): 15
                UnlockConnectorOnEVSideDisconnect (r/w): true
                WebSocketPingInterval (r/w): 30
                WebsocketFragmentEnabled (r/w): true
                chargePointSN (r/w): SN10052412277356
                plugAndChargeMsg (r/w): false
                vendorId (r/w): EN+
```

### What type of operating system or environment does evcc run on?

Windows

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

mfuchs1984, 2025-03-04:
I think you can ignore this error, it should still run.

jeffborg, 2025-03-05:
It should be fine your firmware is new enough 1.0.25.130 earlier versions of the EN+ firmware had bugs, so long as evcc works ignore the message

loryanstrant, 2025-03-08:
> It should be fine your firmware is new enough 1.0.25.130 earlier versions of the EN+ firmware had bugs, so long as evcc works ignore the message

Good to hear, as I've just gotten the same unit installed this week and was seeing the same problem.

closed 2025-03-08 by andig

mfuchs1984, 2025-04-26:
@premultiply I suppose these messages would disappear as soon as evcc implements the security callbacks?
https://github.com/lorenzodonini/ocpp-go/blob/master/ocpp1.6/security/security.go#L15

premultiply, 2025-04-26:
Right. I wasn't aware that it was merged upstream.

andig, 2025-04-26:
That profile is optional. I guess we just shouldn't log any errors then. Not sure where they are coming from.