evcc-io/evcc #19388: Implementierung ANKER SOLIX X1

LadesaeulenBetreiber96450, 2025-03-04:
Nach langem hin und her, ist es endlich geschafft, die Modbus-Schnittstelle wurde beschrieben.

Bitte um Implementierung in evcc das System von Anker Solix X1
https://www.ankersolix.com/de/x1-energy-storage-system-hes

Modbus-Register-Tabelle befindet sich im Anhang.

Werden noch weitere Daten benötigt dann bitte schreibt mir.

[Signal interface Modbus Protocol ANKER Solix X1 28.02.25.xlsx](https://github.com/user-attachments/files/19065300/Signal.interface.Modbus.Protocol.ANKER.Solix.X1.28.02.25.xlsx)

Für eventuelle Feldversuche stehe ich zur Verfügung.

Gruß Dominik Wöhner

andig, 2025-03-04:
Es braucht nur ein Template dafür. PR welcome.

LadesaeulenBetreiber96450, 2025-03-07:
>   Hi @LadesaeulenBetreiber96450,
> ich denke, eine Diskussion wäre hier sinnvoller.
> Ich habe mir mal die Dokumentation angeschaut. Und mal zum Test für dich entsprechend mal ein custom-yaml erstellt, daraus können wir dann ein Template erstellen. Kannst du das mal bitte bei dir testen, ob die Adressen und die Skalierung passen? (einfach die IP durch deine ersetzen)
>
> - type: custom
>   name: grid1
>   power: # Watt
>     source: modbus
>     id: 1
>     uri: 192.168.178.100:502
>     register:
>       address: 10007 # Grid-connected power      [10007]
>       type: input
>       decode: uint16
>     scale: 0,1 # /1000
>   energy: # KWH
>     source: modbus
>     id: 1
>     uri: 192.168.178.100:502
>     register:
>       address: 10025 # Total purchased power     [10025]
>       type: input
>       decode: uint16
>     scale: 0,1
>   currents: # A
> [... 287 lines cut ...]
>     register:
>       address: 31001 #  Rated capacity    [31001]
>       type: input
>       decode: uint16
>     scale: 1

Moin,danke für die Vorarbeit 😊. Ich teste dies die kommenden 2 Wochen mal durch, leider schaffe ich es nicht eher. Melde mich wieder 👍Von meinem iPhone gesendet

vladikliese, 2025-03-10:
Hallo zusammen,

ich verfolge diese Implementierung mit Hochspannung. Weiß jemand ob eine derartige Implementierung mit der Anker Solix Solarbank 2 E1600 Pro für evcc gibt?

zachelnet, 2025-03-10:
Hi,
Leider nein, die Anker solix Solarbank Unterstützung kein modbus. 

Grüße Martin

closed 2025-03-20 by andig

Nasskappe, 2025-04-12:
Hi @LadesaeulenBetreiber96450 , konntest du das Template inzwischen testen?

LadesaeulenBetreiber96450, 2025-04-13:
Guten Morgen, 

leider nein, ich bin momentan dienstlich sehr eingespannt und husche von einer in die nächste Dienstreise. So wie es aussieht bin ich Ende April „Ready for this“. Momentan sieht es so aus 😅…

[image]
[image]

Melde mich

Nasskappe, 2025-04-14:
Hi,

 

wunderbar, danke für die Zwischeninfo.

 

Viele Grüße und viel Erfolg!

 

IMG_1748.jpeg (view on web) <https://github.com/user-attachments/assets/ca679980-23e2-436a-b546-62852e0e4376> 
IMG_1747.jpeg (view on web) <https://github.com/user-attachments/assets/f101aaf6-1862-40e3-a0e3-6860243a153f> 

IMG_1748.jpeg (view on web) <https://github.com/user-attachments/assets/ca679980-23e2-436a-b546-62852e0e4376> 
IMG_1747.jpeg (view on web) <https://github.com/user-attachments/assets/f101aaf6-1862-40e3-a0e3-6860243a153f>

stefanbritz, 2025-06-02:
Hallo, gibt es hier was neue habe die gleiche Anlage! Aber ich weis auch nicht wie ich an die Daten vom MODBus komme.

dmueller23, 2025-06-29:
Hallo zusammen,
und danke @zachelnet für das yaml.
Ich habe ein Solix X1 und das yaml bei mir integriert aber leider scheint der akku den modbus port gar nicht offen zu haben:
```
nmap --script modbus-discover.nse --script-args='modbus-discover.aggressive=true' -p- 192.168.178.xxx
All 65535 scanned ports on Anker.fritz.box (192.168.178.xxx) are closed
```
Kommt sowohl bei der WLAN als auch der LAN IP.
Wenn ich die Doku richtig interpretiere ist der Dongle für das LAN/WLAN per modbus angeschlossen:

```
WLAN Dongle (DG-WF-H)
...
Connector
--
Pin 1 | V+
Pin 2 | RS-485 B (D-)
Pin 3 | RS-485 A (D+)
Pin 4 | GND
```
Das hiesse man könnte da evtl. eine eigene Lösung dran stecken?

Hat da jemand eine Idee?

LG,

Dominik

zachelnet, 2025-07-01:
@dmueller23 Denk ich auch das Klemme 17/18 dafür da ist, hier einfach mit RS485 zu RJ45 Konverter draufgehen,

dmueller23, 2025-07-01:
Konverter ist bestellt, spätestens am WE kann ich's ausprobieren.
LG,
Dominik

dmueller23, 2025-07-06:
Hello,
Aaalso diesen Adapter angeschlossen und konfiguriert:
https://www.waveshare.com/wiki/RS232/485/422_TO_POE_ETH_(B)

Allerdings hadere ich noch mit der Konfiguration.
Habe ich Modbus TCP als Transfer Protocol eingestellt kommt:
```
[site ] ERROR 2025/07/06 22:14:39 pv 2 power: read failed: read tcp 192.168.178.214:40284->192.168.178.80:502: i/o timeout
```

Bei Real_COM

```
 ERROR 2025/07/06 22:12:03 2025-07-06 22:11:57.147569944 +0200 CEST m=+464.554064461 !! 3ms read failed: modbus: length in response header '64002' must not be zero or greater than '254'
```
Mit Modbus Mechanic kommt immer ein  i/o timeout.

Falls da jemand eine Idee oder einen Link hat, wäre ich dankbar.