evcc-io/evcc #19399: Tibber "Socket closed with event 4429 Too many open connections on this server"

chrstmnn, 2025-03-04:
### Describe the bug

Hi,

I am using Tibber as a meter in EVCC and after a few days, the subscription is blocked by Tibber:
Using the access token in the API explorer of Tibber, I receive:
`    "Socket closed with event 4429 Too many open connections on this server; count 2;"`

After re-creating a token and entering it to the EVCC yaml, it works again for a few days.

Currently, I am only using the access token for meter data in EVCC (already stopped using it for tariff data and Home Assistant).

Not using nightly as I don't know how to install on Proxmox. Please advice if needed. Checked that currently there are no (new) issues closed regarding this problem.

### Steps to reproduce

1. Create API token
2. Set up Tibber meter
3. Wait a few days
...

### Configuration details

```yaml
meters:
- type: template
  template: tibber-pulse 
  usage: grid  
  token: <token>
  homeid: <home-id> 
  name: grid1
```

### Log details

```text
pulse ] TRACE 2025/03/04 07:01:33 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:01:33 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:01:38 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:01:38 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:01:43 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:01:43 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:01:48 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:01:48 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:01:54 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:01:54 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:01:59 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:01:59 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:04 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:04 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:09 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:10 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:15 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:15 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:20 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:20 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:25 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:25 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:30 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:30 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:35 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:35 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:40 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:41 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:46 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:46 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:51 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:51 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 07:02:56 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 07:02:56 failed to read JSON message: failed to get reader: use of closed network connection client
[... 491 lines cut ...]
[pulse ] TRACE 2025/03/04 11:33:07 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 11:33:07 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 11:33:12 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 11:33:12 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 11:33:18 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 11:33:18 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 11:33:23 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/04 11:33:23 failed to read JSON message: failed to get reader: use of closed network connection client
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.200.5

chrstmnn, 2025-03-04:
Tibber advices the following for implementing websocket subscriptions:

> Best practices for implementing websocket subscription clients
> When implementing a client for subscribing to live data there are a few scenarios to consider:
> 
> No data due to networking issues: it may be wise to implement reconnect mechanism in case no data has been received for several minutes. **During a retry, the old websocket connection must first be destroyed and proper jitter and exponential delay implemented.**
> A realtime device may get removed: it is recommended to check that home.features.realTimeConsumptionEnabled has a true value always before reconnecting.
> A websocket connection may be forbidden due to invalid authentication token (e.g. a user may have removed her Tibber account). In this case the client must not retry with same token. Instead the authorization process must be ran again with user supplying new credentials.
> Tibber websocket API restart may occur which means the connection is closed with code 1001 and reason Going away. In that case the client can reconnect after a 1-60 second random delay (jitter applied) to avoid all clients reconnecting at the same time.

Maybe this might be the problem.

andig, 2025-03-04:
Wir bräuchten den Teil vom Log, der zu dieser Situation geführt hat. Der Informationsgehalt der 1000 Zeilen ist ja sehr beschränkt…

chrstmnn, 2025-03-04:
Leider enthält das Log nicht mehr, da es scheinbar überschrieben wird.
Ich habe einen neuen Token erstellt und hoffe, beim nächsten Auftreten schneller im Log zu sein.

StefanSchoof, 2025-03-04:
Es gibt auch andere Wege an die Logs zu kommen (wenn dein Log Level zu dem Zeitpunkt ausreichend war): https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

andig, 2025-03-04:
Gerne wieder melden wenn der Teil davor mit erhalten ist!

closed 2025-03-04 by andig

chrstmnn, 2025-03-04:
Im Log finden sich nun bereits zwei erzeugte Verbindungen. Die erste entstand direkt beim Neustart, nachdem der neue API Token von mir gesetzt worden ist.
Danach gab es später einen Aussetzer der API, was dazu geführt hat, dass eine zweite Connection initialisiert wurde. Dies geschah innerhalb weniger Minuten mehrere Male, sodass bereits insgesamt 4 Connections aufgebaut worden sind, was letztendlich zur Sperrung auf Tibber-Seite führen wird.

Meines Erachtens müsste die bestehende Connection weiter v

[evcc-20250304-163343-trace.log](https://github.com/user-attachments/files/19073657/evcc-20250304-163343-trace.log)

erwendet oder zuerst geschlossen werden, bevor eine weitere aufgebaut wird.

```
[pulse ] TRACE 2025/03/04 16:02:23 POST https://api.tibber.com/v1-beta/gql
[pulse ] TRACE 2025/03/04 16:02:23 {"query":"{viewer{websocketSubscriptionUrl}}"}

{"data":{"viewer":{"websocketSubscriptionUrl":"wss://websocket-api.tibber.com/v1-beta/gql/subscriptions"}}}
[pulse ] TRACE 2025/03/04 16:02:23 {"type":"connection_init","payload":{"token":""}} client
[pulse ] TRACE 2025/03/04 16:02:24 {"type":"connection_ack"} server
[pulse ] TRACE 2025/03/04 16:02:24 {"id":"09219a96-9979-48be-b0b5-1cd12ce71a8f","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/03/04 16:02:25 {"id":"09219a96-9979-48be-b0b5-1cd12ce71a8f","type":"next","payload":{"data":{"liveMeasurement":{"power":265,"powerProduction":0,"lastMeterConsumption":29344.7452,"lastMeterProduction":0.8461,"currentL1":null,"currentL2":null,"currentL3":null}}}} server

[...]

[pulse ] TRACE 2025/03/04 16:04:31 failed to read JSON message: failed to get reader: context deadline exceeded server
[pulse ] TRACE 2025/03/04 16:04:37 {"type":"connection_init","payload":{"token":""}} client
[pulse ] TRACE 2025/03/04 16:04:37 {"type":"connection_ack"} server

[...]

[pulse ] TRACE 2025/03/04 16:06:14 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/04 16:06:20 {"type":"connection_init","payload":{"token":""}} client
[pulse ] TRACE 2025/03/04 16:06:20 {"type":"connection_ack"} server

[...]

[pulse ] TRACE 2025/03/04 16:32:13 failed to read JSON message: failed to get reader: context deadline exceeded server
[pulse ] TRACE 2025/03/04 16:32:18 {"type":"connection_init","payload":{"token":""}} client
[pulse ] TRACE 2025/03/04 16:32:18 {"type":"connection_ack"} server
[...]
```

GrimmiMeloni, 2025-03-04:
> Im Log finden sich nun bereits zwei erzeugte Verbindungen. 

Woran machst Du das fest? Im von Dir geteilten Log jedenfalls kann ich das nicht erkennen.

Die Client Library verwirft bei Neuinitialisierung zunächst alle Subscriptions (so fern noch welche aktiv sind) platt, und verwirft danach auch den WebSocket. In evcc parametrisieren wir den Client mit einem Delay von 5s für den Reconnect, was sich dann auch ebenfalls im Bereich dessen bewegt was die Tibber API vergibt.

chrstmnn, 2025-03-05:
Dass bestehende Subscriptions verworfen werden, konnte ich an der Stelle dem Log nicht entnehmen, und würde vermuten, dass da etwas nicht funktioniert, wenn man voraussetzt, dass Tibber die Connections korrekt zählt. 

Wenn ich EVCC abschalte und nur die Home Assistant Integration von Tibber laufen lasse, wird mein API Token jedenfalls (auch über längere Zeit) nicht gesperrt

GrimmiMeloni, 2025-03-05:
Nochmal - im Log gibt es keine zwei Verbindungen. Auch gibt es dort keine Fehlermeldung von der Tibber Seite über zu viele Verbindungen. Es ist weiterhin nicht nachvollziehbar wie Du zu dem Schluß kommst, daß hier zu viele Verbindungen (von evcc) geöffnet werden.

chrstmnn, 2025-03-06:
Hallo,
Danke, verstanden.
Seit heute Nachmittag erhalte ich erneut keine Daten mehr.
Vielleicht bringen diese Log-Auszüge Licht ins Dunkle:

Trace:

```
[pulse ] TRACE 2025/03/06 15:55:40 {"id":"c1a0ed2a-f000-48f8-90f1-3ee7a60b2564","type":"next","payload":{"data":{"liveMeasurement":{"power":247,"powerProduction":0,"lastMeterConsumption":29400.7195,"lastMeterProduction":0.8461,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/03/06 15:55:50 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/06 15:55:55 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/06 15:55:55 {"type":"connection_ack"} server
[pulse ] TRACE 2025/03/06 15:55:55 {"id":"a275710b-927c-45ca-8e47-361e24bb63fa","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/03/06 15:55:55 {"id":"a275710b-927c-45ca-8e47-361e24bb63fa","type":"next","payload":{"errors":[{"message":"unable to start stream api:d49ac8c5aa124bc19069033c351f5194 for device d49ac8c5-aa12-4bc1-9069-033c351f5194; userId ffa22741-47f4-42a6-a200-0702d58e00bd","locations":[{"line":1,"column":28}],"path":["liveMeasurement"]}]}} server
[pulse ] ERROR 2025/03/06 15:55:55 Message: unable to start stream api:d49ac8c5aa124bc19069033c351f5194 for device d49ac8c5-aa12-4bc1-9069-033c351f5194; userId ffa22741-47f4-42a6-a200-0702d58e00bd, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[pulse ] TRACE 2025/03/06 15:55:55 {"id":"a275710b-927c-45ca-8e47-361e24bb63fa","type":"complete"} server
[pulse ] TRACE 2025/03/06 15:55:55 no running subscription. exiting... client
[pulse ] TRACE 2025/03/06 15:56:00 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/06 15:56:00 {"type":"connection_ack"} server
[pulse ] TRACE 2025/03/06 15:56:00 {"id":"ecbf5f18-5514-4a03-aafb-97a8dc4e3635","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"***"}}} client
[pulse ] TRACE 2025/03/06 15:56:00 {"id":"ecbf5f18-5514-4a03-aafb-97a8dc4e3635","type":"next","payload":{"errors":[{"message":"unable to start stream api:d49ac8c5aa124bc19069033c351f5194 for device d49ac8c5-aa12-4bc1-9069-033c351f5194; userId ffa22741-47f4-42a6-a200-0702d58e00bd","locations":[{"line":1,"column":28}],"path":["liveMeasurement"]}]}} server
[pulse ] ERROR 2025/03/06 15:56:00 Message: unable to start stream api:d49ac8c5aa124bc19069033c351f5194 for device d49ac8c5-aa12-4bc1-9069-033c351f5194; userId ffa22741-47f4-42a6-a200-0702d58e00bd, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[pulse ] TRACE 2025/03/06 15:56:00 failed to read JSON message: failed to read: use of closed network connection client
[pulse ] TRACE 2025/03/06 15:56:06 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/03/06 15:56:06 failed to read JSON message: failed to get reader: use of closed network connection client
[pulse ] TRACE 2025/03/06 15:56:11 {"type":"connection_init","payload":{"token":"***"}} client
```

Debug:
```
[pulse ] ERROR 2025/03/06 15:55:55 Message: unable to start stream api:d49ac8c5aa124bc19069033c351f5194 for device d49ac8c5-aa12-4bc1-9069-033c351f5194; userId ffa22741-47f4-42a6-a200-0702d58e00bd, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[pulse ] ERROR 2025/03/06 15:56:00 Message: unable to start stream api:d49ac8c5aa124bc19069033c351f5194 for device d49ac8c5-aa12-4bc1-9069-033c351f5194; userId ffa22741-47f4-42a6-a200-0702d58e00bd, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
```

GrimmiMeloni, 2025-03-06:
Das ist ein Tibber seitiges Problem. Das haben wir jetzt schon häufiger gesehen. Zwischendurch werden die sonst korrekten Details zum subscriben einfach abgelehnt. Basierend auf längeren Logs die einige User geteilt haben, gehe ich davon aus das sich dies wieder von allein fängt. Mehr als es erneut versuchen kann evcc hier nicht tun.